MiczFlor/RPi-Jukebox-RFID #581: Remote control with MQTT

codegrau, 2019-09-26:
Hi there, thank you @MiczFlor and all contributors for this wonderful project. 

Since mqtt is very common for smart home and home automation stuff I was wondering: wouldn't it be cool for phonieboxes to connect to a mqtt broker for other devices to know, which tune is playing and control phoniebox remotely?

- Publish "now playing" to a selected mqtt topic
- Subscribe to a mqtt topic for remote control 

This way you could easily integrate phoniebox to openHAB or HomeAssistant or even use cheap ESP8266 to control or visualize phoniebox contents.

Looking forward to your thoughts.

andreasbrett, 2019-09-27:
Awesome idea! Would love to use this as well with my openHAB setup.

MiczFlor, 2019-09-28:
@codegrau @andreasbrett 
It's a great idea. I have no smart home, it's nice here, but not very smart ;) so I can't help with this idea. Makes sense though and I will be hovering over the pull requests - possibly you can create a small task group.

andreasbrett, 2019-09-28:
I'll see if I can contribute to this feature. I need to dig into the phoniebox infrastructure first and see how and where to insert the mqtt stuff.

@codegrau any ideas on what exact features we would need to control/monitor via mqtt?

andreasbrett, 2019-10-04:
@codegrau 
I think we can close this issue as it is implemented now.

codegrau, 2019-10-04:
Sorry for my late response, but what you've achieved @andreasbrett is incredible! This is exactly what I meant! Thank you for this one!

andreasbrett, 2019-10-04:
@codegrau thanks, it was easier than I thought. Let me know if there's something else that could be done with regards to MQTT.

whenthelight, 2019-10-04:
Hey  @ all i had massive problems with Openhab and Mosquito to get and read States (Commands are without Errors and i have it run till google Home can control my Box :-) , thats so awesome), so test step by step and figured some errors (dont know its only me on rasbian buster?) :
1st: its not phoniebox/attributes/attributeName ist phoniebox/attribute/attributeName
2nd: with sample Conf for service in systemd the working Directory must be WorkingDirectory=/home/pi/RPi-Jukebox-RFID/scripts
not
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
without the directory scripts you wont get attribute Values

Hope this help someone

Thanks to dev for this awesome script and Idea of Mqtt

andreasbrett, 2019-10-04:
> 1st: its not phoniebox/attributes/attributeName ist phoniebox/attribute/attributeName

Fixed the typo in the wiki, thanks for pointing that out!

> 2nd: with sample Conf for service in systemd the working Directory must be WorkingDirectory=/home/pi/RPi-Jukebox-RFID/scripts
> not
> WorkingDirectory=/home/pi/RPi-Jukebox-RFID

Okay, I think I found the bug with that configuration. Typically my scripts are independent from the working directory as I use the script path. Thanks to your hint I found 3 attributes where I did NOT do so. maxvolume, idletime and volstep. Did you have issues with other attributes as well? Those 3 bugs will be fixed once @MiczFlor merges the pull request () into master. It contains some new features and I also added the fixes a minute ago.

whenthelight, 2019-10-04:
> Fixed the typo in the wiki, thanks for pointing that out!
>
>
> Okay, I think I found the bug with that configuration. Typically my scripts are independent from the working directory as I use the script [...]

Hey, so fast :-) i couldn't drink my Pale and have to sit down for testing .
The Problem with the "scripts" path is wenn i start the service the "get/all" doesnt get Values in al Attributes. Not only these 3 but when it has merged i test some Attributes. actually this error appears only in the "get" and "attribute" topics , the cmd have no error and delivers without error. Also in the Syslog i see "get/all" empty Value . When start directly with python3 "script" no error and all topics become their Values

thx

andreasbrett, 2019-10-04:
> PS: the new Function for Folderplay(which is incredible when i get this work as Switch and then in Google Home :-)) need some Parameters , are these the directory Path to Musicalbums ? like : /RPi-Jukebox-RFID/shared/audiofolders/Album1 or only Album1 ?

You simply pass the folder name (Album1). BTW: I primarily use the Basic UI to control openHAB and therefore would implement that with a String element that is linked to the MQTT topic "phoniebox/cmd/playfolder" and make a dropdown in the sitemap with pre-set folder names. Something like that:

	Selection item=phoniebox_playfolder mappings=["Album1"="Tool - Fear Inocolum", "Album2"="Slipknot - We are not your kind"]

Not sure how you would handle that with your Google Home though.

> Update: Change your Commit et voila , thumbsup all works like charm.

Great!

KingKahn123, 2019-10-06:
Hi, I still have an issue with the MQTT function. I have connected the Phoniebox to my MQTT server of my IPSymcon system, but only 4 entries are created:

[image]

The phoniebox/get/all attribute gets updated every 10 seconds when a music file is playing. However no data is written to this attribute. In the log files I can see that the payload for this attribute is NULL. 

Any idea why no information is forwarded to this attribute?

Thx for your help...

Kai

andreasbrett, 2019-10-06:
@kingkahn123
You misunderstood the documentation. Attributes are sent to /attribute/$attributeName (so e.g. /attribute/volume or /attribute/elapsed).

/get/$attributeName only triggers phoniebox to publish that attribute to /attribute/$attributeName. That's only used if you want to receive the updates quicker or at a specific time rather than waiting for the next interval. What you will have to do though is set your smart home tool to monitor each attribute separately (via e.g. /attribute/volume).

andreasbrett, 2019-10-06:
Oh and I see that you're using the Spotify edition. Haven't tested that yet as I'm not using it. Let me know if it works like I described above.

KingKahn123, 2019-10-06:
I have created now one additional attribute manually:

[image]

The details look like this:

[image]

However no updates are received for this attribute.

KingKahn123, 2019-10-06:
In addition, as far as I understand the MQTT function from the Phoniebox is that all attributes are refreshed at a predefined interval, correct? That means during playback of any music file, the attributes are sent to the server every 10 seconds (depending on the config file). This would be recognized by my SmartHome system and the attributes would be created automatically. But only the four are created. So I assume that no data is transferred to my SmartHome system.

whenthelight, 2019-10-06:
Hey, in Mqtt you have to subscribe to an Topic to receive an Attribute. You have to make an Status or SUB in your Software with phoniebox/attribute/Volume an no Value.

It seems that you set up an public with that Topic above and an Value

KingKahn123, 2019-10-06:
But this is how my other attributes, where I receive some data from the Phoniebox, are defined.

[image]

---------

Received information on startup:

[image]

This is how my MQTT server receives the publish get/all from the Phoniebox. The payload is empty. Should it not include some data?

   [image]

whenthelight, 2019-10-06:
Sure ;-) because you use publish to publish plusSpotify to the Topic phoniebox/edition
You need an Function in your Ip_symcon that catchs the Values from the Topics not send some.
Like https://github.com/thomasf68/IPS_MQTT. 
You received this in your mqtt server because you published that  with your Action.

andreasbrett, 2019-10-06:
> Should it not include some data?

No. Please read again the documentation and my description above. The get topic NEVER is filled with payload because the topic itself will trigger an action and does not need a payload as all the necessary information is already in the subtopic (attributename). The reason why your tool detects this topic is because the periodic publishing of attributes is triggered by phoniebox sending /get/all to itself. Just ignore that.

The reason you're not receiving any attributes has to do with you being on the Spotify edition. I will have to check how the management of the player (which I guess is different to the classic edition) works for the Spotify edition.

andreasbrett, 2019-10-06:
@whenthelight He is not publishing anything. His tool just subscribed to all possible topics and phoniebox indeed is sending 4 parameters at the beginning (independent of the player attributes).

Reason why he does not receive any attributes is because that function seems to crash due to some difference between the classic and Spotify edition.

@miczflor
Are there major differences on how I can retrieve player data in the Spotify edition? As you see in my code I mostly use queries to mpd as well as the payout controls.sh script.

KingKahn123, 2019-10-06:
> Sure ;-) because you use publish to publish plusSpotify to the Topic phoniebox/edition
> You need an Function in your Ip_symcon that catchs the Values from the Topics not send some.
> Like [...]

This mqtt module was now integrated in IPSymcon 5.1/5.2. No need to have this module any longer ;-) When I create this attribute initially the value field is empty... After reboot of the Phoniebox the value is filled in by the MQTT publish:

[image]

KingKahn123, 2019-10-06:
> @whenthelight He is not publishing anything. His tool just subscribed to all possible topics and phoniebox indeed is sending 4 parameters at the beginning (independent of the player attributes). [...]

I think so as well as I now checked the log file of my IPSymcon MQTT server and I do not see anything else than the 4 parameters. The get/all is occurring every 10 seconds. But I now understood that it is ok, if it is empty ;-)

Thanks for looking into it

whenthelight, 2019-10-06:
> @whenthelight He is not publishing anything. His tool just subscribed to all possible topics and phoniebox indeed is sending 4 parameters at the beginning (independent of the player attributes).
 **Hey im blind :-) didnt see that he is RX the Publish , sorry for missing this. I read the manual of the IPsymcon :-)** 
> Reason why he does not receive any attributes is because that function seems to crash due to some difference between the classic and Spotify edition.
>
> @MiczFlor
> Are there major differences on how I [...]

What is the output of journalctl -u phoniebox-mqtt-client.service there you see whats going on

andreasbrett, 2019-10-06:
@kingkahn123
Could you please try the latest version of the mqtt daemon? It's not yet merged but includes some major changes that might already get rid of the errors. 

You can get it here:
https://github.com/andreasbrett/RPi-Jukebox-RFID/blob/mqtt-dev/scripts/daemon_mqtt_client.py

KingKahn123, 2019-10-06:
> What is the output of journalctl -u phoniebox-mqtt-client.service there you see whats going on

[image]

andreasbrett, 2019-10-06:
You won't see the errors there as the function that causes the issue is called from a thread and not showing error messages on stdout/stderr.

Please try the latest dev version (see above) and check journalctl again. If I'm correct it should give more output and also attributes should pop up. If not... I will have to dig deeper.

whenthelight, 2019-10-06:
Latest Dev gives a lot of Informations :-)
[image]

whenthelight, 2019-10-06:
Spotify Edition use mopidy, customized Mpd 

https://docs.mopidy.com/en/latest/api/core/#mopidy.core.mopidy.core.PlaylistsController

KingKahn123, 2019-10-06:
Using the new script results in only two more attributes:

[image]

journalctl now only shows this:

[image]

whenthelight, 2019-10-06:
All Music related Attributes are in the Spotify Edition others then normal, because mopidy act different then normal MPD.Thats why the Script cannot geht the Values for the Attributes. Thats why you get only the System Default Values

andreasbrett, 2019-10-06:
@whenthelight
But how does phoniebox web app work then? It uses the same ways of accessing player status.

@kingkahn123 
Could you please add `fetchData()` at the bottom of the script right after `client.loop_start()`? This will allow us to see the error in journalctl.

KingKahn123, 2019-10-06:
> @KingKahn123
> Could you please add `fetchData()` at the bottom of the script right after `client.loop_start()`? This will allow us to see the error in ` journalctl -u phoniebox-mqtt-client.service`

I have added it and this is the only output:

[image]

andreasbrett, 2019-10-06:
That's weird. I honestly don't know what to do right now. Are you able to see and control your box in the web app? Additionally, could you run the following command:

`nc - w 1 localhost 6600`

Then enter `status` and press enter. This should output some lines of attributes of your mpd/mopidy server. Then enter `close` and then press enter to close that tool.

KingKahn123, 2019-10-06:
Yes, I can control the box via the web app...

> nc -w 1 localhost 6600 -->

[image]

status gives following "result":

[image]

andreasbrett, 2019-10-06:
No, enter status while still in the mpd console (that you opened with nc)

KingKahn123, 2019-10-06:
It is not opening anything:

[image]

andreasbrett, 2019-10-06:
My bad. Use that command: `nc localhost 6600`

KingKahn123, 2019-10-06:
Ah ;-)

[image]

andreasbrett, 2019-10-06:
Okay, getting closer. Please save the following code to a file and name it test.py

	import subprocess
	
	cmd = ['nc', '-w', '1', 'localhost', '6600']
	input = 'status\ncurrentsong\nclose'.encode('utf-8')
	status = subprocess.run(cmd, stdout=subprocess.PIPE, input=input).stdout.decode('utf-8')
	print("result =", status)

Then run that with `python3 test.py`

KingKahn123, 2019-10-06:
Done:

[image]

andreasbrett, 2019-10-06:
Alright, that's a good thing. 

Now please add the following lines to test.py and run it again. 

	regex = re.search('\nstate: (.*)\n', status).group(1).lower()
	print("regex =", regex)

KingKahn123, 2019-10-06:
Where should I add it? Right at the end of the script after: print("result =", status) ?

andreasbrett, 2019-10-06:
Yes, right there.

KingKahn123, 2019-10-06:
Ok then I get following error message:

[image]

andreasbrett, 2019-10-06:
Sorry change the first line of the script to

`import subprocess, re`

andreasbrett, 2019-10-06:
(sorry for all the hassle. I'm on my smartphone right now and can't properly help)

KingKahn123, 2019-10-06:
no worries... Thx for the help:

[image]

andreasbrett, 2019-10-06:
Okay, that all looks very good. Could you try running the mqtt script?

First stop the service with `sudo systemctl stop phoniebox-mqtt-client`

Then run the script manually with
`python3 daemon_mqtt_client.py`

KingKahn123, 2019-10-06:
[image]

andreasbrett, 2019-10-06:
Okay, this looks like the subscription to the /get/# topic did not work. Which mqtt server are you using? Could you try and send to /get/volume (any payload) from your smart home software? The daemon script needs to run while you're doing it. It should output that it received something.

KingKahn123, 2019-10-06:
Ok that worked:

[image]

And my MQTT Server (IPSymcon) received the publish and saved it in the attribute:

[image]

andreasbrett, 2019-10-06:
Alright, I think I fixed it. Please download the script once again from latest dev (see link above).

KingKahn123, 2019-10-06:
Perfect! It is working now ;-)

[image]

andreasbrett, 2019-10-06:
Awesome! Thanks for the patience and enjoy it. Let me know if there's something missing that you need/want.

KingKahn123, 2019-10-06:
Are you kidding me?! Thanks for your patience and your help!

MiczFlor, 2019-10-08:
I committed the last pull requests to master:
#600

MiczFlor, 2019-10-08:
And thanks for the lively discussion and great input!

whenthelight, 2019-10-10:
> > PS: the new Function for Folderplay(which is incredible when i get this work as Switch and then in Google Home :-)) need some Parameters , are these the directory Path to Musicalbums ? like : /RPi-Jukebox-RFID/shared/audiofolders/Album1 or only Album1 ?
> 
> You simply pass the folder name (Album1). BTW: I primarily use the Basic UI to control openHAB and therefore would implement that with a String element that is linked to the MQTT topic "phoniebox/cmd/playfolder" and make a dropdown in the sitemap with pre-set folder names. Something like that:
> 
> ```
> Selection item=phoniebox_playfolder mappings=["Album1"="Tool - Fear Inocolum", "Album2"="Slipknot - We are not your kind"]
> ```
> 
> So if I select the Slipknot album from that dropdown the value "Album2" would be sent to the `playfolder` command. Not sure how you would handle that with your Google Home though.
> 

Hey andreasbrett, If i select only 1 Album , i can handle this as Switch with 1 Value (Album1).
Items create as Switch et voila my Son is happy ;-) . Volume are configured as Dimmer  and Lightning. The Reaction Time is awesome fast. Sleeptimer also as Switch with Value in my Case 60 for 60min. Play and Pause also works good as Switch.

georgbachmann, 2019-10-13:
Hey there... I love the idea of controlling my sons phoniebox with mqtt (f.e. to make it work as an alarm-clock). THANK YOU very much for your effort!!!

Maybe I was skimming over this issue here too fast... but is there a readme on how to activate the mqtt support? Like where to I configure my broker?

georgbachmann, 2019-10-13:
Thank you guys a lot for this addon! I can't wait to realize a wake-up-alarm for my kids with it!
I can run it and it also works if I run it manually like so: `python3 RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py`.
But when I follow the instructions on the wiki page and then run `sudo systemctl status phoniebox-mqtt-client` at the end, I always get error messages:

```pi@phoniebox:~ $ sudo systemctl status phoniebox-mqtt-client
● phoniebox-mqtt-client.service - Phoniebox MQTT Client Service
   Loaded: loaded (/etc/systemd/system/phoniebox-mqtt-client.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2019-10-13 19:59:43 BST; 1min 42s ago
  Process: 531 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py (code=exited, status=203/EXEC)
 Main PID: 531 (code=exited, status=203/EXEC)

Oct 13 19:59:42 phoniebox systemd[1]: phoniebox-mqtt-client.service: Failed with result 'exit-code'.
Oct 13 19:59:43 phoniebox systemd[1]: phoniebox-mqtt-client.service: Service hold-off time over, scheduling restart.
Oct 13 19:59:43 phoniebox systemd[1]: Stopped Phoniebox MQTT Client Service.
Oct 13 19:59:43 phoniebox systemd[1]: phoniebox-mqtt-client.service: Start request repeated too quickly.
Oct 13 19:59:43 phoniebox systemd[1]: Failed to start Phoniebox MQTT Client Service.
Oct 13 19:59:43 phoniebox systemd[1]: phoniebox-mqtt-client.service: Unit entered failed state.
Oct 13 19:59:43 phoniebox systemd[1]: phoniebox-mqtt-client.service: Failed with result 'exit-code'.```
Do you guys have any idea what the problem might be?
I already tried rebooting... but didn't work.

[6 more replies]

closed 2023-04-29 by s-martin