MiczFlor/RPi-Jukebox-RFID #583: Spotify LOCAL Media RFID

dMopp, 2019-09-28:
How do i link "LOCAL" Media to RFID with Spotify enabled? The "link" i get seems to be quite useless and linking the old way (assign audio folder) does not playback the folder. Any trick iam missing?

MiczFlor, 2019-10-12:
hi @dMopp 
did you resolve this?

dMopp, 2019-10-12:
No, not really :( I dont really know the right synbtax for local spotify playlists

domu83, 2019-10-13:
same issue on my site... 
I haven't performed any changes, so I'm wondering why all of the sudden the audio folder assignment does not work anymore?!

paulrigger, 2019-10-20:
same issue here:

debug file when trying to play a local file via the WEB-UI: 

  ######### SCRIPT inc.readArgsFromCommandLine.sh (2019-10-20.12:56:57) ##
  # VAR CARDID: 
  # VAR COMMAND: savepos
  # VAR FOLDER: 
  # VAR VALUE: 
#START##### SCRIPT resume_play.sh (2019-10-20.12:56:57) ##
VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
VAR COMMAND: savepos
VAR VALUE: 
VAR FOLDER: 
VAR FOLDER from settings/Latest_Folder_Played: 
  content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders//folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: savepos
  savepos FOLDER: 
#END####### SCRIPT resume_play.sh (2019-10-20.12:56:57) ##
  #START### SCRIPT inc.settingsFolderSpecific.sh (2019-10-20.12:56:57) ##
  # VAR FOLDER from settings/Latest_Folder_Played: 
  # Folder exists: /home/pi/RPi-Jukebox-RFID/shared/audiofolders//folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  # SINGLE TRACK PLAY: OFF
  # # CHANGING: mpc single off
  # SHUFFLE FOLDER: OFF
  # # CHANGING: mpc random off
  #END##### SCRIPT inc.settingsFolderSpecific.sh (2019-10-20.12:56:57) ##
  ######### SCRIPT inc.readArgsFromCommandLine.sh (2019-10-20.12:56:58) ##
  # VAR CARDID: 
  # VAR COMMAND: resume
  # VAR FOLDER: 
  # VAR VALUE: 
#START##### SCRIPT resume_play.sh (2019-10-20.12:56:58) ##
VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
VAR COMMAND: resume
VAR VALUE: 
VAR FOLDER: 
VAR FOLDER from settings/Latest_Folder_Played: 
  content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders//folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: resume
  entering: resume with value OFF
#END####### SCRIPT resume_play.sh (2019-10-20.12:56:58) ##
mpc load TanteAndante && /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh -c=resume
echo  > /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played
VAR Latest_Folder_Played:

paulrigger, 2019-10-20:
The installation script does not configure MPD and the script tries to start it via MPD. But I also did not get it to work with configuring MPD. The Latest_Folder_Played seems to stay empty

ngblume, 2019-10-20:
Do I understand it correctly, that currently the MPD is not configured, in turn diabling playback of local files, since their playback is controlled via MPD?
I got sound from my system via IRIS and Mopidy... So the Spotify part seems to play...
I remember reading somewhere, that with mopidy the indexing of local files would be different.. 
Why isn't the playback of local files also done via mopidy?

Thanks!

paulrigger, 2019-10-20:
Somehow some mp3s startet to work for me, others don't. Could not yet figure out why.
By pressing "Scan Music Library" a playlist file is created in the playlist folder. If you look at the file it either has local files in them "local:track:test2/test2.mp3" or "Spotify:track:id"

ngblume, 2019-10-20:
What I tried was scanning the library and the trying to trigger playback
from the web interface.. That didn't work for all local tracks..

MiczFlor, 2019-10-22:
I changed the install script #611
There was a glitch of logic in there. Now mpd and spotify are not dependent on each other anymore.
If somebody has the knowledge to give this a second set of eyes, please do so.
The new install script can be found here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/installscripts/buster-install-default.sh

MiczFlor, 2019-10-22:
# local mp3 files with Spotify
@ngblume @paulrigger 
Please read here:

https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Spotify-FAQ

>MP3 (local music) handling has completely changed for the +Spotify Edition. The tracks need to be indexed by Mopidy & Mopidy-Spotify (this is the part o...

ngblume, 2019-10-22:
Hey,

thanks for the possible fix.
As i said, I did run the scan. But that didn't change anything regarding playback of local files.
Directly setting them for playback in the web UI resulted in no playback (I'm assuming that such a playback would be handled via MPD).
When I used IRIS (and therefore Mopidy) for Spotify, playback worked like a charm.

Not sure, if me using the HifiberryDAC+ might also add to the problem, but Mopidy worked fine with Hifiberry DAC+.

@MiczFlor Why not use Mopidy for all playback?
And what kind of insights are you looking for...?
I will try and see if I can run a clean install and test your fix...

Cheers
Niels Göran

dMopp, 2019-10-22:
All I need is a howto add „local“ files to card with +Spotify. Iam sure there is a syntax (for the url) I don’t know... 

Mit freundlichen Grüßen
Daniel Quolke

ngblume, 2019-10-22:
Based on what I can see in IRIS (and that works so far) it should be along the following lines:
local:track:Folder_1/Folder_2/File.mp3 (relative to your specified media_dir for mopidy..)

dMopp, 2019-10-22:
And for an album? (I have lots of Christmas CDs ripped :D)

Mit freundlichen Grüßen
Daniel Quolke

ngblume, 2019-10-23:
I get the feeling, that you add the URIs of all tracks when trying to play an album, since I can remove individualy tracks from playback after having set the album for playlist..

MiczFlor, 2019-10-23:
Hi @ngblume @dMopp 
adding local files should work by assigning RFID IDs to folders via the web app.
Does the web app play local files?

domu83, 2019-10-23:
sorry to hi-jack... does this fix also apply to non-buster installations?

thx for your continuous  support!!

dMopp, 2019-10-23:
Nope. Does not work. No playback with local files. 

Mit freundlichen Grüßen
Daniel Quolke

restless1987, 2019-11-11:
Any updates here?

MiczFlor, 2019-11-11:
Hi, quick input from me: I don't recall that the issue with local files on a spotify edition Phoniebox existed before. If anything, in the link above, it clearly describes how to work with local files. Having said this, you guessed it, I personally do not have a spotify account, so I never tested it, this was very much a community contribution. Anybody out there who has local files working with spotify?

Could it be a `mopidy` issue?

restless1987, 2019-11-12:
Just to check that I got it right: When I use the spotify variant, I need to add local files by updating `/etc/mopidy/mopidy.conf` with a valid `[files]` config.

After that, I can play those files via the IRIS webinterface (and add it as a spotify link like `file:whatever...`)?

closed 2020-01-18 by MiczFlor

dMopp, 2020-01-18:
Why closed? Iam still missing a clear howto map songs to rfid over webui?

splitti, 2020-01-18:
dMopp, maybe this is the cause of a corrupt installation. Could you reinstall and of course, save all your data!!!
I made a fresh install and could use local media directly. You asked for a howto:

1. Copy Audio-Files to phoniebox_audiofile Folder
2. Swipe my Card and go to the Web Interface of my box to Card ID
3. Here i just change option a to the folder i want and save
4. The most missing step is to go into the Web Interface to Folders&Files and click on "Scan Music Libary".
5. That's it, play local Music...

Tested with the last Installer...