MiczFlor/RPi-Jukebox-RFID #588: Resume on as default setting

iwishitwassummer, 2019-09-29:
Is ist correct to change _scripts/inc.writeFolderConfig.sh_
`        RESUME="ON"` in the folderConf
to enable to? Or is there another setting to take care that resume is always on, in every folder and automatically created?

MiczFlor, 2019-09-29:
Hi @iwishitwassummer 
good point. It can not be set globally and if you want resume to be the default, yes, you would have to change it in this line:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/abca5b935490f10b10f448b734696a70000cb1ab/scripts/inc.writeFolderConfig.sh#L80

There is a settings page where you can set the behavior of the "second swipe" in the web app. Not the resume default setting though. I only mention this, because the two are interlinked a bit:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#settingssecond_swipe

iwishitwassummer, 2019-09-29:
ok, thanks

closed 2019-09-29 by iwishitwassummer