MiczFlor/RPi-Jukebox-RFID #589: Configuration via one settings file

veloxidSchweiz, 2019-09-29:
Hey,
Thnaks for this great project. It is a really nice use of the RPis and offers many different possibilities. 
I am playing around with the RPi Jukebox a lot and try to find a soutions where everything is done more or less with GPIO pins.
In order to have an overview over all this pins and the configurations options, I would be great to use a single configuration file for all such settings, by only changingf this file it would be much easier to "switch" between configurations.
Almost every handler uses python and with the ConfigParser there is already a module which is really powerful and would probably simplify many things. For example one could easily generate different settignsfile for the different suggested GPIO configurations.
What do you think about such a change. I am already using it more or less for my version and I am happy to share it if there is interest.
cheers

MiczFlor, 2019-10-08:
hi @veloxidSchweiz 
I like the idea. I am already parsing bash config files when running parts in php. So I am sure that there could be "one file to rule them all".
My only concern: depending on the hardware used (dial, buttons) combined with the use of soundcards that require GPIO pins, there have been many different flavours of Phoniebox UIs - and therefore configs. But please show me what you got :)

s-martin, 2020-04-12:
@veloxidSchweiz, this is being implemented in #763, right?

veloxidSchweiz, 2020-04-13:
At least the GPIO settings...
at the moment i would suggest to go for one config file for each matter of concern.
If we use the ~/.config/phonniebox folder we have them all at one point....
The version I have in my fork at the moment seems to work pretty good.
I planned to also add the 'special' Buttons as shutdown button etc. and the change from config to yaml files isn't done it. Otherwise it seems to work quite good: I tested rotary-switches, Two-Buttons and normal buttons already and i added a simple install script..
But due to time constrains in the current times I do not find the time I would need ;(
I

At the moment I am using cfg files and no yaml yet as I

s-martin, 2020-04-13:
Config file for each area is fine for me.

Moving content and config files to dedicated folders is already being addressed in #797.

s-martin, 2023-12-11:
This was implemented in future3.

closed 2023-12-11 by s-martin