MiczFlor/RPi-Jukebox-RFID #592: Can't get audio working

methigel, 2019-10-02:
Hello folks,
I set up a phoniebox installation with spotify. And can not get audio to work. I use the PCM device ist is configured and the systemsound is working. I can play mp3 in vlcplayer. I've set PCM to force in raspi-config. I've configured evrything like description in troubleshooting FAQ.
I've tried to play sound via mpc play command and alway get the error: "error parsing song numbers: example.mp3"
I've installed the example sounds and streams, but nothing worked not via webapp-play and not via RFID-tag.

Has anybody an idea how I can solve the problem?

Greets Marc

Featherwhiter, 2019-10-02:
Hi,
check if your Phoniebox folder contains the 'playlists' folder. 
After updating to the newest version I was missing this folder and this made mpd crash when trying to play sound.

Fl0ete, 2019-10-02:
> Hi,
> check if your Phoniebox folder contains the 'playlists' folder.
> After updating to the newest version I was missing this folder and this made mpd crash when trying to play sound.
> 
> Edit, this is the folder structure:
> /home/pi/RPi-Jukebox-RFID/playlists

solved my issue also described in issue #575.

thanks

Featherwhiter, 2019-10-02:
Glad it worked, I spend hours searching for this issue..

You're welcome!

M0D-F, 2019-10-03:
me too,... thank you !

MiczFlor, 2019-10-06:
Kudos for figuring this out so quickly. While I was away over a long weekend, it seems others used the long weekend to roll up their sleeves and get into their Phoniebox setup :)

closed 2019-10-06 by MiczFlor