MiczFlor/RPi-Jukebox-RFID #594: Sound I think is working but... no audio.

MarkIpswich, 2019-10-02:
Using this version

 2018-03-13-raspbian-stretch.zip

With the Phoniebox Classic (stretch):

RPi Model 3+

No soundcard added to it, just the built in one via the jack thing.

I set this up today for the web streams and that I haven't done what was suggested after completing installing (connecting to usb hub etc) it's power via a power bank (one that works fine) and that I have a 1 volt speaker (not sure if that's enough) and connected it straight from the power bank and the RPi powered from it also. The card reader powered from the RPi itself.

When registering a card and doing the swipe, nothing plays including in the web app.

Though when using some headphones with the jack in the RPi, I get sound but the sort of "mmm" sound if you know what I mean.

So the audio does work, well the sound does (via headphones with the jack, not speaker) but I am getting no audio itself from when swiping the card.

Help please!

Oh... and

I am using a Neuftech USB RFID Reader ID and when checking to see if it works alright (when starting to install the reader etc) and selecting a number from a list it just came up with some numbers and letters and not something like "USB reader" is that an issue? I installed it with the reader plugged into the RPi via one of it's USB ports.

AND...
I tried it powered via a power bank and I get two beeps instead of one, what does this mean?

Featherwhiter, 2019-10-03:
Hi,
I'll try to get your issues sorted:

1. No sound when swiping card
Have you tried if there really is working sound for you in general? For example, have you tried playing a sound file via VLC player or anything like that through headphones and speakers to rule out any issues with the sound and hardware in general?

If all that is working for you, then you should also have a look at your Phoniebox folder structure. Is there a folder called 'playlists'? If not, create it. You can see the issue right before yours, this was the same problem.

2. Hardware naming of the RFID reader
In my case, the reader is a random mix of letters and numbers, so no, that's not an issue, it's normal.

3. Two beeps instead of one
Like, from the reader? I honestly have no clue here, but so far I didn't use a power bank, so I have no experience here.

4. Changing volume
For me, the startup sound is always pretty loud. But the volume for playing files is kept at the level I used previously. I am not sure right now if that is a setting you can choose somewhere, maybe look around in the wiki here or in your web interface and see what you can find.

MarkIpswich, 2019-10-03:
I haven't done method one.

How would I play the sound file please and also to check to see if there is a "playlist" folder and if there isn't one, how to create it?

Thank you.

MarkIpswich, 2019-10-03:
Checked to see if there is a playlist.

No such file or directory.

Exactly what do I type to create one?

Featherwhiter, 2019-10-03:
I guess this should work, if you are in the Phoniebox directory:

mkdir playlists

If you are in the root directory you can add the whole path:

mkdir /home/pi/RPi-Jukebox-RFID/playlists

A reboot is a good idea, if you don't want to reboot you can also restart mpd.

service mpd restart (I hope that's correct)

MarkIpswich, 2019-10-03:
You mean the Phoniebox directory as in being in the terminal?

If not then what's the path to it as I don't know.

MarkIpswich, 2019-10-03:
Done it and it works..... well it plays via the web app but when swiping a card I get no sound from the speaker, connected to either the power bank or the RPi itself.

Would a USB hub work or would it still do the same?

Same goes to headphones, no sound apart from the mmm noise.

I also don't hear the startup sound.

Help!

MarkIpswich, 2019-10-04:
I think this may be the issue.
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/CONFIGURE-stretch#create-settings-for-audio-playout

I am just using a speaker and that's it. No phatbeat, or external soundcard.

I changed to speaker.

Then when coming to this https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/CONFIGURE-stretch#auto-start-the-phoniebox

This failed sudo systemctl enable phoniebox-startup-sound

and when finishing, the web app won't work.

Help.........

MarkIpswich, 2019-10-04:
Done it again with this one liner

For latest updates, try the version in the develop branch:

cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh

I changed this from echo "PCM" > /home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name

to echo "Mono" > /home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name

Is that correct?

as I choose the Mono in the install script as that's the only one that came up in the list.

Is that correct?

Now when doing this:

sudo systemctl start phoniebox-idle-watchdog
sudo systemctl start phoniebox-rfid-reader
sudo systemctl start phoniebox-startup-sound
sudo systemctl start phoniebox-gpio-buttons

The startup sound fails:

This is the error code:

**pi@raspberrypi:~ $ systemctl status phoniebox-startup-sound.service
● phoniebox-startup-sound.service - Phoniebox Startup Sound
   Loaded: loaded (/etc/systemd/system/phoniebox-startup-sound.service; enabled;
   Active: failed (Result: exit-code) since Fri 2019-10-04 12:00:19 UTC; 31s ago
  Process: 2966 ExecStart=/usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/start
 Main PID: 2966 (code=exited, status=203/EXEC)

Oct 04 12:00:19 raspberrypi systemd[1]: Starting Phoniebox Startup Sound...
Oct 04 12:00:19 raspberrypi systemd[2966]: phoniebox-startup-sound.service: Fail
Oct 04 12:00:19 raspberrypi systemd[1]: phoniebox-startup-sound.service: Main pr
Oct 04 12:00:19 raspberrypi systemd[1]: Failed to start Phoniebox Startup Sound.
Oct 04 12:00:19 raspberrypi systemd[1]: phoniebox-startup-sound.service: Unit en
Oct 04 12:00:19 raspberrypi systemd[1]: phoniebox-startup-sound.service: Failed
lines 1-12/12 (END)...skipping...
● phoniebox-startup-sound.service - Phoniebox Startup Sound
   Loaded: loaded (/etc/systemd/system/phoniebox-startup-sound.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2019-10-04 12:00:19 UTC; 31s ago
  Process: 2966 ExecStart=/usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3 (code=exited, status=203/EXEC)
 Main PID: 2966 (code=exited, status=203/EXEC)

Oct 04 12:00:19 raspberrypi systemd[1]: Starting Phoniebox Startup Sound...
Oct 04 12:00:19 raspberrypi systemd[2966]: phoniebox-startup-sound.service: Failed at step EXEC spawning /usr/bin/mpg123: No such file or directory
Oct 04 12:00:19 raspberrypi systemd[1]: phoniebox-startup-sound.service: Main process exited, code=exited, status=203/EXEC
Oct 04 12:00:19 raspberrypi systemd[1]: Failed to start Phoniebox Startup Sound.
Oct 04 12:00:19 raspberrypi systemd[1]: phoniebox-startup-sound.service: Unit entered failed state.
Oct 04 12:00:19 raspberrypi systemd[1]: phoniebox-startup-sound.service: Failed with result 'exit-code'.

-- Support: https://www.debian.org/support
--
-- Unit phoniebox-startup-sound.service has begun starting up.
Oct 04 12:00:19 raspberrypi systemd[2966]: phoniebox-startup-sound.service: Failed at step EXEC spawning /usr/bin/mpg123: No such file or directory
-- Subject: Process /usr/bin/mpg123 could not be executed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The process /usr/bin/mpg123 could not be executed and failed.
--
-- The error number returned by this process is 2.
Oct 04 12:00:19 raspberrypi systemd[1]: phoniebox-startup-sound.service: Main process exited, code=exited, status=203/EXEC
Oct 04 12:00:19 raspberrypi systemd[1]: Failed to start Phoniebox Startup Sound.
-- Subject: Unit phoniebox-startup-sound.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit phoniebox-startup-sound.service has failed.
--
-- The result is failed.
Oct 04 12:00:19 raspberrypi systemd[1]: phoniebox-startup-sound.service: Unit entered failed state.
Oct 04 12:00:19 raspberrypi systemd[1]: phoniebox-startup-sound.service: Failed with result 'exit-code'.**

Finished installing and the web app won't work!

So what is the problem?

HELP PLEASE. Thank you.

closed 2019-10-04 by MarkIpswich