MiczFlor/RPi-Jukebox-RFID #595: No Sound

Marmi86, 2019-10-03:
Hello,
I've installed phoniebox with spotify support,but the Box Play no Music with the linked tags.I testen spotify local music youtube .
I  created the playlists folder   with 777 authorization,no success
The RFID Reader works,I see  the rfid Code in the webgui and can link the Music with the tags

My Hardware
Raspian Stretch lite
Raspberry Pi 2
USB Sound Card 
RC522 RFID Reader 
newest one click installer for Raspian Stretch

MiczFlor, 2019-10-08:
Hi @Marmi86 
did you mix the stretch OS with the buster install script?
Currently, both options are available. If you use the stretch OS, you need to run this install script:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#raspbian-stretch

MiczFlor, 2019-11-11:
There was no answer, silence for a month, I am closing this thread.

closed 2019-11-11 by MiczFlor