MiczFlor/RPi-Jukebox-RFID #598: snd_usb_audio is missing. Help please ASAP. Thank you.

MarkIpswich, 2019-10-07:
I've installed all the one liner scripts but getting the same thing:

$ cat /proc/asound/modules
0 snd_bcm2835
1 snd_usb_audio

Number 1 is missing.

Please can someone tell me exactly why the file is missing?

Is there any chance (with telling me exactly what to do) to get the file put on there?

Thank you.

Do I need to have the USB mono speakers connected when installing the Phoniebox?

Is that why it's missing?

MarkIpswich, 2019-10-07:
Issue solved!

It was the audio output I was meant to do.

MarkIpswich, 2019-10-08:
Well that didn't work.

Tried this:

https://support.screenly.io/hc/en-us/articles/214547803-How-do-I-enable-analogue-audio-output-for-my-Raspberry-Pi-

But I don't see the "Manage Screen" anywhere.

Tried right clicking the sound icon and electing Analouge from HDMI and after rebooting, from headphones I didn't get the "mmm" sound and I heard the startup sound very clearly.

Though however when registering a card and swiping it it does NOT come up as the ID number where the "available audio" is, even though it did when I first installed it.

I am using the one liner for and classic version without spottily.

So why exactly is the ID number not coming up on the "Available Audio

Tried this -

Start - Perferances - Audio Device Settings - Select Controls - Tick PCM. OK. Swipe card, nothing.

BUT I had it with the audio still checked via the sound control at the top.... then I get the "mmm" sound and I can just hear the startup sound.

Then tried this, amixer cset numid=3 2

getting the "mmm" sound and no startup sound.

Using a 3.5mm jack via the headphones.

Please, please, please can someone tell me exactly what is wrong?

I have been trying to get this working every day since last Wednesday.

MarkIpswich, 2019-10-08:
Well I done this

Start - Perferances - Audio Device Settings - Select Controls - Tick PCM

Rebooted then I still got the "mmm" noise in headphones then when selecting the analogue via the sound at the top right then rebooted I head no "mmm" and heard the startup sound very clearly.

Then when I unplugged the HDMI from the tv and turned it on again via the powerbank I get the "mmm" noise again and hear the startup sound.

I register and swipe a card and I still get NO audio.

Do I also need to change the settings to force it to play via 3.5mm jack as I am using headphones?

MiczFlor, 2019-10-08:
Hi @MarkIpswich 
do you get any other audio on the Pi? Like playing YouTube? Or other audio files (outside of Phoniebox)?

MarkIpswich, 2019-10-08:
Downloaded YouTube audio, swiped a card, and nothing. No audio

MarkIpswich, 2019-10-08:
Playlist folder is there in correct directory.

MiczFlor, 2019-10-08:
Did you use the install script for buster on the stretch OS?

MarkIpswich, 2019-10-08:
I tried them all and I still get the problem.

At the moment, using the classic version without spotify.

MarkIpswich, 2019-10-08:
I just noticed that the Youtube folder, there is no files in there as it says in web app.

MarkIpswich, 2019-10-08:
What about this

mpc load "Sunshine Live"

Stream URL;

http://air.radiorecord.ru:8101/rr_128

(playing) #1/1 0:00/0:00 (0%)

Volume 100% repeat off random off single off consume off

mpc status

Volume 100% repeat off random off single off consume off

ERROR: failed to decode http://air.radiorecord.ru:8101/rr_128

MiczFlor, 2019-10-08:
Hi @MarkIpswich 

regarding "does it play at all?" I didn't mean downloading youtube in Phoniebox. I meant simply: without Phoniebox, is there any sound coming from your raspberry pi? Using the browser and go to youtube or play a file in an audio player?

And is there any chance that your Pi is offline? Can you visit websites? In other words: did you type in the right IP address and the like during the install procedure or Phoniebox?

The *ERROR failed to decode* is an issue with the MPD (music player daemon) and is not part of the Phoniebox problem you have. https://www.raspberrypi.org/forums/viewtopic.php?t=97755

Did you try the sample files delivered with Phoniebox?
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#sample-content

MarkIpswich, 2019-10-08:
Not I did not try the sample files.

When buring the image, I load the stretch, put the wifi details in, get the code and paste from browser. Then when rebooting, I can access the phoniebox.de site fine (takes a bit longer than usual) and any other site I can't. But before rebooting, I can.

Yes, all the IP address's are correct. Router IP, static IP... http://192.168.x.xx

Well I looked to see if I have a static IP address and it says dymantic. 

But the web app via http://192.168.x.xx loads fine without any problems.

MarkIpswich, 2019-10-09:
I just looked up the IP address via System Configuration and the last two numbers are different.

Would it be because of that?

MarkIpswich, 2019-10-09:
Well yes I got it working through headphones and now trying for the second time.

No audio!

MiczFlor, 2019-10-09:
Hi @MarkIpswich 
sorry to ask you this: but please do a fresh install :) I just made some changes to the install script.

* Re: audio. You got it working through the headphones. Then you say "no audio!". Can you explain what you mean?
* So you registered a card with a stream and that works. Ok, that's good. Then you say "I haven't done card ID with that one yet" - again, what do you mean?

To remove a folder in the terminal, type:
~~~
rm -rf /home/pi/RPi-Jukebox-RFID/shared/audiofolders/folderName
~~~
Replace folder name with your folder name. Note: if there are spaces in the folder name, you have to "escape" them. Like: `Folder Name` becomes `Folder\ Name` - you add a backslash before the space.

Editing a stream (hopefully doable in the web app soon). For now, in the terminal type 
~~~
nano /home/pi/RPi-Jukebox-RFID/shared/audiofolders/folderName/livestream.txt
~~~

MarkIpswich, 2019-10-09:
I mean I put the web stream link in (Sunshine Live) then tested it via headphones. 100% works.

Then I had to go out so about an hour later or so, it doesn't play.

I mean for Sunshine Live, I haven't made it available yet for swiping. Haven't set it up that way yet.

MiczFlor, 2019-10-09:
Ok, but the Phoniebox is online, you can for example
~~~
ping phoniebox.de
~~~
?

MarkIpswich, 2019-10-09:
Yes. I know that you can use the terminal on PC.

Is there anyway with doing it without a enternal cable?

MarkIpswich, 2019-10-10:
Yet another problem.

I removed the folder as suggested, now I can't change the volume in web app. Just stays at 0% when setting it.

Help please.

closed 2019-10-10 by MarkIpswich