MiczFlor/RPi-Jukebox-RFID #601: One script not working for spotify edition #buster & stretch

Noobelisk, 2019-10-09:
Hi, 
First things first: Wonderful Project! 
I installed Phoniebox on Raspian Stretch (which was working fine) and crashed it after playing around with some settings.  so I tried to restart from scratch with Buster. 
Unfortunatly the one line script is not working for Spotify edition any more: Installation is running fine, but there is no possibility to reach Iris any more in the webinterface - insted webinterface says "classic". 
I tried with Stretch (and according one line script) again and had the same issue. 
Script asks about credential for spotify, but in the web application it says "Phoniebox Classic". 

Can you please verify this issue and - in the best case - correct it?

MiczFlor, 2019-10-09:
Ah, good catch. Let me check.

MiczFlor, 2019-10-09:
Hi @Noobelisk 
could you do me a favour and manually edit the file 
~~~
RPi-Jukebox-RFID/settings/global.conf
~~~
Please change the EDITION line to:
~~~
EDITION="plusSpotify"
~~~

MiczFlor, 2019-10-09:
And @Noobelisk while you are at it, could you tell me what this returns?
~~~
cat RPi-Jukebox-RFID/settings/edition
~~~

Noobelisk, 2019-10-09:
Hi MiczFlor,
thx for your fast reply: 
Without tweaking the global.conf: 
`cat /home/pi/RPi-Jukebox-RFID/settings/edition`
plusSpotify

EDITION in global.conf was "classic" - after changing to  
`EDITION="plusSpotify"`
**everything works!** (Iris reachable and doing fine)

Thanks!!
currently running in STRETCH. Issue closed.

closed 2019-10-09 by Noobelisk