MiczFlor/RPi-Jukebox-RFID #607: (after booting) sound sometimes stuck at 0% (mpdsessions)

Hamsert, 2019-10-13:
Hello,
sometimes after booting the phoniebox the sound is stuck at 0%. I checked the mopidy.log:
`2019-10-13 18:58:52,250 INFO [467:MpdSession-16] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:58:53,282 INFO [467:MpdSession-17] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:58:54,696 INFO [467:MpdSession-18] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:58:54,705 INFO [467:MpdSession-19] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:58:55,729 INFO [467:MpdSession-20] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:58:56,893 INFO [467:MpdSession-21] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:58:58,297 INFO [467:MpdSession-22] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:58:58,304 INFO [467:MpdSession-23] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:58:58,453 INFO [467:MpdSession-24] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:58:59,317 INFO [467:MpdSession-25] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:58:59,337 INFO [467:MpdSession-26] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:00,290 INFO [467:MpdSession-27] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:00,305 INFO [467:MpdSession-28] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:01,308 INFO [467:MpdSession-29] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:01,327 INFO [467:MpdSession-30] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:02,292 INFO [467:MpdSession-31] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:02,296 INFO [467:MpdSession-32] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:03,313 INFO [467:MpdSession-33] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:03,338 INFO [467:MpdSession-34] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:03,382 INFO [467:MpdSession-35] mopidy.mpd.session: New MPD connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:04,289 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:04,301 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:04,315 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:04,331 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:06,301 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:06,304 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:06,339 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:06,343 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:08,471 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:08,481 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:08,483 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:08,496 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:08,500 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:09,338 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:09,351 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:09,449 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:09,665 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:09,667 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:09,688 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:09,695 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:09,781 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:11,663 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:11,667 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:11,752 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:11,765 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:13,673 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:13,681 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:13,703 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:13,706 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:14,736 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:15,661 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:15,663 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:15,682 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:15,687 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:17,661 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:17,662 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:17,681 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:17,710 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:19,662 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:19,665 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:19,680 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:19,682 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:19,754 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:21,662 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:21,667 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:21,677 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:21,681 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
2019-10-13 18:59:23,663 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
[... 97 lines cut ...]
2019-10-13 19:00:07,671 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600
[... cut ...]
2019-10-13 19:00:15,696 WARNING [467:MainThread] mopidy.internal.network: Rejected connection from [::ffff:127.0.0.1]:6600`
what I understand from that is, that there are 20 mpd sessions opened (first 20 lines) and after that further requests get rejected. This can only be fixed by a reboot. Note that this situation happens after booting without any card swiped. Sometimes this also happens after a while of usage.
Why are there so many requests, especially after booting? How to fix this?
Kind Regards and thank you for this Project
Matt

Hamsert, 2019-11-03:
Has no one an idea why this happens or how to fix it? It's really annoying because I can' reliably use the Phoniebox. Also this is my second (one click) install, so a new installation did not fix this issue. Any help would be appreciated.

coperator, 2019-11-24:
I seem to have the same problem. Very annoying.
Unfortunately, I currently do not have the time to investigate this.
Does anyone know if this is a mopidy issue or is this phoniebox specific?

patrickweigelt, 2019-12-25:
Same for me. Sometimes it works, sometimes the sound is down to 0% and nothing works, sometimes sound is ok on 30% and mp3 works but spotify doesn't. In some other thread I read that scanning the music library (via the web app -> Folders and files) helps as is restarts the mopidy service. This indeed works most of the times but it is not ideal since the kids cannot do it themselves and always need me and my phone/computer for it. Does somebody have a fix for this?

Hamsert, 2019-12-27:
Yes scanning the music library is a workaround. What I wonder is, if we have a misconfiguration somehow or if everyone has this problem. I reinstalled Phoniebox once and came to the same result.
The error may happen because my child is very young (18 months) and she sometimes spams the cards on the reader. Could it be programmed that if a card is swiped that old mpd requests get deleted? @MiczFlor

patrickweigelt, 2019-12-27:
In thread #406 others suggest that a wrong system time might be the problem and that a real time clock could solve the problem: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/406#issuecomment-530949259
Is there another way without an rtc to solve this? Would delaying the start of mopidy until the time is synchronized via the internet work? Is everybody without rtc having this issue or is it just some of us?

s-martin, 2020-03-07:
Please check, if #803 fixes this.

Hamsert, 2020-03-10:
Ok, since I did a lot of installs in recent days (upgrading my box to version 2.0 and installing a new box for a friend) I figured something out: I think this whole issue existed because I used to run the one line install script and after that I did the configuration steps (https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/CONFIGURE-stretch). Now I figured, that I don't need to do the configuration steps, if I've chosen the one line install script and this probably led to this strange behavior. Maybe you should clarify on this because at the top of the install page it says "The installation is the first step to get your Phoniebox up and running. Once you have done this, proceed to the configuration." which is not true and will end up in a semi defective box, if you've done the one line install script.
I figured this out on accident since I had one box where I did the configuration steps after the one line install script (develop branch) and this box needed a music library scan after every reboot, otherwise only local files would work and Spotify wouldn't. On the other box I just did the one line install script and after that started some testing and I figured out, that I don't need to do the configuration steps since everything is working already. Also I did not have the "library scan after reboot" problem.
After I figured that out I reinstalled box 1 and the problem was gone.
I hope this helps someone.

s-martin, 2020-03-10:
You are right that the wiki doesn't state explicitly enough that these steps aren't all necessary.

Thanks for testing that.

Issue can be closed then?

Hamsert, 2020-03-10:
As it is right now it looks like it can be closed. Could I reopen it, if I would need to?

s-martin, 2020-03-10:
> As it is right now it looks like it can be closed. Could I reopen it, if I would need to?

Sure, just click reopen.

closed 2020-03-10 by Hamsert

Hamsert, 2020-03-29:
Short feedback after about 20 days of usage: the issue never occurred again.