MiczFlor/RPi-Jukebox-RFID #608: Schedule for shutdown

oneshow01, 2019-10-21:
Is there a way to setup schedule for shutdown? 
I'd like the box to shut down itself in the evening (or at certain times).

At the moment it is connected to a TP-Link Smart Plug which turns the box off and on. This is only power on/off without shutting down. I would like to avoid that.

I think this would be a nice feature... similar to the [Smart Plug Schedule](https://www.tp-link.com/us/support/faq/947/)

[image]

Thank you

dnmtch, 2019-10-21:
you could use this from the command line:

`sudo shutdown -h 03:30`
(or google cron job shutdown)

or use hardware:
https://www.lifehacker.co.uk/2015/11/20/how-to-boot-and-shut-down-your-raspberry-pi-safely-and-more-effectively
http://www.uugear.com/product/witty-pi-3-realtime-clock-and-power-management-for-raspberry-pi/

MiczFlor, 2019-10-23:
Hi @oneshow01 
nice feature request. I will see if I find time to do that. The shutdown options available at the moment are only
* stop player after x minutes
* shutdown after x minutes
* shutdown after x minutes of idle time
@dnmtch pointed you to the right searches. Thanks for that.

andreasbrett, 2019-10-23:
If you already have a smart home solution you might also choose to use the mqtt module and shut down your phoniebox from your smart home server.

https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Smart-Home-remote-control-with-MQTT

UndIch, 2019-12-11:
So is there any progress in this request? Especially the one “shutdown down after idle time” to safe battery? Something like “after 15 min without any song played shutdown?”

andreasbrett, 2019-12-11:
> So is there any progress in this request? Especially the one “shutdown down after idle time” to safe battery? Something like “after 15 min without any song played shutdown?”

That is exactly what the idle shutdown is doing. Set the idle timer to 15 min and you got what you asked for.

UndIch, 2019-12-11:
Stupid me. After having some
Installation problems (see other thread) and using some old versions for a while I didn’t realize this cool and helpful new feature! Sorry for this...

s-martin, 2024-04-19:
Already implemented in V2.

Implementation for V3 will be tracked in #1970 (scheduled for[v3.6](https://github.com/MiczFlor/RPi-Jukebox-RFID/milestone/10))

closed 2024-04-19 by s-martin