MiczFlor/RPi-Jukebox-RFID #609: stretch/Classic 2.0 rc4: resume not working

Mike1020, 2019-10-21:
I've installed 2.0 rc4 from the current one line installer on the tested stretch 9.3. Everything works fine, but somehow I could not get the resume function to work. I've looked at the code and I believe there are some general problems in the scripts that prohibit the resume:  

1. inc.settingsFolderSpecific.sh : At the top, FOLDER is set from the Latest_Folder_Played, which doesn't make sense, as this script is called in playout_controls.sh before the new folder is stored in Latest_Folder_Played. So effectively this resets the new folder to the old one and Latest_Folder_Played is never written correctly. The playlist loaded is still correct, but the wrong folder.dat is written, so the resume will never work. 
Since this script is included this line can simply be removed and FOLDER will be set correctly in playout_controls.sh . 

2. in playout_controls.sh: at first, resume_play.sh -c setpos is called to save postion of the currently played folder. Then resume_play.sh -c resume is called to play the new folder. In both cases, FOLDER is set in resume_play.sh from the Latest_Folder_Played. This is correct for the setpos, but not for the resume, obviously, as the new folder gets only stored in Latest_Folder_Played after this resume script. I've passed the new folder as parameter for the resume script which does the trick: 
in playlistaddplay)
...
# Now load and play
$PATHDATA/resume_play.sh -c=resume -d="${FOLDER}"
...
and in resume_play.sh, any parameters are already read, so only check if it's not passed in which case continue to set it from the Latest_Folder_Played (for setpos, etc). 
if [ -z "$FOLDER" ]
then
    # Get folder name of currently played audio
    FOLDER=$(cat $PATHDATA/../settings/Latest_Folder_Played)
...
3. inc.writeFolderConfig.sh: "&" in filenames will cause the sed command at the end to fail and replace nonsense in the CurrentFilename in folder.dat, so resume will not find the last file in the playlist. Do not use "&" in File- or Foldernames, or figure out how you get the sed to work correctly with "&" :-) 

Not sure if these are proper fixes or I missed something, but it seems to be working for me now, so maybe it helps.

alink75, 2019-11-05:
I have exactly the same situation with the resume function.
Unfortunately i am not that good in programming.
Would you be so kind and explain a little bit deeper where to change the lines you have mentioned in your post.
Especially what to do with the "&" in the inc.writeFolderConfig.

Thank you very much.

ctietze, 2019-11-09:
This issue and  https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/610 share the same root cause.

MiczFlor, 2019-11-11:
I merged the #610 changes to `develop` for testing.

closed 2019-11-11 by MiczFlor