MiczFlor/RPi-Jukebox-RFID #612: ./buster-install-default.sh: Zeile 661: `fi'

Moppel2k, 2019-10-23:
Guten Morgen zusammen

Ich hab mich extra für diese Frage bei GitHup angemeldet daher verzeiht mir bitte wenn mir die allgemeinen Gepflogenheiten hier noch nicht geläufig sind :-)

Diesen Beitrag verfasse ich auf Deutsch da, mir das schneller von der Hand geht und ich momentan schnell Hilfe brauche, bei Bedarf kann ich aber auch auf massige Englischskils zurückgreifen.

Ich habe ein Problem bei Oneline Installation, die bricht ständig ab.

Die Fehlermeldung, die angezeigt wird, ist:

```
./buster-install-default.sh: Zeile 661: Syntaxfehler beim unerwarteten Wort `fi'
./buster-install-default.sh: Zeile 661: `fi'
```

Ich vermute ein Tippfehler in Script, wobei ich das vor ein paar Wochen erfolgreich durchlaufen lassen konnte!?!?

Da ich die Installation auf ein jungfräuliches Debian Bust , die lediglich automatisch Updates installiert hat, ausführe, komme ich nicht auf die mögliche Ursache und finde auch im Netzt keinen Ansatzpunkt! (Bin aber auch eher unerfahren und habe bisher lediglich Images auf Raspberrys installiert.)

Kann mir bei dem Problem jemand helfen?

Danke schon mal für eure Hilfe!

Hier noch der Komplette LOG der Installation:

```
#####################################################
#
# START INSTALLATION
#
# Good news: you completed the input.
# Let the install begin.
#
# Get yourself a cup of something. The install takes
# between 15 minutes to half an hour, depending on
# your Raspberry Pi and Internet connectivity.
#
# You will be prompted later to complete the installation.

Do you want to start the installation? [Y/n] y
OK:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
OK:2 http://archive.raspberrypi.org/debian buster InRelease
Paketlisten werden gelesen... Fertig      
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.      
Statusinformationen werden eingelesen.... Fertig
apt-transport-https ist schon die neueste Version (1.8.2).
at ist schon die neueste Version (3.1.23-1).
gcc ist schon die neueste Version (4:8.3.0-1+rpi2).
git ist schon die neueste Version (1:2.20.1-2).
lighttpd ist schon die neueste Version (1.4.53-4).
mpc ist schon die neueste Version (0.31-1).
mpd ist schon die neueste Version (0.21.5-3).
mpg123 ist schon die neueste Version (1.25.10-2).
php7.3 ist schon die neueste Version (7.3.9-1~deb10u1).
php7.3-cgi ist schon die neueste Version (7.3.9-1~deb10u1).
php7.3-common ist schon die neueste Version (7.3.9-1~deb10u1).
php7.3-fpm ist schon die neueste Version (7.3.9-1~deb10u1).
python-dev ist schon die neueste Version (2.7.16-1).
python-mutagen ist schon die neueste Version (1.40.0-2).
resolvconf ist schon die neueste Version (1.79).
samba ist schon die neueste Version (2:4.9.5+dfsg-5+deb10u1+rpi1).
samba-common-bin ist schon die neueste Version (2:4.9.5+dfsg-5+deb10u1+rpi1).
spi-tools ist schon die neueste Version (0.8.1-1).
ffmpeg ist schon die neueste Version (7:4.1.4-1+rpt1~deb10u1).
python-pip ist schon die neueste Version (18.1-5+rpt1).
python-spidev ist schon die neueste Version (20170223~145721-2).
python3-gpiozero ist schon die neueste Version (1.5.1).
python3-spidev ist schon die neueste Version (20170223~145721-2).
raspberrypi-kernel-headers ist schon die neueste Version (1.20190925+1-1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
W: --force-yes ist veraltet, verwenden Sie stattdessen eine der Optionen, die mit --allow beginnen.
Klone nach 'RPi-Jukebox-RFID' ...
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 5164 (delta 0), reused 1 (delta 0), pack-reused 5158
Empfange Objekte: 100% (5164/5164), 15.40 MiB | 2.67 MiB/s, Fertig.
Löse Unterschiede auf: 100% (3427/3427), Fertig.
Branch 'develop' folgt nun Remote-Branch 'develop' von 'origin'.
Zu neuem Branch 'develop' gewechselt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: evdev==0.7.0 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 7)) (0.7.0)
Requirement already satisfied: spi-py from git+git://github.com/lthiery/SPI-Py.git#egg=spi-py in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 8)) (1.0)
Requirement already satisfied: youtube_dl in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 9)) (2019.10.22)
Requirement already satisfied: pyserial in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 10)) (3.4)
Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 12)) (0.7.0)
Requirement already satisfied: pi-rc522 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 13)) (2.2.1)
Requirement already satisfied: spidev in /usr/lib/python2.7/dist-packages (from pi-rc522->-r requirements.txt (line 13)) (3.3)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: evdev==0.7.0 in /usr/local/lib/python2.7/dist-packages (0.7.0)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: youtube_dl in /usr/local/lib/python2.7/dist-packages (2019.10.22)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: spi-py from git+git://github.com/lthiery/SPI-Py.git#egg=spi-py in /usr/local/lib/python2.7/dist-packages (1.0)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pyserial in /usr/lib/python2.7/dist-packages (3.4)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages (0.7.0)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pi-rc522 in /usr/local/lib/python2.7/dist-packages (2.2.1)
Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages (from pi-rc522) (0.7.0)
Requirement already satisfied: spidev in /usr/lib/python2.7/dist-packages (from pi-rc522) (3.3)
already enabled
Run "service lighttpd force-reload" to enable changes
already enabled
Run "service lighttpd force-reload" to enable changes
### Deleting older versions of service daemons. This might throw errors, ignore them
Failed to disable unit: Unit file idle-watchdog.service does not exist.
Failed to disable unit: Unit file rfid-reader.service does not exist.
Failed to disable unit: Unit file startup-sound.service does not exist.
Failed to disable unit: Unit file gpio-buttons.service does not exist.
rm: das Entfernen von '/etc/systemd/system/rfid-reader.service' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
rm: das Entfernen von '/etc/systemd/system/startup-sound.service' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
rm: das Entfernen von '/etc/systemd/system/gpio-buttons.service' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
rm: das Entfernen von '/etc/systemd/system/idle-watchdog.service' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
### Done with erasing old daemons. Stop ignoring errors!
./buster-install-default.sh: Zeile 661: Syntaxfehler beim unerwarteten Wort `fi'
./buster-install-default.sh: Zeile 661: `fi'
```

Mike1020, 2019-10-23:
Hi, ich hab bisher nur stretch installiert und kann es nicht testen, aber sieht so aus als wenn da ein "then" fehlt, nach Zeile 650. Siehe z.B. Zeile 627. Kannst du einfach einfügen, dann müsste es gehen. "fi" begrenzt nur den "if" Block in bash, das ist kein Fehler.

Moppel2k, 2019-10-23:
Danke für die schnelle Antwort!
... Aber wie mach ich das? Er lädt das script doch online runter oder??

Mike1020, 2019-10-23:
Das script wird heruntergeladen und liegt dann im pi homeverzeichnis. Einfach mit einem Texteditor die Zeile einfügen 
    cd ; nano buster-install-default.sh
und dann nochmal starten
   ./buster-install-default.sh 

Falls du mit dem Editieren Probleme hast, hilft vielleicht dieser Link
https://uberubuntu.info/questions/1785/wie-bearbeitet-man-dateien-einem-terminal-mit-nano
Oder ansonten warten, bis der Fehler behoben wurde, aber ein paar Anpassungen muss man hier und da immer mal machen, da ist ein bisschen Basiswissen auf jeden Fall hilfreich :-)

MiczFlor, 2019-10-23:
Hi @Moppel2k @Mike1020 
I fixed the install script. You can try the one line install process now

closed 2019-10-23 by MiczFlor