MiczFlor/RPi-Jukebox-RFID #617: playout_controls.sh not found

empydempy, 2019-10-26:
Hi all,

this project is great. I am pretty new to linux and I guess this won't be too tough to handle. 
But I haven't found an answer by searching google...or I didn't get it ;-)

Here is what I get when pressing on volU button which I changed to GPIO 5 because I am using hifiberry miniamp. I think the issue I have is just in the beginning where it says playout_controls.sh not found

python3 /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py
**/bin/sh: 1: ./scripts/playout_controls.sh: not found**
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 244, in _call_when_changed
    super(RPiGPIOPin, self)._call_when_changed()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 143, in _call_when_changed
    self.state if state is None else state)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 293, in _call_when_changed
    method(ticks, state)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 197, in _pin_changed
    self._fire_events(ticks, bool(self._state_to_value(state)))
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 368, in _fire_events
    self._fire_activated()
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 397, in _fire_activated
    super(HoldMixin, self)._fire_activated()
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 344, in _fire_activated
    self.when_activated()
  File "/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py", line 26, in def_volU
    check_call("./scripts/playout_controls.sh -c=volumeup", shell=True)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command './scripts/playout_controls.sh -c=volumeup' returned non-zero exit status 127.

Thanks in advance for your help.
empy

MiczFlor, 2019-10-27:
Hi @empydempy,

do other buttons work, which control the behavior of the phoniebox?
Have you altered the script?
It seems that the script is found at a different place:
~~~
File "/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py", line 26, in def_volU
check_call("./scripts/playout_controls.sh -c=volumeup", shell=True)
~~~

empydempy, 2019-10-27:
Hi Miczflor, 

thanks for the quick answer. 
None of the other buttons work.

Acutally, I installed Rasbian (buster) and ran the Jukebox script on another SD-Card to check if it would work then. 
Python and I hope all other tools are installed. 

Then I copied the GPIO-buttons script from the samples to the script folder using this command:
"sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/gpio-buttons.py.sample /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py"

After that I chmod +x the GPIO buttons script (sudo chmod +x /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py)

 and then ran it without alternating anything. (python3 /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py)

I attached the button to GPIO 26 ("next") -> Unfortunately I get the same result :-(
It does recognize the button and starts the script. But then, at least to me, it can't start/find the playout_controls.sh (which is in the directory: /home/pi/RPi-Jukebox-RFID/scripts)

You are saying that the script seems to be found in another place. Here is a listing of my script folder:
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ ls
activate-amplifier.py           playlist_recursive_by_folder.php
daemon_mqtt_client.py           **playout_controls.sh**
daemon_rfid_reader.py           python-phoniebox
**gpio-buttons.py**                 Reader.py
helperscripts                   Reader.py.experimental
idle-watchdog-countdown.sh      Reader.py.kkmoonRFIDreader
idle-watchdog.sh                RegisterDevice.py
inc.readArgsFromCommandLine.sh  resume_play.sh
inc.settingsFolderSpecific.sh   rfid_trigger_play.sh
inc.writeFolderConfig.sh        rotary_encoder_base.py
inc.writeGlobalConfig.sh        rotary-encoder.py
installscripts                  shuffle_play.sh
MFRC522.py                      single_play.sh
pirc522                         userscripts

Should I copy the script or the playout_conrtols somewhere else?
Can it have something to do with buster, instead of stretch?
Do I have to "chmod +x" anything else?

thanks once more :-)
empy

empydempy, 2019-10-27:
Hi all,

it seems that I solved the problem. As soon as I am sure I will post the result.

empydempy, 2019-10-27:
Its working for all buttons tested yet. Next, Vol Up/down....so I guess it will work for the other too.

At the moment I commented shutdown out since I am using an oled.

I will close this Issue and hope it helps others running into this problem. 
@MiczFlor once again thanks for you help! And yet once more thanks for the great project! ;-) 

Here is the modified script which hopefully avoids any trouble with the hifiberry and the oled:

#!/usr/bin/python3
from gpiozero import Button
from signal import pause
from subprocess import check_call

# This script will block any I2S DAC e.g. from Hifiberry, Justboom, ES9023, PCM5102A
# due to the assignment of GPIO 19 and 21 to a buttons

# 2018-10-31
# Added the function on holding volume + - buttons to change the volume in 0.3s interval
#
# 2018-10-15
# this script has the `pull_up=True` for all pins. See the following link for additional info: 
# https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/259#issuecomment-430007446
#
# 2017-12-12
# This script was copied from the following RPi forum post:
# https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-fuer-kinder/?postID=312257#post312257
# I have not yet had the time to test is, so I placed it in the misc folder.
# If anybody has ideas or tests or experience regarding this solution, please create pull requests or contact me.

#def def_shutdown():
 #   check_call("./scripts/playout_controls.sh -c=shutdown", shell=True)

def def_volU():
    check_call("/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=volumeup", shell=True)

def def_volD():
    check_call("/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=volumedown", shell=True)

def def_vol0():
    check_call("/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=mute", shell=True)

def def_next():
    check_call("/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playernext", shell=True)

def def_prev():
    check_call("/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playerprev", shell=True)

def def_halt():
    check_call("/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playerpause", shell=True)

def def_recordstart():
    check_call("/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=recordstart", shell=True)

def def_recordstop():
    check_call("/home/pi/RPi-Jukebox-RFID/playout_controls.sh -c=recordstop", shell=True)

def def_recordplaylatest():
    check_call("/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=recordplaylatest", shell=True)

#shut = Button(3, hold_time=2)
vol0 = Button(13,pull_up=True)
volU = Button(5,pull_up=True,hold_time=0.3,hold_repeat=True)
volD = Button(6,pull_up=True,hold_time=0.3,hold_repeat=True)
next = Button(23,pull_up=True)
prev = Button(22,pull_up=True)
halt = Button(12,pull_up=True)
#reco = Button(6, pull_up=True) # Choose GPIO to fit your hardware
play = Button(27,pull_up=True) # Choose GPIO to fit your hardware

#shut.when_held = def_shutdown
vol0.when_pressed = def_vol0
volU.when_pressed = def_volU
#When the Volume Up button was held for more than 0.3 seconds every 0.3 seconds he will call a ra$
volU.when_held = def_volU
volD.when_pressed = def_volD
#When the Volume Down button was held for more than 0.3 seconds every 0.3 seconds he will lower t$
volD.when_held = def_volD
next.when_pressed = def_next
prev.when_pressed = def_prev
halt.when_pressed = def_halt
#reco.when_pressed = def_recordstart
#reco.when_released = def_recordstop
play.when_pressed = def_recordplaylatest

pause()

closed 2019-10-27 by empydempy