MiczFlor/RPi-Jukebox-RFID #618: Move spotify implementation from mopidy to spotifyd

deisi, 2019-10-27:
Hey everyone.

First of all, thx very much for this project. Its awesome, and we are currently building a setup, having a lot of fun. However I do have a suggestion to make. I most likely will implement it for my setup anyway and maybe you will like it.

Currently the spotify implementation is based on mopidy. As much as I like mopidy, it has quite some flaws. There is e.g. thee need to create a device password. Mopidy is based on python 2.7 up to now https://github.com/mopidy/mopidy/issues/779 and the issue tracking the change to python 3 started in the year 2014, while python 2.7 will be end of life by the end of 2019. It might be they manage to switch by the end of the year, but who knows. Further mopidy is not known to be very fast or lightweight either, resulting in a slower boot up etc. (At least I read people reporting that.)

As an alternative, I would like to suggest [spotifyd](https://github.com/Spotifyd/spotifyd). spotifyd is an extremely lightweight standalone spotify implementation. The ready to use binary has almost zero dependencies and runs out of the box on raspbery pi. Depending on the compiled features it is usually less then 10MB in size and the best part, one wouldn't need the device-password any more, simplifying the setup process. Spotifyd can be controlled via an mpris dbus interfaces. E.g. [playerctl)(https://github.com/altdesktop/playerctl) could be used. Or one could use the dbus interface directly.

Hamsert, 2019-11-03:
Hi Deisi,
since I seem to have a problem with mopidy (https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/607) this might be a way to fix my problem. It would be nice if you could give instructions how to use spotifyd instead of mopidy after you've implemented it on your Phoniebox.
Thanks in advance and cheers
Hamsert

deisi, 2019-11-04:
Hey, I started looking into it and encounter some problems.

- Spotifyd has currently a broken mpris setup making it impossible to use for playing https://github.com/Spotifyd/spotifyd/issues/417
- I then realized that phoniebox is heavily based on MPD. To use Spotifyd together with MPD phoniebox would need some sort of statemachine to keep track of who is playing what. Else there will be conflicts.
- I then took a look if it is possible to have Spotify within MPD. Currently this is not possible but I started a threat in the MPD forum about this. https://forum.musicpd.org/viewtopic.php?f=8&t=7871

MiczFlor, 2019-11-16:
Hi @deisi 
do you think we should close this thread? From the sound of it, mopidy is the best option right now?

deisi, 2019-11-16:
Well I still have it on my list, but we can close it here. Oce I start
moving I might reopen something alike.

djmattyg007, 2019-11-17:
Mopidy is very close to completing the port to python3. It's probably not an issue, as the developer is committed to getting a stable release on python3 before the end of the year.

closed 2020-01-18 by MiczFlor