MiczFlor/RPi-Jukebox-RFID #619: RFID Cards.

MarkIpswich, 2019-10-31:
Not an issue as such...

Looking to get some RFID Cards with the Neuftech USB RFID Reader ID

Just need to know if these will work or if not can someone point me out some that will please as looking for a pack of 50/100/150.

https://www.amazon.co.uk/UHPPOTE-Contactless-Proximity-Control-Attandance/dp/B075F1KQQQ/ref=sr_1_4?keywords=125KHZ+RFID+EM4100+%2F+TK4100+Contactless+Cards&qid=1572549660&s=diy&sr=1-4

MarkIpswich, 2019-11-01:
Also forgot to add, I may get one of those led message scrolling displays that shows what's playing powered via a USB cable.

Any step by  step instructions for getting that set up?

MondMann2019, 2019-11-07:
Yes, these work. Reader and cards both 125khz.

closed 2019-11-08 by MarkIpswich