MiczFlor/RPi-Jukebox-RFID #621: Warten bis WLAN bereit ist (Failed to add some tracks)

MondMann2019, 2019-11-06:
Hallo, 
Ich schreibe das ganze mal auf deutsch. 

Ich konnte das Problem von Spotify via MPD bei mir darauf eingrenzen, dass anscheinend die Verbindung von mpd zu schnell erfolgt und das WLAN (via WLAN-Stick) nich nicht vollständig geladen hat. Wenn ich den MPD Server einmal neustarte (Ordner/Dateien -> Musikbibliothek scannen), dann laufen auch die Spotify Playlisten. 

Ich denke also, dass das WLAN einfach nicht schnell genug ist. Vielleicht reichst es eine Pause von ca. 5-10 Sekunden einzubauen.

punkratz0815, 2019-11-16:
hm?
Ich verstehe dein Problem nicht ganz. Aber, ich glaube in der Raspi.config, gibt es eine option, die besagt "auf netzwerk warten bei boot" - oder ähnlich!
das hat mir zB bei Netzwerkmounts geholfen, dass diese ordentlich eingebunden wurden.

Grüße

_i dont get the whole point, but for example the raspi-config got the option "to wait for network" during boot.
that solved some issues with network mounts at my box..._

MondMann2019, 2019-11-18:
Das scheint das Problem gelöst zu haben. 

Konnte „wait for network at boot“ über diesen Befehl setzen:

sudo raspi-config

Danke und zu. 😁👍🏻

closed 2019-11-18 by MondMann2019