MiczFlor/RPi-Jukebox-RFID #622: Problem with Mute: Still hearing sound ?

SmightyCrown, 2019-11-08:
I installed the standard version on a RPi 3B with the Logitech USB V20 Speakers. I controll the volume with a rotary Knob. Everything is working but when i mute the phoniebox with the rotaryknob button or in the web app , i still hear the sound. It is quiet but audible. The alsamixer is also showing zero when its muted. Can i level the volume elsewhere?

ctietze, 2019-11-22:
Did you check if they are also not fully quiet if you connect the Speaker to a PC, Laptop or whatever?

SmightyCrown, 2019-11-25:
Hey, thank you for replying. 
When i plug it to my pc its muting correct. But i think i found the problem ( but still not the solution): 
When its "muted" in the WebApp, alsamixer is showing "Item: PCM [dB gain: -41.00, -41.00]" with the Logitech USB Speaker. When i change the soundcard to bcm2835 ALSA, and the volume is 0 then it is  showing "Item: PCM [db gain: mute]. With volume 2 its db gain: -102.38. So i think i have to change the db gain value, but i dont know how. Anyone have some suggestions?

closed 2020-01-18 by MiczFlor