MiczFlor/RPi-Jukebox-RFID #625: RFID Reader stopped working after git pull

georgbachmann, 2019-11-11:
Hello,
my phoniebox (with a RC522 reader) was working like a charm for the past year. I recently did a git pull to update for the latest version (I love the new mqtt plugin) and afterwards I followed the instructions here (https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/RFID-Reader-Special) again (basically `cp /home/pi/RPi-Jukebox-RFID/scripts/Reader.py.experimental /home/pi/RPi-Jukebox-RFID/scripts/Reader.py`) and it was working again as expected.

But now, after one more restart of my phoniebox, swiping a card has no effect any more.
Can anybody point me to the logs that I need to take a look at? When looking at `log/debug.log` there is nothing that looks suspicious to me...
Thank for your help guys!

MiczFlor, 2019-11-12:
HI @georgbachmann 
are you using the develop or master branch?

georgbachmann, 2019-11-12:
Hey @MiczFlor ... thanks for your help!. It's the master branch
```
pi@phoniebox:~/RPi-Jukebox-RFID $ git status
On branch master
Your branch is up-to-date with 'origin/master'.
```

MiczFlor, 2019-11-12:
Two early birds meet on github :)

Checking the code base: https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/master/scripts
this script has not been changed in the past 11 months. I assume the error must come from something else. But it's too early for me to thing straight...

georgbachmann, 2019-11-12:
Hehe... good morning to Germany I suppose? :)
Hm.. ok I see :) But the strange thing is that it was working yesterday even after the git pull. But after one more reboot of my raspberry it stopped. Do you have an idea where logs with more information would be located?

georgbachmann, 2019-11-14:
Hm... do you have any idea how  I can debug the problem? Are there any logs that messages get written to that might give me a hint?

georgbachmann, 2019-11-17:
Ok... I did some further inspection today...
So when I run `python daemon_rfid_reader.py`
I get the following.

```
pi@phoniebox:~/RPi-Jukebox-RFID/scripts $ python daemon_rfid_reader.py 
/home/pi/RPi-Jukebox-RFID/scripts/pirc522/rfid.py:78: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(pin_rst, GPIO.OUT)
/home/pi/RPi-Jukebox-RFID/scripts
Card detected.
/home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh: line 73: /home/pi/RPi-Jukebox-RFID/scripts/../shared/latestID.txt: Permission denied
/home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh: line 74: /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_RFID: Permission denied
/home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh: line 250: /home/pi/RPi-Jukebox-RFID/scripts/../shared/latestID.txt: Permission denied
/home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh: line 262: /home/pi/RPi-Jukebox-RFID/scripts/../shared/latestID.txt: Permission denied
/home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh: line 316: /home/pi/RPi-Jukebox-RFID/playlists/Pippi Langstrumpf % Pippi Langstrumpf bekommt feinen Besuch.m3u: Permission denied
/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh: line 409: [: ==: unary operator expected
volume: 50%   repeat: off   random: off   single: off   consume: off
loading: Pippi Langstrumpf % Pippi Langstrumpf bekommt feinen Besuch
/home/pi/RPi-Jukebox-RFID/scripts/inc.settingsFolderSpecific.sh: line 28: [: ==: unary operator expected
volume: 50%   repeat: off   random: off   single: off   consume: off
volume: 50%   repeat: off   random: off   single: off   consume: off
```

So basically a lot of permission denied errors. But playback and changing songs did work...
I now saw that there are quite some files belonging to the root user. I guess I fucked up something when git pulling. After `chown`ing the files to the pi user again, it now works again as expected.

So I guess I can close the issue. Thanks for your help!

georgbachmann, 2020-02-04:
May I ask some more questions about my issue here again... It seems as if the problem isn't 100% fixed yet... I just uploaded two new audio-books and not it's not working again :(

I guess it's still a permission problem I guess...
I am not thaaaat super good at Linux... As which user should the whole phoniebox be started?
Cause f.e. the `gpio-buttons.py` is started as `root`. Other things such as the `daemon_mqtt_client` is run as `pi`.

I'll reset my raspberry and to a fresh install, but if somebody has a better (not that radical) idea, I'd prefer to test that before :)
Reinstalling with airplay and mqtt support again would be quite a bit of work :(

s-martin, 2020-02-19:
Can you try to swipe a card and afterwards post the results of `tail -n 500 /car/log/syslog`?

georgbachmann, 2020-02-22:
Ok... that's quite interesting... It seems that the mqtt part is doing a loooot of things.
I pasted the results here: https://pastebin.com/5Wpc9ZhL
In the logs you can see that I manually pressed the Play/Pause button. But before and afterwards I tried swiping different cards with no effect. Also nothing in the logs? Strange...
Any idea on why that is? Nothing from the card reader?

ZyanKLee, 2020-03-07:
you can activate DEBUG_rfid_trigger_play_sh on the settings page. perhaps that will give us some insight into the issue.
did you check the user permissions for those files in `settings` and `shared`?

georgbachmann, 2020-03-26:
Thanks for the hint! I just enabled some of those debug settings as I am currently experiencing some other problems as well!
Thanks for your help!
Where will those logs appear? So that I can check them in the next few days...

s-martin, 2020-12-18:
Closing this issue now, feel free to reopen, if the problem persists

closed 2020-12-18 by s-martin