MiczFlor/RPi-Jukebox-RFID #627: Stream Phoniebox to Chromecast (work-in-progress)

andreasbrett, 2019-11-12:
I'm currently checking out streaming to Chromecasts using the MPD httpd streaming service. Looks like it's easy to implement but as always with streaming: playback is not realtime but delayed. This is a work-in-progress and might not be a good solution but maybe someone wants to contribute code and/or test this.

Here's what I did to configure it:

## Install [pychromecast](https://github.com/balloob/pychromecast) library
`pip3 install pychromecast`

## Enable MPD httpd service
edit mpd.conf (`sudo nano /etc/mpd.conf`) and add following code block

	audio_output {
		type			"httpd"
		name			"Phoniebox HTTP Stream"
		encoder			"vorbis"
		port			"8000"
		bind_to_address		"0.0.0.0"
		quality			"5.0"		# do not define if bitrate is defined
		#bitrate		"128"		# do not define if quality is defined
		format			"44100:16:1"
		max_clients		"0"		# optional 0=no limit
	}

## Restart MPD
`sudo systemctl restart mpd`

## Use script to play/pause/stop stream on Chromecast
https://github.com/andreasbrett/RPi-Jukebox-RFID/blob/chromecast-dev/scripts/control_chromecast.py

* configure your Chromecast's name in the configuration section of that file
* if you changed the port and audio format in `mpd.conf` you will have to configure this here as well
* play with `python3 control_chromecast.py play`
* pause with `python3 control_chromecast.py pause`
* stop with `python3 control_chromecast.py stop`

MiczFlor, 2019-11-16:
Hi @andreasbrett 
do you think you could create a pull request for this script? and add a page in the wiki :) 
thanks for your contributions!!!
All the best, micz

andreasbrett, 2019-11-16:
I will do so once this is a bit more tested. Currently it's more a proof of concept that I haven't had a chance to test thoroughly (also my phoniebox is currently away for finishing the laqcer).

timoweber1985, 2020-04-16:
Hi @andreasbrett ,
any chance to try this out. The link to the script control_chromecast.py is no longer available. I would really like to test it if possible.

Thanks

Wallnusskaese, 2020-04-26:
Hi, 
any progress on this? 

I would love to do exactly this. 

lg,
Marco

andreasbrett, 2020-04-26:
@timoweber1985 @Wallnusskaese 
Sorry, I had abandonded further developing this as the delay was just too long for my use case. I had accidentally deleted the branch on my fork so the linked script file was not available anymore. It's back up now. Please try again but know that I stopped developing this. You're free to further enhance it if you want to.

Wallnusskaese, 2020-04-26:
@andreasbrett 
Great to hear from you! Is the delay so bad?

andreasbrett, 2020-04-27:
@Wallnusskaese 
It was for my use case and I quickly decided not to bother with it. I guess when reducing buffer size of the http stream of mpd there might be a chance to reduce the delay.

Wallnusskaese, 2020-04-28:
@andreasbrett thanks for keeping up on this. 

Which version of pyChromecast do you use?
I'm on 2.5.2 and figured out that the loooong buffering seems due to a bug in pyChromecast.
As described here: [https://github.com/home-assistant/core/issues/16319](url).

My current workaround (as i cannot update pyChromecast by now) is to issue a play command right after the initial play http-stream command.

Wallnusskaese, 2020-04-28:
To give some background on why I want this to work:

I have made an Phoniebox for my little boy, but sometimes it would be great to just play his audiobooks through our chromecast/google home environment. 
So I soldered an rfid reader onto an esp8266 and connected it over mqtt to my fhem-instance. 
I now plan to run phoniebox on this server ( anyone knows how to run phoniebox within docker??) and cast it to my chromecast.

andreasbrett, 2020-04-28:
> @andreasbrett thanks for keeping up on this.
> 
> Which version of pyChromecast do you use?
> I'm on 2.5.2 and figured out that the loooong buffering seems due to a bug in pyChromecast.

I didn't have an issue with the buffering as that's only an issue when your starting to cast. My issue is more that my actions on the phoniebox become active on the chromecast with a delay. Not an issue with your setup where you'd like to hear audio books and won't switch between them or pause/play often. It's an issue though with the animal sounds that my daughter (1,5y) is triggering with her rfid cards. Every second of delay is too long for the short attention span of our most little ones.

andreasbrett, 2020-04-28:
> To give some background on why I want this to work:
>
> I have made an Phoniebox for my little boy, but sometimes it would be great to just play his audiobooks through our chromecast/google home [...]

Why not use the phoniebox itself for that. Make a script that will toggle the output of mpd from local alsa output to the http stream and start casting it to your chromecast. Assign that script to a rfid tag and you can switch your son's phoniebox from local mode to chromecast mode. I wouldn't go through the hassle to manage 2 phonieboxes for your scenario. Also phoniebox is not what I would call a secure setup. It relies a lot on the standard user having a lot of sudo permissions. Definitely not something I would want to have on my smart home server. Maybe I'm getting your idea the wrong way.

Wallnusskaese, 2020-04-29:
Nope you are right.
> Why not use the phoniebox itself for that.
That was my starting point, but fiddling with my sons phoniebox would cause trouble with him (I tend to break things). I would also love to have a test system where I can test things first, before deploying to the (right-now-working) Box. 
I was hoping to get phoniebox going within docker, I guess that would be ok security-wise.
But unfortunately (for me) thats not quite easy to accomplish.

jo-me, 2021-01-13:
Are you aware of https://github.com/dresdner353/mpd2chromecast ?