MiczFlor/RPi-Jukebox-RFID #628: Push Buttons via GPIO doesn't work with Miniamp

HansMuff82, 2019-11-13:
Hello everyone!
First, I would like to thank you for this great project!

I have some trouble with the push buttons. I know there are several attempts to get them running, I read a lot about it ;) 
My setup is a Raspi 3 B with the Hifiberry Miniamp. I installed a clean Buster lite and the Phoniebox with the "one-line script". Then I changed the configuration like in this post:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/261#issuecomment-433740032, except that I didn't comment the GPIO out. 
I used this pinout scheme:

shut = Button(3, hold_time=2) #PIN5
volU = Button(22,pull_up=True) #PIN15
volD = Button(23,pull_up=True) #PIN16
next = Button(17) #PIN11
prev = Button(27) #PIN13
halt = Button(26) #PIN37

But there is no reaction when I press any button. Music works fine via the Miniamp hub.

Another point is, that I can't activate the GPIOs in the "External Devices & Interfaces"-Section in the WebUI. Every time I click on "Switch On" I stays in the "Disabled"-Mode. 

Does anybody have an idea if there's a problem or a mistake that i don't see?

> ************************************
*** PHONIEBOX INFO
*** version: 2.0 rc4
*** edition: plusSpotify
*** Audio_iFace_Name: Master
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 5
*** Max_Volume_Limit: 75
*** Idle_Time_Before_Shutdown: 0
*** Second_Swipe: RESTART
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: ON
************************************
*** CONF FILES DEFAULT

*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 8.2K Nov 11 19:46 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 2.1K Nov 11 19:46 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 575 Nov 11 19:46 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
ls: cannot access '/etc/php/7.0/fpm/php.ini': No such file or directory
*** /etc/sudoers
-r--r----- 1 root root 784 Nov 11 19:46 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 296 Nov 11 19:46 /etc/systemd/system/phoniebox-rfid-reader.service
-rw-r--r-- 1 root root 340 Nov 11 19:46 /etc/systemd/system/phoniebox-startup-sound.service
-rw-r--r-- 1 root root 289 Nov 11 19:46 /etc/systemd/system/phoniebox-gpio-buttons.service
-rw-r--r-- 1 root root 291 Nov 11 19:46 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Nov 11 20:37 /etc/mpd.conf
music_directory         "/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
#music_directory are changed.
#symbolic links outside of the configured music_directory.
#symbolic links inside of the configured music_directory.
        mixer_control   "Master"                # optional
#mixer_control   "PCM"           # optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 1.8K Jan 24  2019 /etc/dhcpcd.conf
#static ip_address=xxx/24
#static ip_address=xxx/24
#static routers=xxx
#static routers=xxx
option domain_name_servers, domain_name, domain_search, host_name
#static domain_name_servers=xxx 8.8.8.8 fd51:42f8:caae:d92e::1
#static domain_name_servers=>xxx
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw------- 1 root root 178 Nov 11 19:30 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition

*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Nov 11 19:46 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Nov 11 19:46 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
-rw-r----- 1 mopidy root 1.1K Nov 11 19:47 /etc/mopidy/mopidy.conf
username = xxx
*** ~/.config/mopidy/mopidy.conf
-rw-r--r-- 1 root root 3.3K Nov 11 19:47 /home/pi/.config/mopidy/mopidy.conf
#username =
username = xxx

Thanks!

Klaus-Guenther, 2019-11-19:
I have the same issue/Problem with my box.

HansMuff82, 2019-11-19:
So, I think this was a stupid one... 
The problem was, that I commented out the "play button" ("#play = Button(6,pull_up=True)"), but I did not do the same with the "play.when_pressed" function below. So, I think because of that, the script didn't work, and the buttons were disabled.

The following code now works fine:

shut = Button(25, hold_time=2)
#vol0 = Button(13,pull_up=True)
volU = Button(24,pull_up=True,hold_time=0.3,hold_repeat=True)
volD = Button(23,pull_up=True,hold_time=0.3,hold_repeat=True)
next = Button(12,pull_up=True)
prev = Button(26,pull_up=True)
halt = Button(16,pull_up=True)
#reco = Button(6, pull_up=True) # Choose GPIO to fit your hardware
#play = Button(6,pull_up=True) # Choose GPIO to fit your hardware

shut.when_held = def_shutdown
#vol0.when_pressed = def_vol0
volU.when_pressed = def_volU
volU.when_held = def_volU
volD.when_pressed = def_volD
volD.when_held = def_volD
next.when_pressed = def_next
prev.when_pressed = def_prev
halt.when_pressed = def_halt
#reco.when_pressed = def_recordstart
#reco.when_released = def_recordstop
#play.when_pressed = def_recordplaylatest

Best regards Markus

closed 2019-11-19 by HansMuff82