MiczFlor/RPi-Jukebox-RFID #631: Adafruit Speaker Bonnet for Raspberry Pi for sound

MeisterSplinder, 2019-11-14:
Hi, thanks for sharing this nice project!
I tried to use the "Adafruit Speaker Bonnet for Raspberry Pi" for the sound.
I tested the sound card by following the guide from Adafriut (https://learn.adafruit.com/adafruit-speaker-bonnet-for-raspberry-pi/raspberry-pi-usage). And it works. 
After installing the "RPi-Jukebox-RFID" there is no sound. I guess there is an issue with the config files and followed the instructions from "External Non USB Audio DAC ES9023" (https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/External-Non-USB-Audio-DAC-ES9023), but it doesn't work.
Can anybody help me getting this stuff working? I have no more ideas what to do know.

moritzmetz, 2020-01-12:
Does "speaker-test" work?

MeisterSplinder, 2020-01-14:
I installed the speakers without using the RPi-Jukebox project and did the speaker-test: works.
After installing the project the speaker doesn't work anymore. Of course I tried it vice versa but that doesn't work too.
I guess it is a problem with the "mpd.conf", "asound.conf",...and other config settings.

moritzmetz, 2020-01-14:
Have the same problem and some really nasty noise instead of proper sound. It did work again though after deinstalling gpio-zero. Think theres some interference with GPIOs but havnt had time yet to dig deeper.

moritzmetz, 2020-01-14:
Have the same problem and some really nasty noise instead of proper sound. It did work again though after deinstalling gpio-zero. Think theres some interference with GPIOs but havnt had time to dig deeper.

smalchi, 2020-05-05:
I have exctly the same problem. Speaker works until Install Script.

s-martin, 2020-05-26:
Does it work after you executed `sudo apt remove python.gpiozero`?