MiczFlor/RPi-Jukebox-RFID #632: Trouble installing php7

JuliH29, 2019-11-15:
Hi!

I'm in the process of seeting up my new Phoniebox and unfortunately run into problem during the configuration.

The command
`sudo apt-get install apt-transport-https samba samba-common-bin python-dev python-pip gcc raspberrypi-kernel-headers lighttpd php7.0-common php7.0-cgi php7.0 php7.0-fpm at mpd mpc mpg123 git ffmpeg python-mutagen`

returns a set of error messages (that I unfortunately can only post in German, mabe they help you anyway?)
`Hinweis: »php7.0-thrift« wird für regulären Ausdruck »php7.0« gewählt.
Hinweis: »php7.0-remctl« wird für regulären Ausdruck »php7.0« gewählt.
Hinweis: »php7.0-common« wird für regulären Ausdruck »php7.0« gewählt.
Hinweis: »php7.0-curl« wird für regulären Ausdruck »php7.0« gewählt.
Hinweis: »php-remctl« wird an Stelle von »php7.0-remctl« gewählt.
Paket php7.0-common ist nicht verfügbar, wird aber von einem anderen Paket
referenziert. Das kann heißen, dass das Paket fehlt, dass es abgelöst
wurde oder nur aus einer anderen Quelle verfügbar ist.

E: Für Paket »php7.0-common« existiert kein Installationskandidat.
E: Paket php7.0-cgi kann nicht gefunden werden.
E: Mittels des Musters »php7.0-cgi« konnte kein Paket gefunden werden.
E: Mittels regulärem Ausdruck »php7.0-cgi« konnte kein Paket gefunden werden.
E: Paket php7.0-fpm kann nicht gefunden werden.
E: Mittels des Musters »php7.0-fpm« konnte kein Paket gefunden werden.
E: Mittels regulärem Ausdruck »php7.0-fpm« konnte kein Paket gefunden werden.
`

Any idea how I could fix this?
Your help would be very much appreciated!
Thanks,
Julius

closed 2019-11-15 by JuliH29

ctietze, 2019-11-15:
Hi Julius,
what Raspian release did you install? Stretch or Buster? Seems to me that you tried the stretch ap-get command from the wiki page on a Buster release.

The packages in Buster are starting with _php7.3-*_

Best regards
Clemens

JuliH29, 2019-11-15:
Hi Clemens,
Thanks for your help. It's quite likely that I erroneously did what you descriped. I reinstalled Phoniebox and everything worked smoothly!

Julius