MiczFlor/RPi-Jukebox-RFID #636: Different audio outputs and other questions

JuliH29, 2019-11-22:
Hi!

My phoniebox-project is moving along nicely and I would like to thank everybody here who made this possible!
I have several questions:

1. Is it possible to to include several ways to output audio (e.g. internal speakers & headphone jack) and switch between them easily?

2. Is it possible to switch to a different Spotify account after the installation?

3. Is there a way to connect to the Raspi when I'm away from known wifis without physically connecting to the Raspi (=opening the box and attaching cables)? So some sort of adhoc-wifi maybe?

Again thanks for all of your work!
Julius

MiczFlor, 2019-11-24:
Hi @JuliH29 
thanks for your interest, here some answers:

* 1a. Is it possible to include several ways to output audio (e.g. internal speakers & headphone jack) 

In theory yes, but ...

* 1b. and switch between them easily?
Not at the moment. Audio out is handled by the mpd (music player daemon) and this needs to have the audio device in its config file. Haven't thought about how to change this in the web app and without reloading the service.
If you use the audio out (not a sound card) with the [Ground Loop Isolator / Entstörfilter Audio](https://amzn.to/2Kseo0L), you could use a [mechanical switch like this one](https://amzn.to/35oOSCS)?

* Is it possible to switch to a different Spotify account after the installation?

Again, in theory yes, but you would manually need to alter the config files. See these lines in the install script: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/c7faadf07b3ff82428c4e7546b8d277705ec29b9/scripts/installscripts/buster-install-default.sh#L639

* Is there a way to connect to the Raspi when I'm away from known wifis without physically connecting to the Raspi (=opening the box and attaching cables)? So some sort of adhoc-wifi maybe?

Currently, there is no way to connect directly to your Phoniebox without a router in the middle. In terms of "use case" this does not seem to be a big problem (i.e. you might be on the road or on your holidays and the need to add files or the like might not be too big).

You can add more than one wifi to your Phoniebox. See here for a little info on the web app UI of doint htat: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/401#issuecomment-520845166

Yordan1976, 2019-11-26:
Hi @JuliH29 

I used one of these to interrupt the signal to the speakers when plugging in headphones. Works pretty well.

https://de.aliexpress.com/item/32847629394.html

JuliH29, 2019-12-01:
Thank you all for your help and comments!

There is one open question for me:
During the installation process of phoniebox I assigned a static IP to the Raspberry Pi (in my case 192.168.1.28).
Now, if I'm on a different wifi I can't connect to it (even if the login credentials are added to the phoniebox settings) if the structure of the IPs is different (for example 192.168.188.x). 
I understand why this doesn't work, yet it creates a practical problem.

How do I handle this?

Thank you again for your help!

e-maddin, 2020-02-25:
> 3. Is there a way to connect to the Raspi when I'm away from known wifis without physically connecting to the Raspi (=opening the box and attaching cables)? So some sort of adhoc-wifi maybe?

==> you could use one of those mobile WIFI routers - have that configured as a Wifi in the box. then you can connect to the player also when you're travelling.

s-martin, 2020-03-07:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/WLAN-ACCESS-POINT.md

I’ve found this description, which may help to setup the Pi working as access point.

closed 2020-12-15 by s-martin