MiczFlor/RPi-Jukebox-RFID #639: Pictures of Phonieboxes for inspiration 2019/2020

MiczFlor, 2019-11-24:
Hi Phonieboxers, 

please add your designs of finished Phonieboxes in this thread. Also, I am collecting designs for the new calendar 2020 coming out shortly before Christmas. You can see [the 2019 calendar here](https://github.com/MiczFlor/RPi-Jukebox-RFID#the-2019-phoniebox-calendar-is-out).

Use this new thread that I started for sharing pictures and designs, because the old thread was wonderfully picture rich which is why it became very heavy to load. So we start a fresh thread end of 2019 and into 2020. You can find the [old thread of Phoniebox designs for inspiration here](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/5).

Yordan1976, 2019-11-28:
Will you take pictues from 2019 from the old thread for the 2020 calendar? Or should we post a single picture here again for it?

MiczFlor, 2019-11-28:
> Will you take pictues from 2019 from the old thread for the 2020 calendar? Or should we post a single picture here again for it?

I will go through the old thread, as well, of course. In short: I will dig through both.

andreasbrett, 2019-11-29:
So here's what I built for my little one. She's shy of 16 months so this box will be mostly used for animal sounds for now. Love how versatile Phonieboxes are as they will easily grow with your children and their needs. 

[image]

jens7885, 2019-11-30:
@andreasbrett: very impressive!

Here is my more modest design:
[image]

lj2000, 2019-12-01:
This is the second Phoniebox I built up.
The first one is my daughter's Phoniebox.. This one is mine :)
Some weeks ago I started to regenerate my good old tube radio Telefunken Jubilate 8.
It was still working but the sound wasn't as good as one would expect from a tube radio or a tube amp.
So I did all the standard things which are needed for bringing tube radios back to service:

- Replacement of all defective capacitors and resistors
- Intensive Cleaning
- Change of the magic eye and of the amplifier tube

After this repair the radio sound (FM/UKW) was fantastic.
While testing the radio I thought if it would be possible to use the radio's TA input for today's audio signals (MP3, CD or ... Phoniebox :) )

So I searched this topic in the internet and found some hints and solutions how to connect audio line outputs to a tube radio's TA input.

Finally I decided to upcycle my tube radio by adding the typical Phoniebox parts (Raspberry Pi Zero, Power Supply, RFID reader) and an audio cable including a very simple but very good passive audio filter ( see [this link](http://www.michael-twiste.de/roehren/klang.html) for details ).

Now my good old Jubilate 8 tube radio has some additional features which in the 1950's no one had dared to dream: :)

- Internet radio
- Phoniebox player using the RFID cards 
- Further software extensions like Bluetooth loudspeaker :)

And of course all sound features can be played in crystal clear tube amp quality. But this radio only provides one audio channel (mono). In my opinion this doesn't matter. I mixed the left and the right channel in the audio cable using resistors. 

Open topics:

- I am searching for the right knob or better: I am searching for the metal in the middle of this knob. If you have one in stock please contact me. 
- When regenerating this radio I destroyed the cable of the MW antenna. I also want to reactivate the MW/LW antenna . MW and LW is also nowadays nice to hear.
- The USB connection to the RFID reader disturbs the FM/UKW reception. I solved this by using a very good shielded USB cable. When the MW antenna is back I also have to check if this one works without any disturbances
- I want to perform some tests with alsaeq (Equalizer): If this works it could be that I don't need the filter in the audio cable which is mounted on the backside of the radio at the moment.

**Attention: If you want to modify a tube radio by yourself you must be an electronics specialist! Tube radios can be very dangerous. Inside of them there are very high voltages which can kill you. So don't do anything with them if you are not exactly knowing what you are doing there.** 

[image]
[image]
[image]
[image]
[image]
[image]

flo2305, 2019-12-01:
This is my first one from 2018. 
The second one is in the making. 

Classic Design :-)

[image]

MiczFlor, 2019-12-01:
@flo2305 that *is* a Phoniebox classic by now, you are right. Nice 50s green. Looks like you got the dial for volume and buttons to skip to new tracks. If you have a working script combining buttons and dial, I would like to add this to the wiki, possibly you can write something?

flo2305, 2019-12-02:
@MiczFlor Hi. Sorry, but I didn’t realize the volume control in software. It’s the built-in dial from the speakers. I soldered it around and built it into my box.

040medien, 2019-12-02:
My first self-designed speaker. Made with a MPCNC lowrider from 18mm plywood. Turned out to be too loud for a kid's room, so I had to cap the volume in software. The slot is for RFID cards which get a label for each audio book.

[image]

andreasbrett, 2019-12-02:
@040medien that looks great! Also like the card slot. Very nice design!

Jorgen-VikingGod, 2019-12-02:
I'm looking currently for something like this 
[image];
use the original keys and knobs and be able to put into some old tapes (assigned with a little rfid sticker) to play a playlist or stream.
First I have to search a good one at the basement of my parents oO

Hans-Haefner, 2019-12-07:
[image]
[image]

mikebarkmin, 2019-12-09:
@andreasbrett your box looks very nice. What materials did you use?

andreasbrett, 2019-12-09:
> @andreasbrett your box looks very nice. What materials did you use?

Thanks, @mikebarkmin! Here's a list of materials used:

- box from [Obi](https://www.obi.de/aufbewahrungsboxen/kesper-allzweckkiste-kiefer-mit-deckel-8-4-l/p/2715670)
- simple rope from Obi as well
- white lacquer from Hornbach (2k PU acrylic lacquer; 2 coats)
- colored acrylic lacquer... idk, my mother painted the owl and used her own colors
- buttons from AliExpress (very cheap ones, hope they last)

If you need more details just ask.

UndIch, 2019-12-11:
[image]
[image]
[image]
Not completely finished jet - but just a first impression!

dbffm, 2019-12-14:
Final assembly finished last night:
[image]
[image]
[image]

deheingit, 2019-12-16:
> 
> 
> Final assembly finished last night:
> [image]
> [image]
> [image]

Hi dbfmm: Do you have a part list? (especially the box itself. And how did you get the battery leds to show on the back?

Yordan1976, 2019-12-16:
> And how did you get the battery leds to show on the back?
I used fiber optic cables for this.
[image]
[image]

deheingit, 2019-12-16:
a cool, thanks. Are they just glued onto the batteries leds?

andreasbrett, 2019-12-16:
@Yordan1976 Nice! How did you fix them on the battery and where did you get them (haven't been able to find them yet).

Yordan1976, 2019-12-16:
The powerbank has holes there (the LEDs are inside) and I just [bought](https://www.ebay.de/itm/Lichtfaser-Lichtleiter-PMMA-Licht-wo-immer-Sie-wollen-flexible-0-5mm/322419402059?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649) a fitting diameter (0.5 mm here). Because the diameter isn't perfectly constant, i could measure it and choose the right places for the cut.

dbffm, 2019-12-16:
> Hi dbfmm: Do you have a part list? (especially the box itself. And how did you get the battery leds to show on the back?

@deheingit: the box itself is custom tailored medium-density fiberboard (MDF) which is glued (Leim). Front and back are 8mm; the other parts are 12mm. 

I bought optic fiber cable (0,5mm diameter; 2m length) on ebay. On the wood side it is glued; on the powerbank it is just "plugged". It could be a bit tighter but it holds the cable as long as you do not shake the whole box.

Diekoeche, 2019-12-16:
[image]
[image]
We finally Finishing the Phoniebox :)
It’s in a Recycled Box where we bought glasses in. So it’s flexible and easy to open with the zipper. That’s why we didn’t built a charging connector.

I hope it’s inspiringly for someone :)

deheingit, 2019-12-17:
> @deheingit: the box itself is custom tailored medium-density fiberboard (MDF) which is glued (Leim). Front and back are 8mm; the other parts are 12mm.
>
> I bought optic fiber cable (0,5mm diameter; 2m [...]

@dbffm: Ok, thanks a lot

deheingit, 2019-12-18:
@dbffm: One mor thing: where did you get the usb port and the power button. i rally like the combination ;)

dbffm, 2019-12-18:
@deheingit I found both on Aliexpress
€ 7,75 | 22mm mounting diameter Micro USB Female to Male
https://a.aliexpress.com/FcQZ4hkUL

16mm 3V LED Power Push Button https://a.aliexpress.com/oiAuW0sBZ

deheingit, 2019-12-18:
@dbffm : Thanks!!

pfedan, 2019-12-20:
During the last days I also built a Phoniebox for my 3.5 yo daughter. Here you go:

[image]
[image]

UndIch, 2019-12-21:
I already posted some pictures of our „Phoniebox“ - but I wasn’t satisfied with t
Is so I did some sort of makeover and that’s how we like it ... and hopefully my son does, too: 
[image]
[image]
[image]

Parts used: RPi 3b, OnOffShim (top!, 7€ +shipping), Neufatek Reader (soldered for “stop on removal” hardware hack), Logitech S150 (15€, absolut recommended- playing ok sound completely over USB (external soundcard included In the speaker and powered by RPi USB, so turning off via onOffShim!), 10000mAh Licheers Powerbank (BlackFriday Amazon 10€, needed a small one because of the reduced space in the „car“), Push LED Buttons from Conrad (2€ ea), red, yellow led for tail and top lights, some resistors. 

So over all 40€+15+10€+10+10€ spare parts = 85€.

JuliH29, 2019-12-21:
Here is my version of phoniebox, which I just finished. So you so much @MiczFlor and the whole community here.
Happy holidays to all of you!
[image]
[image]
[image]
[image]

isenmann, 2019-12-21:
Also finished my first Phoniebox. The RFID reader is at the front behind the face, so you have to "feed" the tiger with the cards. The buttons for play control (play/pause, next, previous) on top and on the back there is the USB charging port and the on/off button. Volume control on the side between the speakers.

[image]
[image]

For the picture I have removed the baseplate, normally it's closed.
[image]

domu83, 2019-12-23:
bit aged already... here is my classic design box
in the meantime inside upgraded with GPIO extension and RTC module.
Featured w/ on/off shim, akkupack w/ fibre led status, magnetic powercord, matrix display & rotary encoder

[image]
[image]
[image]
[image]
[image]

deheingit, 2019-12-23:
@domu83: Very nice and clean design. Can I ask you a few things?:

1. What kind if magnetic plug did u use. I did some research but didn't find anything i really liked. Also the silver plug on the ouside looks great.

2. What is the RTC Cloock for? I guess the rPi doesn't have a battery, so the time is not stored once shut down. But why do you need an accurate time?

3. You sid u use an OnOffShim, bu on the top I see a "mechanical" switch. How does this work. My understand was that the OnOff switch needs a push button to get a signal and than does the power-off on its own

4. Just out of interest: What did you install in the lower left corner next to the speaker?

5. What kind of display did you use? And how does it work with the phoniebox software?

Thanks a lot
 Dennis

pfedan, 2019-12-23:
@domu83 I bet you will massively improve the sound, if you isolated the back of the speakers from the grid (using wood and hot glue). You basically created an acoustic short cirquit which takes a large portion of the lower frequencies.

domu83, 2019-12-23:
@deheingit Thanks... design could be improved 😉

1. I used this Magic Plug: https://www.amazon.de/dp/B07MTGXDR3?ref=ppx_pop_mob_ap_share and this USB plug: https://www.amazon.de/dp/B01ATBTWVM/ref=cm_sw_r_cp_api_i_.1oaEb9V9N19J
you need to consider the max. load capacity with this concept, because depending on the akku pack used it takes quite long till full charge state.

2. the RTC has a battery attached, so storing the time is not an issue. Actually the RTC is considered to solve the issue with long log in time for the Spotify version (issue #406 ). Unfortunately it does not solve this on my end, still investigating.

3. the mechanical switch/toggle switch is connected to the on/off shim to act as the remote button. I liked the design of the switch as the power button. Not sure if the toggle switch itself, without the shim would do this job as well (considering the powerbank + the power-on after a total power-down, w/ no power at the rpi).

4. not sure how to call that, but thats the logic board/equalizer/control module of the speakers (former housed logitech speakers). missed to mention, that I also use a noise filter (https://www.amazon.de/dp/B01LX0H29W/ref=cm_sw_r_cp_api_i_O.oaEb9T13N49)

5. I used this I2C 44780 2x 16 LCD display (https://www.amazon.de/dp/B07CQG6CMT/ref=cm_sw_r_cp_apip_1I1YQNIY5SKkA), today I would consider to use a OLED display instead to display covers as well. Might be more helpful for kids which can't read so far. Implementation is done thanks to #498 contribution.

@pfedan thanks for the hint. Actually I domnot have any issue with sounds. It's actually pretty good. I had a USB soundcard preciously, but my feeling was, that the rpi onboard was even better. I also use a noise filter, maybe that's why I do not face any issue.

Hans-Haefner, 2019-12-23:
[image]
[image]

Hans-Haefner, 2019-12-23:
Wonderful ideas that were implemented up here.
Some Pictures of my Phoniebox, it will be a Christmas present for tomorrow.

deheingit, 2019-12-24:
> @deheingit Thanks... design could be improved 😉
>
>     1. I used this Magic Plug: https://www.amazon.de/dp/B07MTGXDR3?ref=ppx_pop_mob_ap_share and this USB plug: [...]

@domu83: Thanks a lot for you answers. One follow up question: The USB plug has a Micro female port on the inside, correct? Where did you get a micro-usb to micro usb kabel? 

Have a great christmas time!

deheingit, 2019-12-24:
Ah sorry, my mistake. I guess you just need some kind of extension like: https://www.amazon.de/SunshineTronic-Verlängerungskabel-beschaltet-doppelt-geschirmt/dp/B001F6Y8JC

domu83, 2019-12-24:
@deheingit right... I used this one: DeLock Kabel USB micro-B... https://www.amazon.de/dp/B00B3SBTRQ?ref=ppx_pop_mob_ap_share

camCut, 2019-12-24:
This is my design, my son got it this morning as a present is just as happy as I am. [image]

Reddi82, 2019-12-25:
This is our handpainted box: 😊

[image]
[image]
[image]

Florian1974, 2019-12-28:
Here are a few pictures of the first (more to follow :-) ) phoniebox that I build for my 2 1/2 year old Godchild as  a  Christmas gift, needless to say that she loves it. 
The card reader is in the top side of the cube 

[image]
[image]
[image]

for the box I used 18mm pine bords that are box-jointed and glued together, the edges are routed round, the bottom is screwed on for maintenance. 

Here is a picture of my "development system", which I used for tinkering, learning and evaluating more or less useful ideas.  This is "my phoniebox" that I use as a player on my desk at the moment. I will definetly build a proper one with some different features for myself. 
[image]

Here is a rough list of componets that I used:
- RaspberryPi Zero W
- 2 x MAX98357 I2S class D amplifier
- 2 x Visaton FR7 speaker
- USB card reader 
- 16x2 dot matrix display with I2C adapter
- homemade on/off board inspired by the on/off shim
- Intenso powerbank
- arcade buttons for prev/play/next
- rotary encoder for volume / mute 
- iluminated stainless steel button for on/off 

 Other SW components that I used or modified

- [lcdproc](http://lcdproc.omnipotent.net/) for controlling the display
- [mpdlcd](https://github.com/rbarrois/mpdlcd) as a interface between mpd and lcdproc
- some own modification for mpdlcd to get more screens (default is only one)
- some own stuff to display IP, SSID and time via lcdproc
- some modifications to rotary and gpio scripts to fit my needs (combined fast forward/next and rewind/prev,  integration of my on/off board)-> should be combined, cleaned, documented and made available
- some changes to playout control to get resume working via the play button after power-on 

Things that I plan to do in the future:
- clean and document my changes and then make them availabe for the community
- try to intergrate [lirc](http://www.lirc.org/) just for the fun of it
- whatever else comes to my mind or is wished by my "first client"

@MiczFlor + community,  big thanks for the great project. I try to become more active here as I really like the project

andrebsch, 2020-01-02:
[image]
[image]
[image]
[image]
[image]
[image]

I designed a little adapter PCB to use the Hifiberry Miniamp and the OnOffShim together, it also holds the pull up resistors for the buttons and additional connectors for the display, 3.3 V and 5V  . It fits on all raspi models with the 40 pin headers.

dbffm, 2020-01-06:
> I designed a little adapter PCB to use the Hifiberry Miniamp and the OnOffShim together, it also holds the pull up resistors for the buttons and additional connectors for the display, 3.3 V and 5V . It fits on all raspi models with the 40 pin headers.

@andrebsch I like your idea for designing a PCB for that. Would you mind sharing the PCB design file? Did you use KiCAD or sth. similar?

andrebsch, 2020-01-06:
The Gerber flies for the Adapter PCB can be downloaded here:
https://github.com/andrebsch/Pi-Hat-for-Phoniebox

gsowins, 2020-01-09:
Hello everyone ! This is my first Phoniebox. It will be used to listen to audiobooks from HP Lovecraft with RFID stickers on HP Lovecraft books. 
[image]

dermarzel, 2020-01-14:
@andrebsch I really like your case. Is it a custom made one or could you please list your used components? I appreciate your answer. Thank you!

gsowins, 2020-01-14:
Hi !
Thanks for your comment ! The case is custom made from a lamp I found here
<https://www.leroymerlin.fr/v3/p/produits/lampe-charme-romantique-bois-naturel-mathias-buchen-e1500944306?megaboost&gclid=CjwKCAiA6vXwBRBKEiwAYE7iSzcZN5tlJuctKueuAsjHgqNRAfsyxjHUibJBsT6-0rop3wxJ4zlhThoCJzgQAvD_BwE&gclsrc=aw.ds>
for
a few euros. I took away electrical components and I managed to put inside
a USB RFID reader and a JBL speaker like this one
<https://boutique.orange.fr/accessoires/jbl-go-mini-enceinte-noir> (sound
is amazing ! ). I bought an old vintage radio speaker on eBay for 40 euros
(that was a really nice deal since this kind of item is usually sold at
least 100 euros). The vintage speaker holds to the case thanks to a
plumbing pipe.
I can send you some pictures if you want :)

Thanks again !
Guillaume

fatalerror1981, 2020-01-21:
...and this is mine. Its a remix of [primers](https://www.thingiverse.com/thing:3789675) design on thingiverse but modified to fit 2 speakers and different buttons arrangement.
[image]
[image]
[image]
[image]

Cheers.

deheingit, 2020-01-23:
@fatalerror1981: Thanks for sharing - i really like the design. Maybe I'll try this using my soon to arrive 3D printer ;)

3 question: 
- I guess you printed the dinosaur as well. I was just thinking about how to create cool "phoenies" - do you know of a library with models to use
- you painted it manually after printing, right? What kind of color would you use?
- who did you build in the rfid ship?

THANK YOU!

andrebsch, 2020-01-23:
@dermarzel 
My Case is custom made. Its made from 18 mm plywood and the miggle layer is a 27 mm beech wood. The back panel is made with a 6 mm plywood and for the Front I used a 12 mm plywood.
I made a pattern and then I used a router with a bottom trimming cutter to make the other layers, After this all layers where glued together and sanded until the surface was smooth.

The Housing for the Display is made with a 3d Printer as well as the Knob for the Volume control.

For the carrying strap I drilled a slotted hol on each side and fixed the belt from the inside with 2 scews to the housing.

AS speakers I used Visaton FR7 but FR 10 will fit as well.

At the moment I'm working on my second box and the housing is made in the same style but with a cnc machine, so sanding at the end will not take so long :-)

mduesi, 2020-01-24:
@fatalerror1981: Very nice case! Do you have your own 3D printer or did you use an online printing service?

Diablo-Rojo, 2020-01-28:
Now it is my turn to send a big, big "thank you" to @MiczFlor !!!

**Some background:** As a father of three boys I got in touch with the original Toniebox in Nov 19. Although the overall idea is nice, the total costs for three kids would add up soon (not counting the plastic waste bye each new tonie). So I was looking for an alternative solution and came here. As I have literally no technical background I had to spend some evenings reading and watching YT-videos in order to find my personal setup.

**My project:** I came to the conclusion that I wanted to keep it as simple as possible. So, no buttons, no powerbank, everything managed bye RFID. Even the cases should follow the idea of upcycling in order to avoid any unnecessary waste. So I choose the following parts:

- RaspPi 3b
- Neuftech RFID Reader EM4100
- SanDisk Ultra microSDXC 128 GB
- Trust Leto 2.0 USB Lautsprecher
- Aukru 5V 3A Micro USB Netzteil mit Schalter (incl transparent case & coolers)

During the past weeks I build four boxes in total as each kid got his own and we wanted an additional one in our living room. And here are the results:

Box No 1 - living room:
[image]

Box No 2 - kid 8yo:
[image]

Box No 3 - kid 6yo:
[image]

Box No 4 - kid 3yo: 
[image]

As the parts are the same in each box the interior is also the same in each box. Just as an example the interior of box no 3: 
[image]

All boxes shut off after 30 min of inactivity; therefore all boxes have power adapters that include a switch so that they can start the boxes on their own. As the boxes are ment to be stationary there were no power banks used. Everything works very smooth & stable (apart from issue #747 ). All RFID cards have been painted bye the kids on their own. 

I couldn´t be any happier. So, once again: Thank you @MiczFlor for this beautiful project (hearts)

splitti, 2020-01-28:
@Diablo-Rojo really nice and good examples for simple and clean Boxes! Thanks for sharing!!!

stefantruehl, 2020-02-01:
[image]
[image]
[image]
[image]
[image]

Thank you very much @MiczFlor . Me an my daughter we strongly appreciate the project!!!

GMZ87, 2020-02-03:
Nicht immer einfach, aber das Ergebnis ist um so besser geworden.

Mein Sohn hat am 10.02. Geburtstag und soll die Box bekommen. Er hat sie auch schon des öfteren gesehen und auch den Stoff selbst ausgesucht. Nur ist er bis dato im glauben das die Box für einen anderen Jungen ist ;) Kinder sind was tolles xD

Ich habe die Version ebenfalls mit einer Powerbank gebaut. Diese aber komplett zerlegt und auch die Akkuanzeige in das Gehäuse integriert. Dies fand ich durchaus wichtig, um auch einem Kind deutlich machen zu können wann der Akku alle ist.

Das System läuft auf einem Raspi 2 mit Hifiberry MiniAmp und On/Off Switch wie in der Beschreibung. Nach anfänglichen Installationsproblemen läuft nun alles einwandfrei. 

Was ich etwas anders gemacht habe, waren die Tasten. Normale Arcadetasten finde ich derart hässlich und sie würden auch nicht zur Optik der Box passen. Deswegen Touch Sensoren unter einer Gravierten 4mm Acrylglasscheibe für die Bedienung und nur für das ein und Ausschalten einen Taster auf der Rückseite.

Das Gehäuse selbst besteht aus 4mm Nussholzfunier. Dies wurde im Laser geschnitten. War für uns die einfachste Lösung, da wir son teil zu Haus haben für Frauchens Firma.

Alles in Allem ein sehr sehr schönes Projekt!

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ZyanKLee, 2020-02-05:
Wow, schick. Die herausgereichte Akkuanzeige ist tatsächlich eine gute Idee. "Ist der Akku leer" ist eine häufige Sorge bei unseren Boxen - meistens unbegründet.

nikolaus-online, 2020-02-23:
[image]

[image]

[image]

[image]

Let me begin by telling you, that we adore this project and really love our Phoniebox. I assembled it in December, it was a gift for my two boys. They use the box constantly since then.

Parts used:

- Raspberry Pi 3B
- RAVPower Powerbank 22000mAh
- Larcele Buttons
- Trust Leto 2.0 USB Speakers
- KKmoon IC Cardreader 14443A

The case is just a plastic-container from our local supermarket meant for food. My wife sewed a housing out of felt. Its closed with two big strips of fastener. Its fugly, I know that. But it works like charm. :)

A thing I like very much is the implementation of MQTT. I added the box to my [openHAB-Setup](https://www.openhab.org/) and can now monitor the usage and change volume etc. straight out of my Smarthome-Menue. I totally recommend looking into that, you can do amazing stuff with it. Two examples I realized with Phoniebox and MQTT:

- When I drive home after work, the box announces "Dad is on his way home", when the box is switched on.
- My boys have one card they use, when they want to play "circus". It starts circus-themed music on the box, but also crowd-cheering on the smart speaker in their room and it changes the light to a colorful theme. (Hue Lights)

**One thing I would love to have is a lock-function (Reader and GPIOs) via MQTT. That would be nice.**

[image]

I also cast the tracks that are played to my [Awtrix-LED-Controller](https://blueforcer.de/awtrix-2-0/).
[image]

I designed all the cards with a simple Photoshop-Template I created. So my kids can see whether its a audiobook or music etc.

[image]

andreasbrett, 2020-02-23:
> One thing I would love to have is a lock-function (Reader and GPIOs) via MQTT. That would be nice.

You mean lock the buttons and reader? That's already possible. You can start and stop the rfid and gpio services. Send "start" or "stop" to the following mqtt topics.

- phoniebox/cmd/rfid
- phoniebox/cmd/gpio

Love your circus lights solution as well as the daddy's coming home notification. Cool ideas!

[58 more replies]

closed 2021-03-05 by ZyanKLee