MiczFlor/RPi-Jukebox-RFID #640: Spotify

swissmaster1, 2019-11-24:
My Musicbox is running well with your software on it. But since a few weeks there is a problem with spotify. I’m not able to play Spotify tracks when I start the box. I found out that everytime when I restart the mopidy server over the Iris websurface Spotify is working again and I can play tracks and playlists over Spotify.

MiczFlor, 2019-11-24:
I can't find the thread, but somewhere another Phonieboxer with Spotify (I don't use that option... yet) said that Spotify sometimes takes long for the authentication process and then it interrupts. Or something like that. Possibly somebody with Phoniebox +Spotify experience can jump in to this discussion.

domu83, 2019-11-24:
it's issue #406 https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/406#issuecomment-530949259

swissmaster1, 2019-11-24:
Thanks for your answers. I‘ll check if it has something to do with date, time and rtc. Would be strange for me if that is the problem.

jschm06, 2019-11-27:
I have the same thing here going on but yet no idea how to solve it. Anyone fit into Spotify error codes?

@swissmaster1 
I guess you find something similar to: 

>2019-11-27 17:06:36,930 ERROR [462:SpotifyEventLoop] spotify.session: Spotify login error: <ErrorType.OTHER_PERMANENT: 10 >

in your /var/log/mopidy/mopidy.log after starting your box.

After restart your Mopidy server over the IRIS webinterface it should look like:

> 2019-11-27 17:12:19,091 INFO [1588:SpotifyEventLoop] spotify.session: Spotify logged in
2019-11-27 17:12:19,093 INFO [1588:SpotifyEventLoop] mopidy_spotify.backend: Logged in to Spotify in online mode
2019-11-27 17:12:19,546 INFO [1588:SpotifyBackend-3] mopidy_spotify.web: Logged into Spotify Web API as XXX (Spotify user name)
2019-11-27 17:12:19,548 INFO [1588:SpotifyBackend-3] mopidy_spotify.backend: Started spotify backend

Anyone got an Idea to solve the startup problem?

swissmaster1, 2019-12-14:
I was maybe to solve this problem.
It looks that when the musicbox started the network wasn‘t ready. Therefore it was not able to login properly in Spotify. I now choosed the option in raspi-config wait for network before boot. A first test was successful. Hopefully it works now.

restless1987, 2019-12-21:
I just fixed my local playback issues by installing the correct codes to playback mp3:

`sudo apt-get install gstreamer1.0-plugins-bad -y` 

I found the solution by monitoring mopidys logs with `sudo tail -f /var/log/mopidy/mopidy.log` while trying to play some of the local files.

Might be an idea to integrate this step into the autoinstaller-script.

Rougie85, 2020-01-04:
> 
> 
> I was maybe to solve this problem.
> It looks that when the musicbox started the network wasn‘t ready. Therefore it was not able to login properly in Spotify. I now choosed the option in raspi-config wait for network before boot. A first test was successful. Hopefully it works now.

This isn't a suitable solution for me, because maybe WiFi is disabled, the Box is used at a friends home etc.
Is it possible to restart the mopidy server automatically e.g. after the startup sound was played? But I don't know how...

s-martin, 2020-03-07:
Please check, if this is fixed with the upgrade to Mopidy 3 #803

Rougie85, 2020-03-18:
> 
> 
> Please check, if this is fixed with the upgrade to Mopidy 3 #803

I made the update today, looks like spotify is working right after start. Thanks!

s-martin, 2020-03-19:
So I close this issue, feel free to reopen it, if the problem still exists.

closed 2020-03-19 by s-martin