MiczFlor/RPi-Jukebox-RFID #645: Version  2.0 rc6 start and shutdown no sound

Leobo71, 2019-12-02:
After the latest update no start and shutdown sound. In earlier version everything perfect.
Im use USB Soundcard. Playing tracks  sound is audible.

MiczFlor, 2019-12-02:
Hi, did you just update or did you run the install script? The services have changed, possibly that's the reason why. And the current version is built for buster and tested on buser (not jessie anymore).
You "should" be able to run the install on top of your existing version. I tried it, it recognizes the current install, asks to use existing files and cardIDs. If you use the default audio file location, it should work.

Leobo71, 2019-12-02:
Hello thank you for quick response. I have made new installation of one line installer on buster. I just installed 2.0 rc4 again and try updating to 2.0 rc6.

Leobo71, 2019-12-02:
Update or new installation I have no sound start. Played tracks are audible.

donhomie, 2019-12-03:
Same problem here. Did not do a fresh install, but upgraded from rc5. Startup and shutdown sounds worked fine before the upgrade. Everything else still seems to work fine.

I do have a problem resuming playlists from the last track played, but that was already the case in rc5. Even though I enabled this in the webinterface, it simply always starts playing from the first track.

ps: as this is the first post I just wanted to say a big thank you for this great project! If my daughters will actually return the prototype to me I might post a picture of the finished product before Christmas.

Leobo71, 2019-12-03:
OK, I'm not alone with the problem. List, track  shuffle and start sound don't work for me.
2.0 rc6 one line install without Spotify. Whether reinstallation or update same problem.

MiczFlor, 2019-12-04:
Hi @Leobo71 @donhomie 
could you please check if you are in the master branch or the develop branch? You do this by going to the terminal, move into the Phoniebox directory and type
~~~
git branch
~~~
If there is a asterisk (star) next to `develop`, you need to type
~~~
git checkout master
~~~
Then reboot and try again.
I tested the fresh install twice, worked for me, but I believe that it doesn't work for you. So I am trying to circle in on the problem.
All the best and thanks for sharing the problems, helps everybody to make it better.

Leobo71, 2019-12-04:
Hi @MiczFlor .Thanks for your time. I tested everything without success. Buster lite or stretch, develop or master. No start sound. Pi zero with USB Hub, USB Audio card and USB RFID. What else noticed that I have set "Skip to Next track". It works.
Loop List or Track don't work for me. How can I download 2.0 rc 4 again?
Thanks

Leobo71, 2019-12-06:
Hi @MiczFlor, I install JustBoom Dac Zero on my Pi Zero, The self effect. No Start Up Sound. Everything else works.

Leobo71, 2019-12-07:
Hi @MiczFlor 
This `After=default.target`dont work for me.
This `After=network.target iptables.service firewalld.service rfid-reader.service`work fine.

MiczFlor, 2019-12-13:
Hi @Leobo71 
what soundcard do you use?
And are there any other differences in your solution than here?
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/misc/sampleconfigs/phoniebox-startup-sound.service.stretch-default.sample

Leobo71, 2019-12-14:
Hi @MiczFlor I have one raspberry zero whit usb soundcard and the other justboom dac zero. whit this 'After=default.target' I have don't start up sound. 
'After=iptables.service firewalld.service' or 'After=firewalld.service rfid-reader.service' 
I hear this start sound.

s-martin, 2020-04-13:
Is this still an issue?
You could post the results of `tail -n 500 /var/log/syslog`.

s-martin, 2020-06-26:
Should be fixed with #1026

closed 2020-06-28 by s-martin