MiczFlor/RPi-Jukebox-RFID #647: Mopidy-Spotify

chrstn-hsbrg, 2019-12-05:
I've installed Phoniebox. Local files are scanned and played back. There is a problem with the Spotify integration.

The iris/settings/spotify shows me the following. 

````
SERVER
Status: Connected
Username: *****
Host: *****
Port: 6680

SERVICES
Spotify: Authorized
Mopidy-Spotify extension is not running - you will not be able to play any Spotify tracks
````

Does anyone have an idea why the extension doesn't seem to be running? 

Thanks and Greetings
Christian

My Setup:
````
System
Betriebssystem: Raspbian
Beschreibung: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Phoniebox Setup
Version: 2.0 rc4
Edition: Plus Edition (feat. Spotify Integration)
Mopidy Server Status: active (running) since Thu 2019-12-05 08:28:28 GMT; 20min ago
````

MiczFlor, 2019-12-06:
I have no Spotify myself, but this has been reported a few times and apparently it might have to do with the Pi clock. Read here and follow the links:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#spotify-does-not-connect-or-is-really-slow-to-connect

chrstn-hsbrg, 2019-12-23:
Thx for your answer. I haven't had time to fix the problem. But that sounds like a very good approach.

closed 2019-12-23 by chrstn-hsbrg

timomeinen, 2020-11-08:
In my setup the problem with this error messsage was a missing Spotify password in `/etc/mopidy/mopidy.conf` `[Spotify]`-section.