MiczFlor/RPi-Jukebox-RFID #648: No matter what I do - I can't get it running (One-line-install Script)

UndIch, 2019-12-05:
HI,

I really tried to find a solution on my own - but as I'm new to the Pi and to this kind of programming as well - I think I need your help.

I've tried to install the Phoniebox for about 30 times now - but I can't get it running. 

My Setup: PI3b, Neufatek Reader, SanDisk Extreme 32GB microSDHC - and because I thought it might be the SD - additional a Samsung MB-ME32GA/EU EVO Select 32.
I'm using the latest stable release form Buster 
I connect via SSH (works fine) 
Then I start the install script and- bam: 
`#####################################################
#
# START INSTALLATION
#
# Good news: you completed the input.
# Let the install begin.
#
# Get yourself a cup of something. The install takes 
# between 15 minutes to half an hour, depending on 
# your Raspberry Pi and Internet connectivity.
#
# You will be prompted later to complete the installation.

Do you want to start the installation? [Y/n] Y
Get:1 http://archive.raspberrypi.org/debian buster InRelease [25.2 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]                            
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [259 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Fetched 13.3 MB in 8s (1,600 kB/s)                                                                                                
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gcc is already the newest version (4:8.3.0-1+rpi2).
gcc set to manually installed.
The following additional packages will be installed:
  attr exim4-base exim4-config exim4-daemon-light fontconfig fontconfig-config fonts-dejavu-core gir1.2-glib-2.0 git-man
  guile-2.2-libs ibverbs-providers libaacs0 libadplug-2.2.1-0v5 libao-common libao4 libaom0 libass9 libasyncns0 libaudio2
  libaudiofile1 libavahi-client3 libavc1394-0 libavcodec58 libavdevice58 libavfilter7 libavformat58 libavresample4 libavutil56
  libbdplus0 libbinio1v5 libbluray2 libboost-atomic1.67.0 libboost-iostreams1.67.0 libboost-regex1.67.0 libboost-system1.67.0
  libboost-thread1.67.0 libbs2b0 libcaca0 libcairo2 libcdio-cdda2 libcdio-paranoia2 libcdio18 libcephfs2 libchromaprint1
  libcodec2-0.8.1 libcroco3 libcups2 libcurl3-gnutls libdatrie1 libdc1394-22 libdrm-amdgpu1 libdrm-common libdrm-nouveau2
  libdrm-radeon1 libdrm2 liberror-perl libexpat1-dev libfaad2 libfam0 libfftw3-double3 libfl2 libflac8 libflite1 libfluidsynth1
  libfontconfig1 libgc1c2 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgfapi0 libgfrpc0 libgfxdr0
  libgirepository-1.0-1 libgl1 libgl1-mesa-dri libglapi-mesa libglusterfs0 libglvnd0 libglx-mesa0 libglx0 libgme0 libgnutls-dane0
  libgpgme11 libgraphite2-3 libgsasl7 libgsm1 libharfbuzz0b libibverbs1 libice6 libid3tag0 libiec61883-0 libiso9660-11 libixml10
  libjack-jackd2-0 libjansson4 libjbig0 libjs-sphinxdoc libjs-underscore libkyotocabinet16v5 libldb1 liblilv-0-0 libllvm8
  libltdl7 liblzo2-2 libmad0 libmailutils5 libmariadb3 libmikmod3 libmms0 libmodplug1 libmp3lame0 libmpcdec6 libmpdclient2
  libmpg123-0 libmysofa0 libnfs12 libnorm1 libnspr4 libnss3 libntlm0 libogg0 libopenal-data libopenal1 libopenjp2-7 libopenmpt0
  libopus0 libout123-0 libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpgm-5.2-0 libpixman-1-0 libportaudio2
  libpostproc55 libpulse0 libpython-all-dev libpython-dev libpython2-dev libpython2.7 libpython2.7-dev libpython2.7-minimal
  libpython2.7-stdlib librados2 libraw1394-11 librsvg2-2 librsvg2-common librubberband2 libsdl2-2.0-0 libsensors-config
  libsensors5 libserd-0-0 libshine3 libshout3 libsidplayfp4 libsm6 libsmbclient libsnappy1v5 libsndfile1 libsndio7.0 libsodium23
  libsord-0-0 libsoxr0 libspeex1 libsratom-0-0 libssh-gcrypt-4 libswresample3 libswscale5 libtdb1 libtevent0 libthai-data
  libthai0 libtheora0 libtiff5 libtwolame0 libunbound8 libupnp13 libva-drm2 libva-x11-2 libva2 libvdpau-va-gl1 libvdpau1
  libvidstab1.1 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx5 libwavpack1 libwayland-client0 libwayland-cursor0
  libwayland-egl1 libwebp6 libwebpmux3 libwildmidi2 libx11-xcb1 libx264-155 libx265-165 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0
  libxcb-present0 libxcb-render0 libxcb-shape0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0 libxcursor1 libxdamage1 libxfixes3 libxi6
  libxinerama1 libxkbcommon0 libxrandr2 libxrender1 libxshmfence1 libxss1 libxt6 libxtst6 libxv1 libxvidcore4 libxxf86vm1
  libyajl2 libzmq5 libzvbi-common libzvbi0 libzzip-0-13 lighttpd-modules-ldap lighttpd-modules-mysql mailutils mailutils-common
  mariadb-common mesa-va-drivers mesa-vdpau-drivers mysql-common php-common php7.3-cli php7.3-json php7.3-opcache php7.3-readline
  python-all python-all-dev python-asn1crypto python-cffi-backend python-configparser python-crypto python-cryptography
  python-dbus python-dnspython python-entrypoints python-enum34 python-gi python-gpg python-ipaddress python-keyring
  python-keyrings.alt python-ldb python-pip-whl python-pkg-resources python-samba python-secretstorage python-setuptools
  python-six python-talloc python-tdb python-wheel python-xdg python2-dev python2.7 python2.7-dev python2.7-minimal
  python3-colorzero python3-rpi.gpio samba-common samba-dsdb-modules samba-libs samba-vfs-modules spawn-fcgi tdb-tools
  va-driver-all vdpau-driver-all x11-common
Suggested packages:
  exim4-doc-html | exim4-doc-info eximon4 spf-tools-perl swaks ffmpeg-doc git-daemon-run | git-daemon-sysvinit git-doc git-el
  git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn adplug-utils libsndio6.1 nas libbluray-bdj cups-common fam
[... 2089 lines cut ...]
  ; type=INTEGER,access=rw---R--,values=1,min=-10239,max=400,step=0
  : values=400
  | dBscale-min=-102.39dB,step=0.01dB,mute=1
rm: cannot remove '/home/pi/RPi-Jukebox-RFID/settings/global.conf': No such file or directory

#
# INSTALLATION FINISHED
#
#####################################################

If you are using an USB RFID reader, connect it to your RPi.
(In case your RFID reader required soldering, consult the manual.)
Have you connected your USB Reader? [Y/n] Y
./buster-install-default.sh: line 833: cd: /home/pi/RPi-Jukebox-RFID/scripts/: No such file or directory
python2: can't open file 'RegisterDevice.py': [Errno 2] No such file or directory
chown: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory

DONE. Let the sounds begin.
Find more information and documentation on the github account:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/
pi@raspberrypi:~ $ 
`

Whats going on there?

PanoLolas, 2019-12-05:
Seems your raspi isn't able to download the packages needed. Are you shre you are connected with the internet and the connection is running?

If so, your dhcp server i.e. router might not work as it should.

Try:
`sudo nano /etc/resolv.conf`

and then add the lines
```
nameserver 8.8.8.8
nameserver 8.8.4.4
```
That's the google DNS server. Hope this works for you.

MiczFlor, 2019-12-06:
Hi @UndIch 
my thoughts: your Pi and another device struggle over the same IP address in your WiFi network. Seems that the Pi had connectivity and then lost it. That should not happen during the install process. It could happen after reboot - if the SSID and IP etc are wrong.
My suggestion:
1. reinstall a plain buster on the SD card
2. Connect to the Internet via Raspbian, monitor, keyboard, the works and do NOT set a static IP 
3. start the install script and do NOT change the WiFi settings
4. Log into your router and assign a static IP address n the router for the Phoniebox

marmacman, 2019-12-06:
Having exactly the same errors.  Tried a number of times now with the same result.  Using a raspberry pi zero w, headless with buster lite.  Please let me know if you sort out the problem and how you managed to do it...all the best!

UndIch, 2019-12-07:
I just did both recommendations- I deleted the pi from the fritzbox and let the fritzbox “choose” a new is for it - and I added the google name server (found it recommended somewhere else as well  a couple of weeks ago and forgot about it...). And what shall I say? It’s working - even the soldering of the “hardware” hack is working.

closed 2019-12-17 by UndIch