MiczFlor/RPi-Jukebox-RFID #649: Issue with HifiBerry DAC

maxpousre, 2019-12-06:
Hi, Sorry to bring this up but I'm not sure where to turn. My HifiBerry Dac+ seems to fail after configuration of Phonie Box. I have followed the guides here but am still having an issue. Here are the steps I have taken. 

1. Raspbian Buster install 
2. Edit the boot config file 
`dtoverlay=hifiberry-dac`
3. Edit /etc/asound.conf
```
`pcm.hifiberrydac {  
	type softvol  
	slave.pcm "plughw:0"  
	control.name "Master"  
	control.card 0  
}  
pcm.!default {  
	type       plug  
	slave.pcm  "hifiberrydac"  
} `
```
4. I can play mp3 files through VLC or mplayer. The led on the board illuminates when it start the song. The control 'Master' can be found under alsamixer. Audio is good.  

5. I run the Phoniebox install script for Buster, and ensure that the iFace Name is 'Master'. 
6. Edit /etc/mpd.conf
```
audio_output {  
	type		"alsa"
	name		"My ALSA Device"
	mixer_control	"Master"	# optional
}  
audio_output {  
                enabled         "yes"
		type            "alsa"
		name            "HiFiBerry DAC"
		device          "sndrpihifiberry"
		auto_resample   "no"
		auto_channels   "no"
		auto_format     "no"
		dop             "no"
}   
```
7. Restart mpd service. 
8. At this point I can play mp3s in mplayer or in phoniebox via the web interface. 
9. Reboot. 
10. Now the DAC's led lights up at boot (stays lit) and I get no audio from phoniebox. I can see the progression of the song but no audio. I have checked the GPIO Pins to make sure they aren't interfering. Do you have any other ideas of what I might check to trouble shoot the issue?

Leobo71, 2019-12-06:
Hi, Problem is with GPIO pins. Maybe that will help you  
`cd home/pi/RPi-Jukebox-RFID`
and 
`sudo systemctl disable phoniebox-gpio-buttons`
If you want to reactivate 
`sudo systemctl enable phoniebox-gpio-buttons`

maxpousre, 2019-12-07:
Thank you @Leobo71 . I thought changing the Pin numbers would have been enough but I obviously missed something there, as it was definitely the problem. I'll review my set up again and make sure that they won't interfere as I do plan on using the GPIO pins. Thanks. again.

closed 2019-12-07 by maxpousre