MiczFlor/RPi-Jukebox-RFID #650: SONOS Integration

oneshow01, 2019-12-06:
Hi,
Has anyone already managed to integrate the Phoniebox into the Sonos system/ audio output --> Sonos speakers?

Thanks

sassen, 2019-12-06:
@oneshow01 have a look to this https://github.com/maddox/magic-cards
I have it up and running on a Rpi Zero W and integrated with Sonos.

oneshow01, 2019-12-06:
Hello, sounds interesting. Thanks to you!
But I would prefer to solve this with my existing Phoniebox, I love this box....

sassen, 2019-12-06:
I see, lets see if anyone here knows how to integrate Sonos with Phoniebox. I would be interested, I miss the GPIO integration in my current setup.

UndIch, 2019-12-06:
I think it was discussed in #499 - you might want to check on that.

s-martin, 2020-04-12:
Closing as duplicate.

closed 2020-04-12 by s-martin