MiczFlor/RPi-Jukebox-RFID #651: Missing Feature Delete Directory

Hans-Haefner, 2019-12-07:
I miss a way to delete an entire directory or I just overlook what?

ctietze, 2019-12-09:
Hi @Hans-Haefner ,
as far as I know deleting folders is only possible via SSH or via SAMBA. Within the webapp you can only delete single tracks.

BR Clemens

Hans-Haefner, 2019-12-11:
Would be nice if something would be installed, there are aspirations to it?

MiczFlor, 2019-12-13:
Yep, would be nice, won't be this christmas, though ;)

Ardios, 2019-12-19:
Hey all,

at first, sorry for my broken english.

@Hans-Haefner 
I added a file manager maybe it will help you, here is the link:

[https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/510](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/510)

Erechtheus, 2019-12-22:
@Ardios It might make sense to add your file manager as a pull request?  

BTW: Duplicate of #380

s-martin, 2020-04-12:
Closing as it’s a duplicate of #380

closed 2020-04-12 by s-martin