MiczFlor/RPi-Jukebox-RFID #654: Volume Control not working with Hifiberry MiniAmp

marmacman, 2019-12-08:
This project has introduced myself to the raspberry pi and coding and so I am a complete beginner.  I have the Phoniebox working on my pi zero W and I am able to play and change tracks with the USB RFID reader however I can not change the volume.  Adjusting the volume using the RFID reader and in the phoniebox web app does not do anything, it is stuck at 100%.  I am using a hifiberry miniamp.  Initially I had difficulty getting any sound from the phoniebox using this miniamp.  Before I was able to get this to work I was able to use my RFID card reader to alter volume levels within the web app but once I actually got the miniamp to play audio then the ability to alter the volume levels was lost.  I would be very grateful if someone can give me some useful advice.

Leobo71, 2019-12-08:
Hi,
Have you change mpd.conf mixer_type "hardware" to  "software".

marmacman, 2019-12-08:
Thank You so much!!  This has fixed it.  This is a steep learning curve for me but really enjoying it!  Once this project is done I will start from the beginning with something a little more simple. once again thanks for your quick response.

closed 2019-12-08 by marmacman