MiczFlor/RPi-Jukebox-RFID #656: GPIO buttons scripts conflict with I2C

domu83, 2019-12-11:
Hi,
already somehow discussed in issue #406 
I have found during the problem of powering up Mopidy/Spotify connection and trying to fix via RTC, that it seems there is a I2C blockage. I experienced this first with my I2C LED. So once phoniebox code is loaded I2C devices seems not reachable anymore. 
For example with the LED this potential conflict is expressed in the fact, that after powerup, the LED gets refreshed with old date & time (before time sync) and does not get refreshed during the uptime of the phoniebox. 
With my basic code-knowledge I saw that with flexible pinout in 'gpio-buttons.py.rotaryencoder.sample' at least GPIO3 seems to be used for shutdown.

Can anyone please also have another look at this? I may do some tryouts in the code during the weekend.

thanks!!

domu83, 2019-12-14:
okay, checked out... so if shutdown button in the GPIO is not used, you can comment out the shutdown part in the gpio-buttons.py 

```
def def_shutdown():
    check_call("./scripts/playout_controls.sh -c=shutdown", shell=True)
```

`shut = Button(3, hold_time=2)`

`shut.when_held = def_shutdown`

@swissmaster1

with that I2C works back again

closed 2019-12-14 by domu83

mangenehm, 2019-12-14:
Thanks @domu83! Cannot wait to check this out.