MiczFlor/RPi-Jukebox-RFID #657: Spotify Connect?

AaronMeyers, 2019-12-11:
Hi there, I apologize for using a github issue to ask a question, but I came across this project while looking for a way to use RFIDs to play albums on Spotify.

Looks great, but I'm looking for something just to start the album on whichever device of mine is currently playing already. I mostly use Spotify Connect which is built into my Yamaha AVR so I wouldn't want to actually use the audio out from the RPi Jukebox.

Can anyone tell me if this is already possible with the spotify edition? If not, would it be a relatively straightforward modification to make?

ddbkva, 2020-01-03:
hey 
i use this one https://github.com/dtcooper/raspotify 
its a simple code that enable the spotify connect function on your pi.
i use it on a pi zero with the software from micz, and im very happy

mriffault, 2020-01-28:
Hi,

Sorry if I misunderstood the question, or the answer.

I would like to use the Phoniebox with RFIDs for my kid, AND occasionally Spotify Connect to play music from my phone to the Phoniebox, to pick some music out of the playlists sometimes. Will it be possible? Raspotify is required in this use-case? or RPi-Jukebox-RFID may be compatible with Spotify Connect out of the box...? Will Raspotify and RPi-Jukebox-RFID work together on the same RPi without problem?

Thanks for the answer.

domu83, 2020-01-28:
Hi @mriffault 
Spotify connect is not included in any version of the phoniebox. The Spotify version enables you to play Spotify on the box by using a webinterface.
During the weekend I installed raspotify on top of the phoniebox and it worked like a charme in parallel. So just install it and use it. If you do not plan to attach RFID cards to a spotify uri for your kids, than the classic version of phoniebox plus raspotify suits you well

ZyanKLee, 2020-01-29:
@mriffault you might want to use the webinterface of the phoniebox to choose your spotify playlists/albums of choice. That works rather good for me and my family.

mriffault, 2020-01-29:
Thanks @domu83 to have tested it for me. I'm glad it could work together, both of the possibilities are interesting for me.
Last thing which will missing for me is the possibility to put some Spotify tracks offline (to play without an Internet connection) on the phoniebox.

MiczFlor, 2020-01-29:
Hi @mriffault 
you might want to try 
https://github.com/dtcooper/raspotify
and install this on your Phoniebox for Spotify Connect

Having local files from spotify on the Phoniebox is not easily possible. Phoniebox at this stage connects like a streaming app to Spotify - I don't see the possibility to download files (possibly I am wrong...?)

domu83, 2020-01-29:
@MiczFlor Yep, I agree with your conclusion. 
Airplay could be a solution for the offline content. I installed according to the wiki(https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/AirPlay_ShairportSync),  but I have volume issues (literally playing at 1% or so) which I was not able to fix on my fingertips but I can confirm you can use Phoniebox + Raspotify + Shareport on the installation

mriffault, 2020-01-29:
Ok, thanks for your reflections. So I will plan to implements a Bluetooth-speaker-like behaviour to be able to send musics (offline on my phone) to the Phoniebox when there is no internet connection (maybe a switch to enable the Bluetooth of the Pi, a pair button, etc). Typically I'm thinking about usage during car travels.

thomas-wolf, 2020-02-29:
I wonder if the future of the spotify part of phoniebox might be spotify connect. Mopidy-Spotify is using pyspotify project which is deprecated and not maintained any longer by Spotify. There is no replacement for it. It might stop working any time.

With spotify connect running on the pi, playback could be controlled over Spotify Web API. RFID cards (as well as the web app) could be used to trigger API calls in the background which in turn would start the playback of the desired playlist/album/track on spotify connect. Any thoughts on that?  Do not know if there is a command line only version of it at all. But I guess the main principle would work out. Might require somewhat more resources. I might do a little prototype once I find a little time for it.

My kids really love using spotify on my phoniebox and the thought that it could stop gives me some headache. A few days ago, there was an outage. Spotify somehow was not working on Phoniebox (while it was doing okay using officially supported Spotify clients). I was really relieved when this outage turned out to be a temporary one later on. 

It might be some work to get spotify connect integrated with the local and streaming parts of phoniebox.

s-martin, 2020-02-29:
Any help and contributions to the Spotify edition are highly appreciated. I don't use Spotify myself, so I try to help others fixing the issues, but mostly on infrastructure level.

AFAIK there's currently nobody else who can contribute to Spotify implementation, so if you support a lot of people will thank you.

thomas-wolf, 2020-02-29:
I was actually trying to figure out what's needed for adding offline support to spotify extension when I stumbled upon the issues with pyspotify.

It's hard to tell how long the current solution will continue working. Could be years, could only be months. There definitely is a big community which still uses pyspotify. I guess Spotify is not going to shut down the interface actively at some time in the future. But when they come across the need for breaking changes in spotify's fundamentals which would also break pyspotify, then they probably won't take the effort to adapt this old, unsupported interface.

So I wonder if it is worth to search for alternative solutions for spotify edition right now .

ZyanKLee, 2020-03-01:
At least I'm aware of this issue, sure to former engagements with Kodi and libreelec.
In this very project we merely consume other projects and provide some "glue" to make them pay nicely together.
We definitely should look for alternatives to the pyspotify-using mopidy-spotify. But writing our own mpd compatible daemon with Spotify support is probably out of the scope of this project.

ZyanKLee, 2020-03-01:
Most probably the devs of mopidy-spotify are aware of this issue, too. So there might be some solution in the works. Searching for one is on my to-do list, but if anyone beats me to it, I won't complain. :wink:

ZyanKLee, 2020-03-01:
This is the information from the project itself. So I think there is no need to worry about the deprecation of pyspotify and libspotify.

> Mopidy-Spotify is dependent on pyspotify, a wrapper for Spotify's libspotify C library. libspotify was deprecated in 2015 with no replacement. It is unmaintained, functionally limited, and also now unavailable from the Spotify developer site. Where possible we are moving to use Spotify's Web API instead. However, native playback is still only possible using libspotify and there is no official way for us to provide some Spotify features.

thomas-wolf, 2020-03-01:
Hey, thanks for sharing your thoughts. 

My point is that Spotify currently does not provide any API which allows for streaming media content except for iOS and Android (and except for libspotify which is not supported officially any longer). And I don’t think this is going to change anytime soon. This is why I don’t think Pyspotify will find a way around. 

The only option I see is to install spotify connect and use official WebAPI to control playback. 

Or just to stick with libspotify and pyspotify and hope it keeps running.

ZyanKLee, 2020-03-01:
If there is a way to make Spotify connect work in the context of the phoniebox without breaking the overall UX to control playback via RFID cards it would be a nice addition.
We need to make sure though, that controlling playback through either technology does not lead to unwanted behavior. For example: playback through Spotify connect has to be stoppable with RFID cards. Playback started via RFID cards needs to be stopped when Spotify connect is being used to play something else. Volume levels need to be consistent even when switching from one to the other.

thomas-wolf, 2020-03-05:
I did some research and I am pretty sure that there is a way. 

Installing spotify connect on a rasp is easy. It can be done with a one-liner. I've tried it and it worked out nicely in peaceful co-existance with phoniebox software. I was then able to control playback on the rasp using spotify app on my smartphone and chosing the rasp from the device list.

Then I had a closer look at the Web API. There is the Player endpoint which allows you to control on which device the playback happens as well as the actual playback. This endpoint is still in beta, though.

The whole Spotify API is nicely wrapped by spotipy which makes it even easier to use in python.  So I'd go for full support in phoniebox, meaning you'd be able to use gpio buttons, RFID cards and so on no matter whether playback is on file system, webstream / podcast or spotify. 

Maybe I get the chance to do a quick prototype anytime soon.

But let me be clear: As long as libspotify is working, I think this solution is clearly preferable as the whole integration with mpd is much better and it fits  better into phoniebox architecture. Using spotify connect in addition to mpd comes along with some downsides (definitely no offline option for spotify playback as it is discussed for the mopidy-spotify extension / higher resource consumption as spotify connect needs to run as daemon as well / more complex playout of commands / higher overall complexity of architecture).

There is also the possibility that someone is going to reverse engineer the proprietary protocol spotify is using in spotify connect in order to receive the music stream and implements it in a mopidy extension. 

So maybe for the time being, it is sufficient to have a concept / an idea of what to do if one day libspotify stops its service.

thomas-wolf, 2020-03-05:
BTW, in the meantime i noticed there are / were some projects around dealing with reverse engineering of spotify connect. And there is also discussion ongoing in mopidy community. So maybe I'll do some more research on that.

ZyanKLee, 2020-03-05:
Thanks for looking into it, Thomas. I really appreciate the thought, work
and time you put into this.

One downside I noticed with other Spotify connect appliances like my a/v
receiver is that I can't easily setup another Spotify account for my
children and put it into the device to use only that.

This is exactly what I'm doing right now with our two phonieboxes: one family
contract and four accounts for my wife, two children and me.
No conflicts, everybody can listen to what they want.

thomas-wolf, 2020-03-06:
ZyanKLee, actually I suppose that most users of phoniebox spotify edition do have a family account. So do I. 

You're right, for my quick test of spotify connect, I had to log out of my account on spotify mobile phone client. I just did that in order to test spotify connect client on rasp quickly and see if it is working.

The coolest thing about phoniebox is that so easy to use. You shall not need a mobile phone to control playback on it. Of course, you could still use GPIOs and RFID cards to control spotify playback using spotify web API. So you can still have separate accounts for you and each phoniebox you own. 

We only need to extend the command playout script in a way that it knows whether the source of the RFID card shortcut is spotify or webstream/local filesystem. This can be achieved by looking up the shortcut on file system, I guess. If the source is spotify, then the script would rather call spotify web API than executing a local mpd command. Of course, we need to take care about stopping mpd playback and all of these things.

It seems kind of odd to call an API  whose server can be located anywhere in the world from phoniebox just for sake of controlling playback on the very same local phoniebox. We'll need to have a look at latency for this. However, this is the way spotify forces us to do it, I guess.

So I suggest I'll do a prototype of this to the point where we see how controlling the playback with the web API works out. Might take me some time as the panic corona virus causes currently turns the part-time job of my wife into a full time job. So there is currently not much time left for me for doing such things, but I'll share my experience once I've managed do find some time. This should be sufficient for now as fallback scenario for libspotify.

s-martin, 2020-03-06:
> So I suggest I'll do a prototype of this to the point where we see how controlling the playback with the web API works out. 

That would be really great!

mriffault, 2020-03-12:
> Thanks for looking into it, Thomas. I really appreciate the thought, work and time you put into this. One downside I noticed with other Spotify connect appliances like my a/v receiver is that I [...]

I already use raspotify (a libspotify alternative, named librespot, for raspberry), and I know that we can provide an API key to the library through a configuration file. So you could put your children account API key into it. But I guess that to trigger a music play (for example from your phone) through Spotify Connect (on the phoniebox of your child) you will have to be connected with his account on your phone. I don't have a family account so I don't know if it could be a real drawback.

thomas-wolf, 2020-03-14:
I noticed that there is some kind of detection of spotify connect devices on same WLAN. Don't know for sure if it is limited to the account that the spotify connect device itself is associated with (if any). 

However, let me again emphasize that controlling the playback with your smartphone is only an addon. The main controls shall always reside on the phoniebox device - actually the easy to use card reader and buttons. And still there's the web app that comes with phoniebox - guess with some effort, you could also integrate spotify connect with this web UI.

Unfortunately, corona keeps us busy so I cannot spend any time on the PoC at the moment.

wolfthom, 2020-03-23:
For those of you who understand German: there is a good article about Spotify connect on heise / tech stage 
https://www.techstage.de/ratgeber/Ratgeber-Spotify-Connect-Vernetzte-Lautsprecher-ab-40-EUR-4687982.html

veloxidSchweiz, 2020-05-10:
The discussion waas continued at least partially in #883