MiczFlor/RPi-Jukebox-RFID #660: Spotify - Play (full) album from an artist page randomly

luke666s, 2019-12-11:
is there a chance to play the full albums from an artist page randomly? If i assign the artistpage to a rfid card and click on random it plays single tracks form all albums randomly.... 
my idea was to create a card for "Benjamin Blümchen" (spotify:artist:1l6d0RIxTL3JytlLGvWzYe) and if my son puts the card on phonie box one album of the >143 albums is played.

UndIch, 2019-12-17:
Hi - unfortunately that’s not implemented by Spotify (see this old post on their help
Page: https://community.spotify.com/t5/Live-Ideas/Music-Shuffle-by-Album/idi-p/18457) there could be some kind of solution where you store all AlbumLinks ID to a txt and then somehow shuffle that one?

maurermax, 2020-04-04:
@luke666s have you found any solution to that. maybe by creating a folder of all albums? I am still trying to figure out on how to configure this on RPi-jukebox side

comdoxx, 2021-01-02:
I'd like to emphasize this enhancement again.

To achieve this, a "playlist" of Spotify album links could be generated. The options can be enhanced.

Suggestion:
- audiofolders/spotify:album:123456.../ and so forth shall exist with all the files inside.
- audiofolders/spotify:random:Benjamin Blümchen/ has to be created. The spotify.txt can hold all album descriptors. title.txt is the Name of the Playlist ("Zufälliges Benjamin Blümchen Album"), the folder.conf holds necessary data and the cover.jpg may be uploaded on creation (Only necessary for selection, in the playlist only the actual album will appear).
- To clarify: What to do on second swipe? Stop/Resume? Always a new random album?

s-martin, 2021-01-02:
Hi, I think this would be a useful enhancement.

PRs are always welcome.