MiczFlor/RPi-Jukebox-RFID #661: Installation funzt nicht

mimi-ac, 2019-12-12:
Hallo zusammen,

ich habe heute meine Raspberry 
* mit Raspbian Buster Lite installiert
* SSH eingerichtet
* Wlan eingerichtet und dann
* versucht Phoniebox zu installieren.

Nachfolgend das Ergebnis:

`#####################################################
#
# START INSTALLATION
#
# Good news: you completed the input.
# Let the install begin.
#
# Get yourself a cup of something. The install takes
# between 15 minutes to half an hour, depending on
# your Raspberry Pi and Internet connectivity.
#
# You will be prompted later to complete the installation.

Do you want to start the installation? [Y/n] Y
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [25.2 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0                                                                                                                                                              MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [260 kB]
Fetched 13.3 MB in 41s (324 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'linux-headers-4.9.0-6-all' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-rpi' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-all-armhf' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-rpi2' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-common' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-common-rt' for regex 'linux-headers-4.9'
Note, selecting 'php7.0-thrift' for regex 'php7.0'
Note, selecting 'php7.0-remctl' for regex 'php7.0'
Note, selecting 'php7.0-common' for regex 'php7.0'
Note, selecting 'php7.0-curl' for regex 'php7.0'
Note, selecting 'php-remctl' instead of 'php7.0-remctl'
Package php7.0-common is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

W: --force-yes is deprecated, use one of the options starting with --allow inste                                                                                                                                                             ad.
E: Package 'php7.0-common' has no installation candidate
E: Unable to locate package php7.0-cgi
E: Couldn't find any package by glob 'php7.0-cgi'
E: Couldn't find any package by regex 'php7.0-cgi'
E: Unable to locate package php7.0-fpm
E: Couldn't find any package by glob 'php7.0-fpm'
E: Couldn't find any package by regex 'php7.0-fpm'
./stretch-install-default.sh: line 394: git: command not found
./stretch-install-default.sh: line 396: cd: RPi-Jukebox-RFID: No such file or di                                                                                                                                                             rectory
./stretch-install-default.sh: line 397: git: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
Error for wireless request "Set Power Management" (8B2C) :
[... 89 lines cut ...]
amixer: Cannot find the given element from control default

#
# INSTALLATION FINISHED
#
#####################################################

If you are using an USB RFID reader, connect it to your RPi.
(In case your RFID reader required soldering, consult the manual.)
Have you connected your USB Reader? [Y/n] Y
./stretch-install-default.sh: line 693: cd: /home/pi/RPi-Jukebox-RFID/scripts/:                                                                                                                                                              No such file or directory
python2: can't open file 'RegisterDevice.py': [Errno 2] No such file or director                                                                                                                                                             y
chown: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such                                                                                                                                                              file or directory
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such                                                                                                                                                              file or directory

DONE. Let the sounds begin.
Find more information and documentation on the github account:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/
`

Was mache ich falsch?

Grüße und Danke

Christian

kabakakao, 2019-12-12:
Für mich sieht es so aus, als hättest du das Script für Debian Stretch benutzt und nicht das für Buster.
In deinem Logauszug wird nämlich versucht php-fpm7.0 zu installieren. Bei Buster wird aber php-fpm7.3 mitgeliefert.
Probiere mal folgenden Aufruf:
`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh`

mimi-ac, 2019-12-12:
Super, das hat erst mal funktioniert. Danke

kabakakao, 2019-12-13:
Sehr gut. Schließt du dann das Issue? Es wird hier sonst sehr unübersichtlich.

closed 2019-12-13 by MiczFlor