MiczFlor/RPi-Jukebox-RFID #662: RFID-Reader

mimi-ac, 2019-12-12:
Hallo,

ich habe folgenden RFID-Reader:
https://www.amazon.de/Neuftech-Reader-Kartenleseger%C3%A4t-Kartenleser-Kontaktlos/dp/B018OYOR3E/ref=sr_1_1?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=Neuftech+USB+RFID+Reader+ID+Kartenleseger%C3%A4t+Kartenleser+Kontaktlos+Card+Reader+f%C3%BCr+EM4100&qid=1576182475&s=computers&sr=1-1

Der Reader ist erkannt worden und die Karten-Id erscheint auch in der Weboberfäche. 
Zusätzlich wird die Karteneingabe aber auch jeweils als Eingabe in der Konsole verarbeitet. Ist das so gewollt?

Grüße

Christian

MiczFlor, 2019-12-13:
Wo ist das Problem?

mimi-ac, 2019-12-13:
Na ja, es wir halt jeweils eine Aktion in der Konsole ausgeführt, die natürlich zu nichts führt.
Bsp.:  
pi@raspberrypi ~ $  234235453
und das führt natürlich zu nichts. Ist das so richtig?

UndIch, 2019-12-17:
Das sollte nur dann passieren, wenn du die Konsole aktiv offen hast - der Reader macht nichts anderes als die gespeicherten Zahlen auf dem Chip lesen und wie eine Tastatur an den pi zu übergeben denke ich.

mimi-ac, 2019-12-17:
okay, das verstehe ich, aber wie mache ich die Konsole "zu".
Wenn der Pi hochgefahren ist, ist ja automatisch der Linux-Prompt da, oder sehe ich das falsch?

s-martin, 2023-12-12:
Closing as won’t fix

closed 2023-12-12 by s-martin