MiczFlor/RPi-Jukebox-RFID #663: Podcasts/Radio take very long to start playing and block app

JuliH29, 2019-12-12:
Hi!

My phoniebox is in its final phase and seems to be running smoothly, except for podcasts & internet radio.
They take 15 seconds to 2 minutes to start playing (no matter if I start them via RFID or the interface in the Browser).
I tried out several podcasts & radios, they behave all more or less the some.
While Phoniebox tries to connect to the podcast/radio, there is no way to interact for me. The web interface is responsive but it doesn't do anything. If Phoniebox can't connect to the source, the only way to regain control is to reboot the whole system

The internet connection is fine, Spotify starts playing right away and is not at all affected by the problem.
I run Phoniebox 2.0 RC6

Any idea why that is?
Thanks,
Julius

JuliH29, 2019-12-12:
Some more information:

This podcast doesn't seem to work at all (have to reboot)
[](https://feeds.lagedernation.org/feeds/ldn-mp3.xml)

These are very slow too slow
[](https://kinder.wdr.de/radio/diemaus/audio/diemaus-30/diemaus-30-104.podcast)
[](http://stream.laut.fm/kinderradio)

JuliH29, 2019-12-12:
Some more information:

This podcast doesn't seem to work at all (have to reboot)
[https://feeds.lagedernation.org/feeds/ldn-mp3.xml](https://feeds.lagedernation.org/feeds/ldn-mp3.xml)

These are very slow too slow
[(https://kinder.wdr.de/radio/diemaus/audio/diemaus-30/diemaus-30-104.podcast](https://kinder.wdr.de/radio/diemaus/audio/diemaus-30/diemaus-30-104.podcast)
[(http://stream.laut.fm/kinderradio](http://stream.laut.fm/kinderradio)

MiczFlor, 2019-12-13:
Hi @JuliH29 
Die Lage der Nation is a monster of an XML file. :) I took a look at it... but that shouldn't be a problem.
Thing is: I just made a couple of screenshots and tested podcasts on my two text boxes, both without Spotify, and both without any problems.
I have no idea what it could be, but possibly Spotify / mopidy plays a role?
Another thought would be DNS entries in the router?

JuliH29, 2019-12-15:
Hi @MiczFlor 
Thanks for your help. It was indeed the DNS entry...

MiczFlor, 2019-12-16:
Hi @JuliH29 
it makes me both happy and feel old at times that I know so many possible traps :)
Enjoy the Phoniebox and do not forget to post a picture!!!
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/639

closed 2019-12-16 by MiczFlor