MiczFlor/RPi-Jukebox-RFID #664: Can't access webapp: 403 Forbidden

JuliH29, 2019-12-15:
Hi!

I just installed Phoniebox (develeope branch on buster) and cannot access the web app afterwards, I just get "403 Forbidden".
As DNS I have set the IP of my router.

During the installation I was asked `Wenn Ihr Computer IP-Adress-Informationen von einem DHCP-Server im       
   Netzwerk bezieht, ist es möglich, dass der DHCP-Server auch              
   Informationen über im Netz vorhandene WINS-Server                        
   (»NetBIOS-Name-Server«) zur Verfügung stellt. Eine Änderung der Datei    
   smb.conf ist erforderlich, damit die über DHCP angebotenen               
   WINS-Einstellungen automatisch aus der Datei /etc/samba/dhcp.conf        
   übernommen werden.                                                       
                                                                            
   Sie müssen das Paket dhcp-client installiert haben, um diese             
   Funktionalität nutzen zu können.                                         
                                                                            
   Soll smb.conf so abgeändert werden, dass per DHCP angebotene             
   WINS-Einstellungen verwendet werden?                                     
                                                                            
                      <Ja>                        <Nein>         `

Here I hit yes.

Also, during the installation process I saw some error messages:
`Could not find a version that satisfies the requirement evdev==0.7.0 (from versions: )
No matching distribution found for evdev==0.7.0`

My Raspian installation was completely new/clean. So Apache was not installed.

I hope that helps to find the issue.
Thanks,
Julius

JuliH29, 2019-12-18:
Here you can find the output of the terminal during the whole installation process (via ssh):
[https://www.dropbox.com/s/r8h7zon1k0nbtj5/Phoniebox%20Installationsprotokoll.rtf?dl=0](https://www.dropbox.com/s/r8h7zon1k0nbtj5/Phoniebox%20Installationsprotokoll.rtf?dl=0)

I would really appreciate some support if anyone finds a moment before christmas...
Thanks

JuliH29, 2019-12-18:
This fixed it for me:
[https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/648#issuecomment-562331868](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/648#issuecomment-562331868)

Maybe it would be worth including the thing with the nameservers in the troubleshooting guide?

ZyanKLee, 2020-01-19:
@MiczFlor this can be closed

closed 2020-02-19 by s-martin