MiczFlor/RPi-Jukebox-RFID #666: Webradio stuttering

carver79, 2019-12-17:
This worked for month now, but since yesterday my mp3 Radio-Streams are stuttering. I know this is already mentioned as a buffer issue in the Manual with a troubleshooting link - but there I can't find any solution... any ideas beside setting up from scratch?!
(Another box with exactly the same setup is still playing 128kb streams without any issues)

ZyanKLee, 2020-01-19:
Easiest thing might be to see if stuttering is dependend on position in your home.
Then see if some part of your hardware might cause the issue - for example: if you use usb wifi dongles on both systems - interchange them and see if the stuttering wanders to the other system.
Try switching the sd-cards. How about now?

Only after that I would setup from scratch and see if this fixes the issue - for whatever reason that might be.

s-martin, 2020-04-12:
Does this issue still persist?

closed 2020-05-25 by s-martin