MiczFlor/RPi-Jukebox-RFID #668: Volume control does not work

Diekoeche, 2019-12-18:
Hey everyone, I'm new here!

Raspbian GNU/Linux 9.3 (stretch) - release 9.3 - 
Phoniebox 1.1.9
Classic edition (barebones)

The volume changes aren't working - not via RFID- not via Web-App and not even via GPOI-Buttons!
But - I can see in the Web-App that for example 'Mute' is active - but its not really muted!
The only way we could control the volume was by typing percentages in an document, but sorry I dont know where that was - I tried so much for months.
Any Ideas what went wrong?
I managed to change from AUX Jack to the USB-output by changing the alsa.conf
but that didn't solve the problem.

Thank you so much for your help!

Diekoeche, 2019-12-18:
/home/pi/RPi-Jukebox-RFID/scripts/helperscripts/Analytics_AfterInstallScript.sh

************************************
*** PHONIEBOX INFO
*** version: 1.1.9
*** edition: classic
*** Audio_iFace_Name:
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 10
*** Max_Volume_Limit: 100
*** Idle_Time_Before_Shutdown: 10
*** Second_Swipe: PAUSE
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: OFF
************************************
*** CONF FILES DEFAULT
 
*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 9,5K Aug 31 19:39 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 1,1K Aug 31 19:39 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 398 Aug 31 19:39 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
-rw-r--r-- 1 root root 70K Aug 31 19:39 /etc/php/7.0/fpm/php.ini
*** /etc/sudoers
-r--r----- 1 root root 784 Aug 31 19:39 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 296 Aug 31 19:39 /etc/systemd/system/phoniebox-rfid-reader.service
-rw-r--r-- 1 root root 340 Aug 31 19:39 /etc/systemd/system/phoniebox-startup-sound.service
-rw-r--r-- 1 root root 289 Aug 31 19:39 /etc/systemd/system/phoniebox-gpio-buttons.service
-rw-r--r-- 1 root root 291 Aug 31 19:39 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Aug 31 19:39 /etc/mpd.conf
music_directory         "/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
        mixer_control   ""              # optional
#       mixer_control   ""              # optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 130 Aug 31 19:39 /etc/dhcpcd.conf
static ip_address=***
static routers=***
static domain_name_servers=***
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw-r--r-- 1 root root 220 Dez  5 20:30 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition
 
*** /etc/locale.gen
-rw-r--r-- 1 root root 8,9K Aug 31 18:54 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 8,9K Aug 31 18:54 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
ls: Zugriff auf '/etc/mopidy/mopidy.conf' nicht möglich: Datei oder Verzeichnis nicht gefunden
cat: /etc/mopidy/mopidy.conf: Datei oder Verzeichnis nicht gefunden
*** ~/.config/mopidy/mopidy.conf
ls: Zugriff auf '/home/pi/.config/mopidy/mopidy.conf' nicht möglich: Datei oder Verzeichnis nicht gefunden
cat: /home/pi/.config/mopidy/mopidy.conf: Datei oder Verzeichnis nicht gefunden

MiczFlor, 2019-12-19:
Hi @Diekoeche 
would you consider updating to buster and version 2.0 rc? I don't want to support the older versions.

Diekoeche, 2019-12-20:
I can try...
But sorry what do you mean by version 2.0 rc?
I habe to Google everything  I do....😬

M0D-F, 2019-12-22:
Hi, I think I have a similar problem.
I`ve updated to actual Version.

@Diekoeche 
cd /home/pi/RPi-Jukebox-RFID
git pull

But my Volume Button`s still not working.

Do you have any other Ideas to try?

root@raspberrypi:/home/pi# /home/pi/RPi-Jukebox-RFID/scripts/helperscripts/Analytics_AfterInstallScript.sh
************************************
*** PHONIEBOX INFO
*** version: 2.0 rc7
*** edition: plusSpotify
*** Audio_iFace_Name: PCM
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 3
*** Max_Volume_Limit: 100
*** Idle_Time_Before_Shutdown: 0
*** Second_Swipe: PAUSE
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: ON
************************************
*** CONF FILES DEFAULT

*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 8.2K Oct  3 14:55 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 2.1K Oct  3 14:55 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 575 Oct  3 14:55 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
ls: cannot access '/etc/php/7.0/fpm/php.ini': No such file or directory
*** /etc/sudoers
-r--r----- 1 root root 784 Oct  3 14:55 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 296 Oct  3 14:55 /etc/systemd/system/phoniebox-rfid-reader.service
-rw-r--r-- 1 root root 340 Oct  3 14:55 /etc/systemd/system/phoniebox-startup-sound.service
-rw-r--r-- 1 root root 289 Oct  3 14:55 /etc/systemd/system/phoniebox-gpio-buttons.service
-rw-r--r-- 1 root root 291 Oct  3 14:55 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 13K Dec 16  2018 /etc/mpd.conf
music_directory         "/var/lib/mpd/music"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
#       mixer_control   "PCM"           # optional
#       mixer_control   "PCM"           # optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 131 Oct  3 14:55 /etc/dhcpcd.conf
static ip_address=192.168.178.128/24
static routers=192.168.178.1
static domain_name_servers=8.8.8.8 192.168.178.1
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw-rw-r-- 1 root netdev 149 Oct  3 14:55 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition

*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Oct  3 14:55 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Oct  3 14:55 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
-rw-r----- 1 mopidy root 1.1K Oct  3 14:55 /etc/mopidy/mopidy.conf
username = line@datenschleuder.com
*** ~/.config/mopidy/mopidy.conf
ls: cannot access '/root/.config/mopidy/mopidy.conf': No such file or directory
cat: /root/.config/mopidy/mopidy.conf: No such file or directory

s-martin, 2020-04-12:
Does this issue still persist?

M0D-F, 2020-04-14:
hello, I have to admit that I am not dealing with the problem right now. The new switches are still late in delivery - ali Express :-)

Thanks for asking. I will report when the switches are installed

hortator, 2020-06-29:
Hello,

just a thought. After a upgrade from an RC, don't know the exact version to the newest build. Volumeup and Volumedown suddendly stopped working from web and the buttons. Nexttrack etc. did work.

After some searching I did notice that in alsamixer suddendly Headphone was listed. But not labled PCM. After changing PCM to Headphone in mpd.conf and Audio_iFace_Name it started working again.

Before the update PCM worked just fine.

So have you checked with alsamixer?

//s

PS: Phoniebox is an amazing project.

s-martin, 2020-06-29:
> Hello,
> 
> just a thought. After a upgrade from an RC, don't know the exact version to the newest build. Volumeup and Volumedown suddendly stopped working from web and the buttons. Nexttrack etc. did work.

That's a different issue which was introduced with the May 2020 update of Raspberry Pi.
Please see #973

Stocki92, 2020-07-05:
Hi all,
i am new here, too. I have a phoniebox running since a year and i´m now trying to get a second one running. It works, but i cant change the volume. Not via RFID, not via Web-App and not via encoder.

Stocki92, 2020-07-05:
/home/pi/RPi-Jukebox-RFID/scripts/helperscripts/Analytics_AfterInstallScript.sh
************************************
*** PHONIEBOX INFO
*** version: 2.0 - ed888da - master
*** edition: classic
*** Audio_iFace_Name: PCM
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 3
*** Max_Volume_Limit: 100
*** Idle_Time_Before_Shutdown: 0
*** Second_Swipe: RESTART
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: ON
************************************
*** CONF FILES DEFAULT
 
*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 8,2K Jul  5 21:57 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 2,1K Jul  5 21:57 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 575 Jul  5 21:57 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
ls: Zugriff auf '/etc/php/7.0/fpm/php.ini' nicht möglich: Datei oder Verzeichnis nicht gefunden
*** /etc/sudoers
-r--r----- 1 root root 784 Jul  5 21:57 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 295 Jul  5 21:57 /etc/systemd/system/phoniebox-rfid-reader.service
-rwxr-xr-x 1 root root 396 Jul  5 22:36 /etc/systemd/system/phoniebox-startup-sound.service
-rw-r--r-- 1 root root 290 Jul  5 21:57 /etc/systemd/system/phoniebox-gpio-buttons.service
-rw-r--r-- 1 root root 291 Jul  5 21:57 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Jul  5 21:57 /etc/mpd.conf
music_directory		"/home/pi/RPi-Jukebox-RFID/shared/audiofolders"

s-martin, 2020-07-05:
Can you please post the results of `tail -n 500 /var/log/syslog`?

Stocki92, 2020-07-05:
sure:

tail -n 500 /var/log/syslog
Jul  5 22:19:45 PhonieboxHL kernel: [    1.761061] NET: Registered protocol family 10
Jul  5 22:19:45 PhonieboxHL kernel: [    1.762393] Segment Routing with IPv6
Jul  5 22:19:45 PhonieboxHL kernel: [    1.837541] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
Jul  5 22:19:45 PhonieboxHL kernel: [    1.837557] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Jul  5 22:19:45 PhonieboxHL kernel: [    1.840237] smsc95xx v1.0.6
Jul  5 22:19:45 PhonieboxHL kernel: [    1.932352] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:cf:00:5b
Jul  5 22:19:45 PhonieboxHL kernel: [    1.932833] usb 1-1-port2: over-current change #1
Jul  5 22:19:45 PhonieboxHL kernel: [    2.547213] usb 1-1.3: new low-speed USB device number 4 using dwc_otg
Jul  5 22:19:45 PhonieboxHL kernel: [    2.690117] usb 1-1.3: New USB device found, idVendor=16c0, idProduct=27db, bcdDevice= 0.01
Jul  5 22:19:45 PhonieboxHL kernel: [    2.690134] usb 1-1.3: New USB device strings: Mfr=1, Product=0, SerialNumber=0
Jul  5 22:19:45 PhonieboxHL kernel: [    2.690144] usb 1-1.3: Manufacturer: HXGCoLtd
Jul  5 22:19:45 PhonieboxHL kernel: [    2.701750] input: HXGCoLtd Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/0003:16C0:27DB.0001/input/input0
Jul  5 22:19:45 PhonieboxHL kernel: [    2.761588] random: systemd: uninitialized urandom read (16 bytes read)
Jul  5 22:19:45 PhonieboxHL kernel: [    2.767736] hid-generic 0003:16C0:27DB.0001: input,hiddev96,hidraw0: USB HID v1.10 Keyboard [HXGCoLtd] on usb-3f980000.usb-1.3/input0
Jul  5 22:19:45 PhonieboxHL kernel: [    2.775642] random: systemd: uninitialized urandom read (16 bytes read)
Jul  5 22:19:45 PhonieboxHL kernel: [    2.776957] random: systemd: uninitialized urandom read (16 bytes read)
Jul  5 22:19:45 PhonieboxHL kernel: [    2.929415] i2c /dev entries driver
Jul  5 22:19:45 PhonieboxHL kernel: [    3.067264] usb 1-1.4: new low-speed USB device number 5 using dwc_otg
Jul  5 22:19:45 PhonieboxHL kernel: [    3.212830] usb 1-1.4: New USB device found, idVendor=1113, idProduct=3163, bcdDevice= 0.03
Jul  5 22:19:45 PhonieboxHL kernel: [    3.212848] usb 1-1.4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
Jul  5 22:19:45 PhonieboxHL kernel: [    3.212857] usb 1-1.4: Product: MEDION MD 86996
Jul  5 22:19:45 PhonieboxHL kernel: [    3.229538] input: MEDION MD 86996 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:1113:3163.0002/input/input1
Jul  5 22:19:45 PhonieboxHL kernel: [    3.298844] hid-generic 0003:1113:3163.0002: input,hidraw1: USB HID v1.00 Keyboard [MEDION MD 86996] on usb-3f980000.usb-1.4/input0
Jul  5 22:19:45 PhonieboxHL kernel: [    3.323681] input: MEDION MD 86996 Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:1113:3163.0003/input/input2
Jul  5 22:19:45 PhonieboxHL kernel: [    3.324243] input: MEDION MD 86996 System Control as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:1113:3163.0003/input/input3
Jul  5 22:19:45 PhonieboxHL kernel: [    3.387631] input: MEDION MD 86996 Consumer Control as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:1113:3163.0003/input/input4
Jul  5 22:19:45 PhonieboxHL kernel: [    3.388113] hid-generic 0003:1113:3163.0003: input,hiddev97,hidraw2: USB HID v1.00 Mouse [MEDION MD 86996] on usb-3f980000.usb-1.4/input1
Jul  5 22:19:45 PhonieboxHL kernel: [    3.704825] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Jul  5 22:19:45 PhonieboxHL kernel: [    4.901797] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
Jul  5 22:19:45 PhonieboxHL kernel: [    4.904391] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
Jul  5 22:19:45 PhonieboxHL kernel: [    4.904409] [vc_sm_connected_init]: start
Jul  5 22:19:45 PhonieboxHL kernel: [    4.909876] [vc_sm_connected_init]: installed successfully
Jul  5 22:19:45 PhonieboxHL kernel: [    4.912665] media: Linux media interface: v0.10
Jul  5 22:19:45 PhonieboxHL kernel: [    4.973470] videodev: Linux video capture interface: v2.00
Jul  5 22:19:45 PhonieboxHL kernel: [    5.031517] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
Jul  5 22:19:45 PhonieboxHL kernel: [    5.032993] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
Jul  5 22:19:45 PhonieboxHL kernel: [    5.058991] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
Jul  5 22:19:45 PhonieboxHL kernel: [    5.095583] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
Jul  5 22:19:45 PhonieboxHL kernel: [    5.120944] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
Jul  5 22:19:45 PhonieboxHL kernel: [    5.120984] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
Jul  5 22:19:45 PhonieboxHL kernel: [    5.127578] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
Jul  5 22:19:45 PhonieboxHL kernel: [    5.127619] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
Jul  5 22:19:45 PhonieboxHL kernel: [    5.138902] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
Jul  5 22:19:45 PhonieboxHL kernel: [    5.138947] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[... cut ...]
Jul  5 22:45:36 PhonieboxHL dhcpcd[457]: eth0: Router Advertisement from fe80::925c:44ff:fe70:f52a
Jul  5 22:45:36 PhonieboxHL dhcpcd[457]: wlan0: Router Advertisement from fe80::925c:44ff:fe70:f52a
Jul  5 22:48:32 PhonieboxHL dhcpcd[457]: eth0: Router Advertisement from fe80::925c:44ff:fe70:f52a
Jul  5 22:48:32 PhonieboxHL dhcpcd[457]: wlan0: Router Advertisement from fe80::925c:44ff:fe70:f52a
Jul  5 22:51:13 PhonieboxHL dhcpcd[457]: eth0: Router Advertisement from fe80::925c:44ff:fe70:f52a
Jul  5 22:51:13 PhonieboxHL dhcpcd[457]: wlan0: Router Advertisement from fe80::925c:44ff:fe70:f52a

s-martin, 2020-07-11:
@Stocki92, could you please try https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#raspberry-pi-update-may-2020-broke-my-audio

Caliandroid, 2020-07-23:
I had the same problem after a complete new installation because of a broken power cable (i thought it was the sd card).
Everything worked out of the box besides the volume buttons.  
I'm happy that i've found this issue with @s-martin 's link to the right solution  (i had to replace PCM by Headphone in the three files) on my RaspberryPi 4

Stocki92, 2020-08-09:
@s-martin : 
I've gone through the troubleshooting. After changing the three files, volume contol works via web app, but still not with the encoder.
What am I doing wrong?

marcohorstmann, 2020-08-19:
Had the same/similar problem yesterday when I reactivated my daughters box and
all worked exept of volume controls. I figured out that playout_controls.sh brings some
odd messages. To fix this I have done this:
1) Check which sound modules was discovered
cat /proc/asound/modules
 0 snd_bcm2835
 1 snd_usb_audio
2) Changed /usr/share/alsa/alsa.conf
```
defaults.ctl.card 0
defaults.pcm.card 0
```
to
```
defaults.ctl.card 1
defaults.pcm.card 1
```
3) Created a /etc/asound.conf file
```
cm.!default {
    type hw
    card 1
}
ctl.!default {
    type hw
    card 1
}
```
4) With amixer scontrols I see there is no PCM device
```
Simple mixer control 'Speaker',0
Simple mixer control 'Mic',0
Simple mixer control 'Auto Gain Control',0
```
5) Changed to "Speaker" to the settings/Audio_iFace_Name

6) Changed these lines in /etc/mpd.conf
I set it to device 1,0 and from mixer_control PCM to Speaker
```
audio_output {
        type            "alsa"
        name            "My ALSA Device"
#       device          "hw:0,0"        # optional
        device          "hw:1,0"        # optional
#       mixer_type      "hardware"      # optional
#       mixer_device    "default"       # optional
#       mixer_control   "PCM"           # optional
        mixer_control   "Speaker"               # optional
#       mixer_index     "0"             # optional
}
```

Sound was working before. But I think the main point here was to change the mixer_control from PCM to Speaker.
After this change I was able to modify volume via web and GPIO. I figured out that it seems that if mixer_control is
not set correctly you have in mpd commandline no "volume" value and because of this the playout_controls.sh
script is crashing with script error messages.

OlafStyle, 2020-11-17:
@marcohorstmann but I cant change alsa.conf. You dont have permission to save

s-martin, 2020-11-17:
> @marcohorstmann but I cant change alsa.conf. You dont have permission to save 

You need to add `sudo` before your command.

s-martin, 2022-05-31:
Closing. Feel free to reopen, if the problem still occurs.