MiczFlor/RPi-Jukebox-RFID #669: GPIO: Pull-up resistor needed?

Erechtheus, 2019-12-18:
This is pretty much a Newby question, but maybe someone else stumbles across this: I am currently working on the buttons for my Jukebox. However, according to the respective [manual](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/GPIO-BUTTONS.md) I am not (100%) sure if pull-up resistors are needed or not. The image im the documentation shows some resistors in the soldering, but the oreilly links use the internal raspberry resistors.  Also the documentation of gpiozero does not explicitly mention the resistors. In any case, maybe this part could be briefly mentioned in the documentation? 

FYI: I think the resistors are not needed, but I would appreciate some feedback on that matter.

Leobo71, 2019-12-20:
I have no resistance and Phoniebox works well. But don't use LED.

Erechtheus, 2019-12-21:
Hi @Leobo71, finally found my time to test this and you're right. Resistors not needed

closed 2019-12-21 by Erechtheus