MiczFlor/RPi-Jukebox-RFID #674: Can't change MP3-Details (Title, Artist, ...) via Web-interface

Reddi82, 2019-12-21:
I think the problem are some blanks in my file- and folder-names. Here is my solution:

In file "RPi-Jukebox-RFID/htdocs/trackEdit.php" change following lines:

Actual 178: $exec = "mv ".$post['folder']."/".$post['filename']." ".$Audio_Folders_Path."/".$_POST['folderNew']."/";
New 178: $exec = 'mv "'.$post['folder'].'/'.$post['filename'].'" "'.$Audio_Folders_Path.'/'.$_POST['folderNew'].'/"';

Actual 202: $exec = 'mid3v2 --artist "'.replaceUmlaute($post['trackArtist']).'" '.$post['folder'].'/'.$post['filename'];
New 202: $exec = 'mid3v2 --artist="'.replaceUmlaute($post['trackArtist']).'" "'.$post['folder'].'/'.$post['filename'].'"';

Actual 207: $exec = "mid3v2 --song '".replaceUmlaute($post['trackTitle'])."' ".$post['folder']."/".$post['filename'];
New 207: $exec = 'mid3v2 --song="'.replaceUmlaute($post['trackTitle']).'" "'.$post['folder'].'/'.$post['filename'].'"';

Actual 212: $exec = "mid3v2 --album '".replaceUmlaute($post['trackAlbum'])."' ".$post['folder']."/".$post['filename'];
New 212: $exec = 'mid3v2 --album="'.replaceUmlaute($post['trackAlbum']).'" "'.$post['folder'].'/'.$post['filename'].'"';

Actual 219: $exec = "mid3v2 --TCOM '".replaceUmlaute($post['trackComposer'])."' ".$post['folder']."/".$post['filename'];
New 219: $exec = 'mid3v2 --TCOM="'.replaceUmlaute($post['trackComposer']).'" "'.$post['folder'].'/'.$post['filename'].'"';

Actual 244: $exec = "mid3v2 -l ".$post['folder']."/".$post['filename'];
New 244: $exec = 'mid3v2 -l "'.$post['folder'].'/'.$post['filename'].'"';

Because I can't move any files from folder to folder, I found this solution (with actual/old Line the path will doubled (path from line 172+375 and 173+375: /home/pi/RPi-Jukebox-RFID/shared/audiofolders/home/pi/RPi-Jukebox-RFID/shared/audiofolders/.../Xy.mp3):

Actual 375: print "              <option value='".$keyfolder."'";
New 375: print "              <option value='".$audiofolder."'"; //$keyfolder

ctietze, 2019-12-23:
Hi @Reddi82,
I can confirm your issues... I wonder if mid3v2 changed its API. I will try to provide a PR with the bugfixes.
BR Clemens

s-martin, 2020-03-07:
Did #695 fix this? So this issue could be closed.

shoex, 2020-03-14:
> Did #695 fix this? So this issue could be closed.

For me it doesnt work. Tried today with an fresh install.

s-martin, 2023-05-05:
Closing this now. Feel free to reopen, if the problem persists.

closed 2023-05-05 by s-martin