MiczFlor/RPi-Jukebox-RFID #675: wie kann man Phoniebox updaten

Ghostsl84, 2019-12-22:
Hallo,

ich weiß ist eine blöde frage, aber ich habe das problem, das ich die Phonibox gerne updaten möchte aber leider nicht herausfinde wie das funktionieren soll.

Meine letzte Version ist:
Phoniebox Setup
Version
1.1.9-rc7
Edition
Classic edition (barebones)
MPD Server Status
active (running) since Mon 2019-12-16 18:17:05 CET; 5 days ago

also die classic version ohne spotify.
vielleicht kann mir jemand weiterhelfen?

vielen Dank
Ghostsl

Leobo71, 2019-12-22:
`cd /home/pi/RPi-Jukebox-RFID`
`git pull`

M0D-F, 2019-12-22:
Hi,
finde Sie gar nicht so blöd :) Zumindest nicht bei meinem Wissensstand.

Meine Hardware ist gerade fertig geworden. 

Weihnachten steht vor der Tür und meine Laut Leise Buttons gehen nicht. Dahe das updaten.

Ich bekomme diese Meldung:
root@raspberrypi:/home/pi/RPi-Jukebox-RFID# git pull
remote: Enumerating objects: 316, done.
remote: Counting objects: 100% (316/316), done.
remote: Compressing objects: 100% (96/96), done.
remote: Total 481 (delta 248), reused 269 (delta 217), pack-reused 165
Receiving objects: 100% (481/481), 1.14 MiB | 2.34 MiB/s, done.
Resolving deltas: 100% (300/300), completed with 39 local objects.
From https://github.com/MiczFlor/RPi-Jukebox-RFID
   8ea66eb..eac5f70  develop    -> origin/develop
   3e2c101..103ef65  master     -> origin/master
Updating 8ea66eb..eac5f70
error: Your local changes to the following files would be overwritten by merge:
        scripts/daemon_mqtt_client.py
Please commit your changes or stash them before you merge.
Aborting
root@raspberrypi:/home/pi/RPi-Jukebox-RFID#

M0D-F, 2019-12-22:
habe die entsprechende Datei umbenannt,.. dann ging es

Ghostsl84, 2019-12-23:
Hallo

habe jetzt den Update Befehl versucht

cd /home/pi/RPi-Jukebox-RFID
git pull

jedoch bekomme ich folgende Meldung und das Update ging nicht

remote: Enumerating objects: 367, done.
remote: Counting objects: 100% (367/367), done.
remote: Compressing objects: 100% (96/96), done.
remote: Total 860 (delta 299), reused 320 (delta 268), pack-reused 493
Empfange Objekte: 100% (860/860), 5.43 MiB | 1.39 MiB/s, Fertig.
Löse Unterschiede auf: 100% (553/553), abgeschlossen mit 40 lokalen Objekten.
Von https://github.com/MiczFlor/RPi-Jukebox-RFID
   5eab6ef..103ef65  master     -> origin/master
   fb5c069..eac5f70  develop    -> origin/develop
 * [neues Tag]       v1.1.9     -> v1.1.9
Aktualisiere 5eab6ef..103ef65
error: Ihre lokalen Änderungen in den folgenden Dateien würden durch den Merge
überschrieben werden:
        htdocs/inc.setLanguage.php
        htdocs/inc.setStoptimer.php
        scripts/rotary_encoder_base.py
Bitte committen oder stashen Sie Ihre Änderungen, bevor sie mergen.
Abbruch

vielen Dank für die Hilfe

Leobo71, 2019-12-23:
Du kannst frische installation durchführen, da wirst du gefragt ob deine Einstellungen behalten möchtest.

Ghostsl84, 2019-12-23:
vielen Dank und wie mache ich das?
Ich möchte nicht alle Karten und Bücher/Musik verlieren bzw. das alles wieder neu machen.

Danke

Leobo71, 2019-12-23:
Oh, wenn alles läuft dan lieber lassen das system so wie er ist, eventuell ein backup vorher machen.
Dann normal Verbindung über SSH mit RPi und "One Line installer" drauf, enter und folgst Einweisung, Wirst du gefragt ob Bibliothek, Einstellungen usw soll behalten werden. Da wird auch ein backup erstellt. Aber wie gesagt sicher ist sicher werde früher besser alles selber sichern.
Gruss

splitti, 2019-12-23:
Bitte bedenke, dass Du Updates eher vom OS ziehen solltest.
Das GitHub Projekt sind ja keine Sourcen, sondern eher der Klebstoff, der das Projekt zusammen hält und in Form bringt. Klar kann es ein paar Verbesserungen geben oder sogar Features. Wenn Du aber nicht gezielt etwas davon brauchst, würde ich das Projekt nicht aktualisieren. Falls Du Dich dennoch entschließen solltest das Projekt upzudaten, können Probleme folgen. Daher unbedingt ein Backup erstellen.

Das OS kannst Du ja einfach mit
`sudo apt update
sudo apt upgrade` 
auf den aktuellen Stand bringen.

ZyanKLee, 2020-01-19:
ohje - ich denke ich werde die Tage mal eine Anleitung schreiben wie ich meine Systeme aktualisiere. Leider gehört da tatsächlich etwas mehr dazu, da der Update-Teil des Install scripts derzeit kaputt ist.

Ghostsl84, 2020-01-19:
Das wäre wirklich nett und es wäre super, wenn man das als Anleitung aufnehmen könnte

MiczFlor, 2020-01-19:
Hi @Ghostsl84 
das stimmt, das sollte ich mal machen. Wenn man noch nicht auf `buster` ist sondern `stretch`, dann kommt man leider um einen neuen Install nicht herum... so viel kann ich shcon sagen...

MiczFlor, 2020-01-19:
@Ghostsl84 
I wrote a quick page for the wiki:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Upgrade

s-martin, 2022-12-24:
Closing as now described in the wiki

closed 2022-12-24 by s-martin