MiczFlor/RPi-Jukebox-RFID #677: Hardware Buttons Forward/Backward Vol +/- does not work

M0D-F, 2019-12-23:
Hi,
at first, thank you for this great Projekt.

I have some Issues with the GPIO Buttons Vol up / down and also Forward and Backward.

Is there a possibility to test the buttons and see it in the command line?

Thanks for your help

pi@raspberrypi:~ $ /home/pi/RPi-Jukebox-RFID/scripts/helperscripts/Analytics_AfterInstallScript.sh
************************************
*** PHONIEBOX INFO
*** version: 2.0 rc7
*** edition: plusSpotify
*** Audio_iFace_Name: PCM
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 7
*** Max_Volume_Limit: 100
*** Idle_Time_Before_Shutdown: 60
*** Second_Swipe: PAUSE
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: ON
************************************
*** CONF FILES DEFAULT

*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 8.2K Oct  3 14:55 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 2.1K Oct  3 14:55 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 575 Oct  3 14:55 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
ls: cannot access '/etc/php/7.0/fpm/php.ini': No such file or directory
*** /etc/sudoers
-r--r----- 1 root root 784 Oct  3 14:55 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 296 Oct  3 14:55 /etc/systemd/system/phoniebox-rfid-reader.service
-rw-r--r-- 1 root root 340 Oct  3 14:55 /etc/systemd/system/phoniebox-startup-sound.service
-rw-r--r-- 1 root root 289 Oct  3 14:55 /etc/systemd/system/phoniebox-gpio-buttons.service
-rw-r--r-- 1 root root 291 Oct  3 14:55 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 13K Dec 16  2018 /etc/mpd.conf
music_directory         "/var/lib/mpd/music"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
#       mixer_control   "PCM"           # optional
#       mixer_control   "PCM"           # optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 131 Oct  3 14:55 /etc/dhcpcd.conf
static ip_address=192.168.178.128/24
static routers=192.168.178.1
static domain_name_servers=8.8.8.8 192.168.178.1
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw-rw-r-- 1 root netdev 149 Oct  3 14:55 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition

*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Oct  3 14:55 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Oct  3 14:55 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
-rw-r----- 1 mopidy root 1.1K Oct  3 14:55 /etc/mopidy/mopidy.conf
username 
*** ~/.config/mopidy/mopidy.conf
ls: cannot access '/home/pi/.config/mopidy/mopidy.conf': No such file or directory
cat: /home/pi/.config/mopidy/mopidy.conf: No such file or directory

Florian1974, 2019-12-28:
I try to give a few hints. 
If 2 of the 5 buttons are working, I would start with checking the pinout. 
IO number vs Pin number can be quiete confusing with the RaspberryPi. https://www.raspberrypi.org/documentation/usage/gpio/ has picture containing both information. 

For debugging on the command line there are several options: 
- stop the  gpio service and call the script from the command line. This will print the changes mpd sees everytime a button is pressed. Try it first with Play to see what it looks with a working button
` sudo systemctl stop phoniebox-gpio-buttons.service`
`cd ~/RPi-Jukebox-RFID`
`./scripts/gpio-buttons.py`
- as above but additionaly edit gpio-buttons.py and add a print("button xyz was pressed  ") in the def def_but section. see example below. this should print the message if the button is pressed
`def def_volD():`
`    print("button volume down was pressed")`
`    check_call("./scripts/playout_controls.sh -c=volumedown", shell=True)`

Squelsh, 2019-12-30:
For debugging purposes there is the handy command line utility `gpio` which can be installed via
`sudo apt-get install wiringpi`
Have a look at http://wiringpi.com/the-gpio-utility/ for all commands.
The command `gpio readall` shows you that state of all I/Os.
You can have a "live view" by executing `watch -n 0.1 gpio readall`

M0D-F, 2020-01-08:
@Florian1974 @Squelsh 
Thanks a lot! This helped me to find the issue.
My buttons are not working relieable. I ordered new ones. 
When they arrived and I have some time, I give you a feedback :)

Maybe meanwhile the reaction failure with the buttons is already fixed.

Thanks again!

closed 2020-01-15 by MiczFlor