MiczFlor/RPi-Jukebox-RFID #679: Volume control reacts with delay

JuliH29, 2019-12-25:
Hi!

Somehow the volume control on my Phoniebox (running 2.0 RC6) always reacts with a couple of seconds of delay.
I build a circuit board as [discriped in the wiki](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons#optional-step-wire-hardware-pullup-resistor). All the other buttons work flawlessly.
Any idea how I could fix that?

Thanks,
Julius

dbffm, 2019-12-26:
Do you use Buttons or a rotary encoder to control the volume? I have the same behaviour (delay) with a rotary encoder but don't have a solution yet.

JuliH29, 2019-12-27:
I have to buttons to controll the volume.

s-martin, 2020-03-07:
Is this still an issue?

JuliH29, 2020-03-07:
> Is this still an issue?

Yes! It’s not on my list of priorities for the moment though. It’s not ideal but I can live with it.

s-martin, 2020-03-07:
What version are you currently using?

Could you also post the result of `git status`?

JuliH29, 2020-03-08:
I'm using 2.0 RC8

> Could you also post the result of `git status`?

> Your branch is up to date with 'origin/develop'.
> 
> Changes not staged for commit:
>   (use "git add <file>..." to update what will be committed)
>   (use "git checkout -- <file>..." to discard changes in working directory)
> 
> 	modified:   htdocs/api/playlist/playsinglefile.php
> 	modified:   htdocs/api/playlist/shuffle.php
> 	modified:   htdocs/api/playlist/single.php
> 	modified:   htdocs/inc.controlPlayer.php
> 	modified:   htdocs/inc.loadControls.php
> 	modified:   htdocs/inc.loadCover.php
> 	modified:   htdocs/inc.setDebugLogConf.php
> 	modified:   htdocs/inc.setVolume.php
> 	modified:   htdocs/search.php
> 	modified:   htdocs/utils/Files.php
> 	modified:   htdocs/utils/Strings.php
> 
> Untracked files:
>   (use "git add <file>..." to include in what will be committed)
> 
> 	settings/Change_Volume_Idle
> 	settings/Lang
> 	settings/Min_Volume_Limit
> 	settings/debugLogging.conf
> 	settings/global.conf
> 
> no changes added to commit (use "git add" and/or "git commit -a")

Groovylein, 2020-05-17:
I'm having the same issue.

Groovylein, 2020-05-17:
First investigation:
`gpio-buttons.py` and `playout_controls.sh` acting fast. There must be some delay after calling:

```
echo -e setvol +$VOLPERCENT\\nclose | nc -w 1 localhost 6600
```

Groovylein, 2020-05-22:
Found this thread:
https://forum-raspberrypi.de/forum/thread/45100-phoniebox-verzoegerung-beim-einstellen-der-lautstaerke-ueber-drehregler/

After doing following steps recommended in [this post](https://forum-raspberrypi.de/forum/thread/45100-phoniebox-verzoegerung-beim-einstellen-der-lautstaerke-ueber-drehregler/?postID=408399#post408399) no delay detected.

```
sudo apt-get install mopidy-alsamixer
```
modify `/etc/mopidy/mopidy.conf` to:
```
[audio]
mixer = alsamixer
output = alsasink
mixer_volume = 30

[alsamixer]
card = 0
control = PCM
```

I will have a look if we can use this in our install script

MiczFlor, 2020-05-22:
Hi @Groovylein 
that would be awesome, but possibly it should not only go into the install script?
`mopidy` is an extra only installed *if* Spotify is used. And, I assume, if Google Music is used, too - which we don't have in the install script but on the wiki here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Enable-Google-Play-Music-GMusic
And in the `requirements-gmusic.txt` here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/requirements-gmusic.txt
There also is a `requirements-spotify.txt` file here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/requirements-spotify.txt

s-martin, 2020-05-22:
@Groovylein and @JuliH29: do you experience this issue with Spotify edition (i.e. mopidy) or classic edition?

If it's Spotify edition we should add the mentioned changes https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/679#issuecomment-632700468 in the section where Spotify is installed.

I think we don't need to update the requirements files, because it's an apt package.

Groovylein, 2020-05-22:
@s-martin I'm having the Spotify version. Did not try it with the basic version since the phoniebox is in use.

s-martin, 2020-05-22:
I have the classic edition and never experienced this issue.

I think for the moment we can assume it only applies to Spotify edition.

JuliH29, 2020-05-22:
I do experience the issue and am using the Spotify edition.
I will try to solution described by @Groovylein within the next days and report back here.

Groovylein, 2020-05-24:
My experience:
- The volume is lower than the other mixer.
- I need to change the startup volume
- I need to change the increase volume percentage

dbffm, 2020-05-25:
I can confirm that it is working much better (2.0 rc6; Spotify Edition). 
The only change I had to make was to set 
`control = Master` 
in mopidy.conf since I use a hifiberry miniamp. 

> After doing following steps recommended in [this [...]

s-martin, 2020-05-25:
Great to hear this fixes the issue.

If somebody adds a PR we add this ASAP.

JuliH29, 2020-06-14:
I just saw that version 2.1 RC1 is on the develop channel. Are there any release notes available? Is the thing with the delayed volume control part of it?
I definitely would have created a pull request. Unfortunately I don't have the necessary expertise. Thank you for all your great work here!

s-martin, 2020-06-14:
> I just saw that version 2.1 RC1 is on the develop channel. Are there any release notes available? Is the thing with the delayed volume control part of it?

Here you can find all the issues and PRs, which are supposed to be part of 2.1  - please check still open and already closed:
https://github.com/MiczFlor/RPi-Jukebox-RFID/milestone/3

JuliH29, 2020-06-14:
> Here you can find all the issues and PRs, which are supposed to be part of 2.1 - please check still open and already closed:
> https://github.com/MiczFlor/RPi-Jukebox-RFID/milestone/3
Great, thanks! I did not discover that yet!

s-martin, 2020-06-14:
Many issues should be allocated to a milestone, so you can click on that.

StefanD986, 2020-12-21:
I am using the **classic** version but I still get 1-6 seconds delay between turning the rotary encoder and MPD reacting.

For me the length of the delay depends on how quickly I turn the knob. If I go 1 step, it takes about 1 second for MPD to ackknowledge:
```
Dec 21 20:51:38 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:51:39 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
Dec 21 20:51:39 raspberrypi PhonieboxGPIOControl[5679]: OK  
```

1 second is already annoying enough, but it gets even worse when I turn the knob quickly, the delay until MPD reacts to the first increment is much longer (in the example below about 6 seconds):
```
Dec 21 20:52:44 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:52:44 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:52:44 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:52:44 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:52:44 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:52:45 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:52:45 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:52:45 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:52:45 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:52:45 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:52:45 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:52:51 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
Dec 21 20:52:51 raspberrypi PhonieboxGPIOControl[5679]: OK
Dec 21 20:52:51 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
Dec 21 20:52:51 raspberrypi PhonieboxGPIOControl[5679]: OK
Dec 21 20:52:51 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
Dec 21 20:52:51 raspberrypi PhonieboxGPIOControl[5679]: OK
Dec 21 20:52:51 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
Dec 21 20:52:51 raspberrypi PhonieboxGPIOControl[5679]: OK
Dec 21 20:52:51 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
Dec 21 20:52:51 raspberrypi PhonieboxGPIOControl[5679]: OK
Dec 21 20:52:52 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
Dec 21 20:52:52 raspberrypi PhonieboxGPIOControl[5679]: OK
Dec 21 20:52:52 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
Dec 21 20:52:52 raspberrypi PhonieboxGPIOControl[5679]: OK
Dec 21 20:52:52 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
Dec 21 20:52:52 raspberrypi PhonieboxGPIOControl[5679]: OK
Dec 21 20:52:52 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
Dec 21 20:52:52 raspberrypi PhonieboxGPIOControl[5679]: OK
Dec 21 20:52:52 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
Dec 21 20:52:52 raspberrypi PhonieboxGPIOControl[5679]: OK
Dec 21 20:52:52 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
Dec 21 20:52:52 raspberrypi PhonieboxGPIOControl[5679]: OK
```

Since I am using the classic version I think it the previous solutions (installing mopidy-alsamixer) doesn't seem to make much sense to me. Any other suggestions?

s-martin, 2020-12-21:
@veloxidSchweiz, any ideas?

michnixweiss, 2021-01-09:
I am using a Raspi Zero W with classic version and WM8960.
Same problem here.
So the previous solution wasn't promising I gave it a try, but no improvement. Are there other things to try?

jonach66, 2021-01-30:
Had the same problem on a Pi Zero, found a alternative solution running an extra service & a python script:

systemd service - test_rotary_class.service:
```
[Unit]
Description=Test Rotary Class
After=multi-user.target
 
[Service]
Type=simple
ExecStart=/usr/bin/python /home/pi/scripts/test_rotary_class.py
Restart=on-abort
 
[Install]
WantedBy=multi-user.target
```

and class definition - rotary_class.py
```

#!/usr/bin/env python
#
# Raspberry Pi Rotary Encoder Class
# $Id: rotary_class.py,v 1.2 2014/01/14 07:30:07 bob Exp $
#
# Author : Bob Rathbone
# Site : http://www.bobrathbone.com
#
# This class uses standard rotary encoder with push switch
#
#
import RPi.GPIO as GPIO

class RotaryEncoder:
    CLOCKWISE=1
    ANTICLOCKWISE=2
    BUTTONDOWN=3
    BUTTONUP=4
    PINC_UP=5
    PINC_DOWN=6
    PIND_UP=7
    PIND_DOWN=8
    rotary_a = 0
    rotary_b = 0
    rotary_c = 0
    last_state = 0
    direction = 0
    # Initialise rotary encoder object
    def __init__(self,pinA,pinB,pinC,pinD,button,callback):
        self.pinA = pinA
        self.pinB = pinB
        self.pinC = pinC
        self.pinD = pinD
        self.button = button
        self.callback = callback
        GPIO.setmode(GPIO.BCM)
        # The following lines enable the internal pull-up resistors
        GPIO.setwarnings(False)
        GPIO.setup(self.pinA, GPIO.IN, pull_up_down=GPIO.PUD_UP)
        GPIO.setup(self.pinB, GPIO.IN, pull_up_down=GPIO.PUD_UP)
        GPIO.setup(self.pinC, GPIO.IN, pull_up_down=GPIO.PUD_UP)
        GPIO.setup(self.pinD, GPIO.IN, pull_up_down=GPIO.PUD_UP)
        GPIO.setup(self.button, GPIO.IN, pull_up_down=GPIO.PUD_UP)
        # Add event detection to the GPIO inputs
        GPIO.add_event_detect(self.pinA, GPIO.FALLING,callback=self.switch_event)
        GPIO.add_event_detect(self.pinB, GPIO.FALLING,callback=self.switch_event)
        GPIO.add_event_detect(self.pinC, GPIO.BOTH,callback=self.pinc_event, bouncetime=400)
        GPIO.add_event_detect(self.pinD, GPIO.BOTH,callback=self.pind_event, bouncetime=400)
        GPIO.add_event_detect(self.button, GPIO.BOTH,callback=self.button_event, bouncetime=400)
        return
    # Call back routine called by switch events
    def switch_event(self,switch):
        if GPIO.input(self.pinA):
            self.rotary_a = 1
        else:
            self.rotary_a = 0

        if GPIO.input(self.pinB):
            self.rotary_b = 1
        else:
            self.rotary_b = 0

        self.rotary_c = self.rotary_a ^ self.rotary_b
        new_state = self.rotary_a * 4 + self.rotary_b * 2 + self.rotary_c * 1
        delta = (new_state - self.last_state) % 4
        self.last_state = new_state
        event = 0

        if delta == 1:
            if self.direction == self.CLOCKWISE:
            # print "Clockwise"
                event = self.direction
            else:
                self.direction = self.CLOCKWISE
        elif delta == 3:
            if self.direction == self.ANTICLOCKWISE:
                # print "Anticlockwise"
                event = self.direction
            else:
                self.direction = self.ANTICLOCKWISE
        if event > 0:
            self.callback(event)
        return
    # Push button event
    def pinc_event(self,pinC):
        if GPIO.input(pinC):
            event = self.PINC_UP
        else:
            event = self.PINC_DOWN
            self.callback(event)
        return
    # Push button event
    def pind_event(self,pinD):
        if GPIO.input(pinD):
            event = self.PIND_UP
        else:
            event = self.PIND_DOWN
            self.callback(event)
        return
    # Push button event
    def button_event(self,button):
        if GPIO.input(button):
            event = self.BUTTONUP
        else:
            event = self.BUTTONDOWN
            self.callback(event)
        return
    # Get a switch state
    def getSwitchState(self, switch):
        return GPIO.input(switch)
# End of RotaryEncoder class

```

...and the responding python script - test_rotary_class.py:

```
#!/usr/bin/env python
#
# Raspberry Pi Rotary Test Encoder Class
#
# Author : Bob Rathbone
# Site : http://www.bobrathbone.com
#
# This class uses a standard rotary encoder with push switch
#
import subprocess
import sys
import time
from rotary_class import RotaryEncoder

# Define GPIO inputs
PIN_A = 12 # Pin 32
PIN_B = 16 # Pin 36
PIN_C = 23 # Pin 16
PIN_D = 24 # Pin 18
BUTTON = 20 # Pin 38

# This is the event callback routine to handle events
def switch_event(event):
	if event == RotaryEncoder.CLOCKWISE:
		print ("Clockwise")
		# subprocess.call("/usr/bin/mpc volume +10")
		subprocess.call("mpc volume +1", shell=True)
	elif event == RotaryEncoder.ANTICLOCKWISE:
		print ("Anticlockwise")
		subprocess.call("mpc volume -1", shell=True)
	elif event == RotaryEncoder.BUTTONDOWN:
		print ("Button down")
		# mute
		subprocess.call("mpc toggle", shell=True)
	elif event == RotaryEncoder.PINC_DOWN:
		print ("Pin C down")
		# mute
		subprocess.call("mpc next", shell=True)
	elif event == RotaryEncoder.PIND_DOWN:
		print ("Pin D down")
		# mute
		subprocess.call("mpc prev", shell=True)
	elif event == RotaryEncoder.BUTTONUP:
		print ("Button up")
	return

# Define the switch
rswitch = RotaryEncoder(PIN_A,PIN_B,PIN_C,PIN_D,BUTTON,switch_event)

while True:
	time.sleep(10)
```
I`m using this with a rotary encoder for volume (includes a button for play/stop), and two buttons for previous & next audio file.

Enable the systemd service, scripts to the right folders...

kr
jo

veloxidSchweiz, 2021-02-03:
Hi,
sorry for the inconvenience. And sorry for the late reply, my github messages got sorted into a folder and i didn't check it.
For all RPi Zero users:
It actually seems to be a speed problem. The implementation of gpiozero was giving similar issues. This is quite 'interesting' though. I hope i find some time to to dig into it in one of the next days.

@StefanD986 : This is interesting,
Dec 21 20:52:45 raspberrypi PhonieboxGPIOControl[5679]: INFO:GPIODevices.rotary_encoder:RotaryVolumeControl: Calling functionIncr 1
Dec 21 20:52:51 raspberrypi PhonieboxGPIOControl[5679]: OK MPD 0.21.4
You get the information that the rotary knob was turned and the functionIncr was called, but the 'feedback' of mpd taks multiple seconds. One second delay for changing the volume by one is already to much. This is a little unclear for me what the cause.   This could be investigated adding a logger output in https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/function_calls.py#L21

I do see one issue with the rotary implementation: 
It should not try to increase everytime by one, but should cal the functionIncr X where X is bigger than 1 as soon as you turn faster. This is something we should investigate... maybe changing the timeBase could help here...

qwert1337, 2023-09-11:
This is how I solved it for my pi zero with hifiberry miniamp:

sudo apt-get install mopidy-alsamixer

alsa.conf
```
pcm.!hifiberry {
  type hw card 0
}
ctl.!default {
  type hw
  card 0
}
pcm.!default {
    type            plug
    slave.pcm       "softvol"   #make use of softvol
}
pcm.softvol {
    type            softvol
    slave {
        pcm         "hifiberry"      #redirect the output to hifiberry (instead of "hw:0,0")
    }
    control {
        name        "PCM"       #override the PCM slider to set the softvol volume level globally
        card        0
    }
}
```

mopidy.conf
```
[audio]
mixer = alsamixer
output = alsasink
mixer_volume = 10

[alsamixer]
card = 0
control = PCM
``` 

For me the volume set in % appears to be lower after these changes, so you will probably need to adapt the volume steps and the defaults.