MiczFlor/RPi-Jukebox-RFID #680: Standard shuffle option does not work for Spotify playlists calls

skuenstl, 2019-12-25:
Hi all,

i use the phoniebox to call Spotify Playlist only. No local music files are stored on my phoniebox.

My problem is that the standard shuffle function for a stored card does not work. Spotify starts with shuffle off every time although the shuffle option is switched on for the stored card. Because of that fact I always start the web interface after the phoniebox is up and running to switch shuffle on at the Spotify page of the phoniebox.

Does anyone know that problem and can give some hints to fix it?

MiczFlor, 2020-01-04:
I have tested shuffle only for local files. Would be surprised if it worked for Spotify out of the box :)

skuenstl, 2020-01-12:
okay thx for your explanation.