MiczFlor/RPi-Jukebox-RFID #681: Hifiberry AMP2 Setup für PhonieBox

helge1979, 2019-12-25:
Hallo,

ich bin langsam am verzweifeln, beim Einrichten vom hifberry amp2 zusammen mit der Phoniebox.

Mittleweile bin ich so weit, dass die Weboberfläche alle meine Ordner anzeigt und die MP3's auch abspielt (zumindest wird der Liedfortschritt angezeigt). Allerdings kommt aus den Lautsprechern kein Sound. Fast, wie wenn Mute noch irgendwie konfiguriert wäre.

Im MPD Log keine Auffälligkeiten, laut log wird auch hier das MP3 File abgespielt. Wenn ich mir den MPC anschaue sagt dieser mir auch, dass er gerade das entsprechende MP3 File abspielt.

Bitte um Feedback :-) Per paypal war auch was für die Kaffeekasse drin ;-)

splitti, 2019-12-26:
Hallo Helge,

Ich kenne mich zwar eher mit dem MiniAmp aus, aber einige Probleme sind davon unabhängig...

Kriegst du den Bootsound? Wenn der nicht da ist, dann stimmt was grundlegendes nicht, z.b. config txt.
Eine weitere Option ist die asound.conf und ob das richtige iface eingetragen ist.
Was auch gerne vorkommt ist, das nicht verwendete Gpios in der gpiobuttons.conf geladen werden, diese aber eigentlich vom Hifiberry werden... 

Vielleicht hilft dir das?

Grüße
Splitti

helge1979, 2019-12-26:
Nein, habe kein Bootsound...

Wenn ich einen Speaker-Test mache läuft dieser zwar, ich bekomme aber trotzdem kein Sound

pi@neleplayer:~ $ speaker-test -c2 -Dplughw:0 -twav

speaker-test 1.1.8

Playback device is plughw:0
Stream parameters are 48000Hz, S16_LE, 2 channels
WAV file(s)
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 128 to 131072
Period size range from 64 to 65536
Using max buffer size 131072
Periods = 4
was set period_size = 32768
was set buffer_size = 131072
 0 - Front Left
 1 - Front Right
Time per period = 0.741728

Was ist denn das richtige iface für die asound.conf? Und wo finde ich die gpiobuttons.conf?

splitti, 2019-12-26:
Irgendwas stimmt dann grundlegend nicht. Weißt du die notwendigen GPIOs für den Amp2? Ggf. kollidieren diese mit irgendwas... Wie hast du den Amp2 denn in der config.txt eingetragen?

Benzinbruder, 2019-12-27:
Hallo helge1979,

hast du die Zuordnung der GPIOs in `gpio-buttons.py` kontrolliert? Die GPIOs dürfen NICHT doppelt belegt werden. Ich hatte ein ganz ähnliches Problem wie du, und damit hatte ich es dann gelöst.

Laut [HiFiBerry](https://www.hifiberry.com/docs/hardware/gpio-usage-of-hifiberry-boards/) benötigt der AMP2 die GPIOs 2-3 sowie 18-21. Diese GPIOs dürfen in `gpio-buttons.py` nicht zugeordnet werden. Wenn dort also z.B. etwas Ähnliches wie `shut = Button(3)` zu finden ist, wird der AMP2 nicht funktionieren.

helge1979, 2019-12-27:
danke für den Hinweis, habe in der gpio-buttons.py die entsprechenden Zeilen auskommentiert. Aber der HifiBerry macht immer noch kein mux...

helge1979, 2019-12-27:
> Irgendwas stimmt dann grundlegend nicht. Weißt du die notwendigen GPIOs für den Amp2? Ggf. kollidieren diese mit irgendwas... Wie hast du den Amp2 denn in der config.txt eingetragen?

config.txt

```
dtoverlay=hifiberry-dacplus
# Enable audio (loads snd_bcm2835)
#dtparam=audio=on
```

splitti, 2019-12-27:
Es sind doch immer die gängigen Fehler... Also die gpio-buttons.conf hat 100% keine Doppelbelegungen? Hast Du Buttons oder Display angeschlossen, falls ja, schreib Mal deine gpio-buttons hier rein.
Unabhängig von der gpio-buttons.conf sollte dennoch der Bootsound kommen. Das ist zumindest meine Erfahrung, wenn es eine Doppelbelegung gibt.

Alsamixer ist hoch gedreht?

Hast du schon Mal in meinem Blog wegen der asound.conf geschaut? http://splittscheid.de/selfmade-phoniebox/
Dort steht es beschrieben, wie die asound.conf gepflegt werden kann. Dann muss aber das iface noch angepasst und die Box gebootet werden. Mein Bauchgefühl sagt aber, dass vorher was nicht geht wegen dem fehlenden Bootsound.

helge1979, 2019-12-27:
hab es nun am Laufen, hab das image nochmal komplett neu aufgesetzt...jetzt habe ich nur noch das Problem, dass der MPD nicht automatisch beim booten startet, aber das sollte des kleinste Problem sein.

Hab mir Dein Blog angeschaut...Cooles Teil...

Display überlege ich mir aber auch noch...

helge1979, 2019-12-30:
Danke euch beiden für die Hilfe.

1 Sache funktioniert noch nicht und zwar der startupsound.

Wenn ich den Dienst starte wird der Sound wiedergegeben, allerdings nicht automatisch beim Systemstart.

Habt Ihr da noch Ideen, an was das liegen könnte?

Benzinbruder, 2019-12-30:
Sorry, aber mir fällt dazu nichts ein. Phoniebox startet automatisch beim Start, oder?

splitti, 2019-12-30:
Was meist du mit
Wenn ich den Dienst starte? Wochen Dienst meinst du bitte

helge1979, 2020-01-02:
das meine ich damit:

sudo systemctl status phoniebox-startup-sound.service
● phoniebox-startup-sound.service - Phoniebox Startup Sound
   Loaded: loaded (/etc/systemd/system/phoniebox-startup-sound.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

Wenn ich den Dienst manuell starte:
sudo systemctl start phoniebox-startup-sound.service

wird mir der Start Sound wieder gegeben.

Jemand eine Idee, an was das liegen könnte?

splitti, 2020-01-03:
Nutzt Du die Classic oder Spotify Version?
Ich hab gestern Abend mal Buster und alles drum herum installiert. Hab massive Probleme mit dem mpd Service, da stimmt m.E. gerade generell etwas nicht. Ich hätte die Spotify Version installiert...

MiczFlor, 2020-01-04:
Hi @helge1979 
Ich finde den Thread gerade nicht, aber wenn ich mich recht erinnere, dann gab es unterschiedliche Ergebnisse in Abhängigkeit von den Soundcards / Mini-Jack bei Änderungen im service file:

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/misc/sampleconfigs/phoniebox-startup-sound.service.stretch-default.sample 

Diese beiden Zeilen funktionieren mit Mini-Jack out, aber (frag nicht...) bei anderen Soundkarten mussten da andere Werte rein.
~~~
After=default.target
WantedBy=multi-user.target
~~~

Benzinbruder, 2020-01-04:
Also ich hab eine Classic auf buster am Laufen und mittlerweile funktioniert alles soweit wie gedacht.
Ich nutze allerdings einen MiniAmp von HiFiBerry.

tcse10, 2020-01-04:
Hi,
ich habe das gleiche Problem mit einem HIfiBerry Miniamp und einer hedless Installation. 
Das default.target wird leider offenbar trotzdem auf graphical.targe gesetzt
`pi@phonibox:~ $ systemctl get-default
graphical.target
`
Gibt es einen speziellen Grund, warum der hier after=default.target verwendet wird? Mir erschient after=alsa-state.service sinnvoller.

helge1979, 2020-01-04:
@tcse10 Perfekt! Damit hat es geklappt! 1000 Dank

closed 2020-01-04 by helge1979

Hans-Haefner, 2020-05-09:
Auch wenn das hier schon etwas älter ist wollte ich noch etwas anmerken:
Bitte beachten das man den HifiBerry AMP2 nicht zusammen mit dem ON/OFF-Shim einsetzen kann da dieser den GPIO4 belegt und es somit Probleme mit dem Mute des HifiBerry AMP2 geben wird.

Benzinbruder, 2020-05-09:
> 
> 
> Auch wenn das hier schon etwas älter ist wollte ich noch etwas anmerken:
> Bitte beachten das man den HifiBerry AMP2 nicht zusammen mit dem ON/OFF-Shim einsetzen kann da dieser den GPIO4 belegt und es somit Probleme mit dem Mute des HifiBerry AMP2 geben wird.

Thanks for pointing this out!
This is valuable information for newcomers trying to figure out what components to use!

PR #697 (2020-01-08): changed "After" to alsa-state instead of default.target
make sure startup sound also works on heedless installations. Fixes #681