MiczFlor/RPi-Jukebox-RFID #683: Add Architecture description

joshwapohlmann, 2019-12-28:
Very often I find myself in the position, that I'd like to contribute, but have to spend hours to just understand what part of the project/code does what.
In many cases this is not necessary for most small features and I feel this would lower the barrier for people to actually contribute small changes if they don't have to dig through all the code.

MiczFlor, 2020-01-04:
Hi @joshwapohlmann 
good point. And needs to be done the sooner the better - so that more people can contribute. Can you point me to an example file that you think would help?
Digging through Phoniebox code is also digging through many different styles of coding :) being open source and benefiting from people like yourself commiting their code, makes it a very colourful code base.

joshwapohlmann, 2020-01-05:
@MiczFlor Maybe add a coding style section in the CONTRIBUTION.md, e.g. python pep8.
Also point to the architecture document (maybe a structure.md in /doc ?).
I have no example file at hand, sorry :(. A general overview about the components would help:
So far I see 3 "components":
1. python for reading the RFID cards.
2. python for controling mpd depending on RFID value read.
2. PHP website for interfacing with MPD and other controls. (or does it interface with the python script?)
3. a ton of shellscripts for installation and configuration.

If you could clarify this, add a few more details who is talking to who, doing what ..... this would be great. Bonus points for pointing roughly at the code in the repository :)

Squelsh, 2020-01-06:
And please clarify if **mopidy** requires **mpd** or if it is a replacement. I still can't tell.

Squelsh, 2020-01-06:
Found it in the installer script: Mopidy replaces mpd in the Spotify version... 

```
# Spotify config
if [ $SPOTinstall == "YES" ]
then
        sudo systemctl disable mpd
        sudo systemctl enable mopidy

```

I think, this would be really helpful in the description.

s-martin, 2020-01-12:
I agree this would be very helpful.

In addition some roadmap what’s planned to do or could be picked up by contributors could help. E.g. is it planned to migrate the bash scripts to python or should it stay this way?

MiczFlor, 2020-01-15:
Hi @s-martin 
Phoniebox has come a long way and still carries a lot of infant code with it. The "never change a running system" policy has made it accessible and stable enough for many. I remember a few points in the history (e.g. replacing `cvlc` with `mpd`) which created many sleepless nights :)
If there was a number of committed developers, who spend some of their downtime, I would be happy to be in a more managerial role and community contact. Then a roadmap would be good.
I still would like to keep this as simple as possible: so that many, many people can contribute. I often get messages that this is somebody's first Pi / Linux / coding project and that makes me very happy. If Phoniebox were to become a superslick install but difficult to contribute code base, it would lose some of its strength.

joshwapohlmann, 2020-01-15:
@MiczFlor Thanks for the statement. I understand the priorities you make in terms of the design.
Before discussing the future course of this project I'd (still) like to stress the importance of documenting the **current** design and structure so that everybody is on the same page. 

I will provide a first graphical representation of the project as I understand it (without digging too much into the code) and you could help me correct and clarify bits and parts of it.
When finished we can add this graphic/diagram to the README or simply drop it into the docs/ folder. 

What do you think?

MiczFlor, 2020-01-15:
Hi @joshwapohlmann 
Yes, please! And thanks for the effort. What tool will you use for the graph(s)?

joshwapohlmann, 2020-01-15:
@MiczFlor I will use inkscape to create a SVG image.

joshwapohlmann, 2020-01-15:
Okay,
First let's explain the code structure.
/ - root directory, here is everything to get started and install the project.
/display - code for connecting and controlling a display attached to the respberry pi. See directorie's README for details
/docs - documentation and howtos for the project besides the README
/htdocs - PHP web interface 
/logs - not holding any source files
/misc - sample configuration files, mp3 testfiles
/scripts - ???
/scripts/helperscripts - help to configure and manage your phoniebox
/scripts/installscripts - installation scripts which modify the system to work with/as a phoniebox
/scripts/pirc522 - python driver for the RFID module
/scripts/python-phoniebox - ???
/scripts/userscripts - ???
/settings - settings how the phoniebox should behave (other locations/files as well ???)
/shared/audiofolders - files available to be played
/shared/shortcuts - ???
/tests -  ???

Is this listing correct @MiczFlor? Could you please fill in the gaps marked with "???"
After this I continue to draw the diagram.

MiczFlor, 2020-01-18:
Hi @joshwapohlmann 
to make the process a bit transparent, I started to add a wiki page
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Architecture_FoldersFiles
possibly that could be a space where we can finalize the content and then you take it to inkscape

Squelsh, 2020-01-26:
Hi Micz,

the folder structure is already a nice starting point. Could you clarify in 3 sentences, which backend is responsible for which music? As I still don't understand if local music / playlists in the Spotify Edition are handled by mopidy or its mpd server part or completely different.

I only noticed that the playlist filestructure differs between Sotify and classic edition.

Squelsh, 2020-01-26:
Sorry, just found this one: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/745#issuecomment-577190340
But I still don't understand why my mopidy installation can't handle the .m3u files

s-martin, 2022-12-24:
For version 2 there’s a description in the wiki, for future3 there’s an extensive documentation

closed 2022-12-24 by s-martin